stronnag/mwptools #87: Bullet GCSS minor protocol changes

danarrib, 2021-02-10:
Hello Jonathan. Just to let you know that I made some minor changes on the Bullet GCSS protocol this weekend.

All the field that were "double" are now integers. Which means that protocol doesn't send the decimal separator (dot) anymore.

* GPS Latitude and Longitudes needs to be divided by 10000000
* Current Draw, GPS HDOP, Battery and Cell Voltages needs to be divided by 100
* Flight mode is not a String anymore. It's now an Integer that is translated on the client side:

Id | Flight Mode
-- | --
1 | MANUAL
2 | RTH
3 | A+PH
4 | POS H
5 | 3CRS
6 | CRS
7 | WP
8 | ALT H
9 | ANGLE
10 | HORIZON
11 | ACRO

I don't plan to change the protocol structure anytime soon... I plan, however, to implement some kind of cryptography, that will encode the entire message. But I'm not sure yet the details... 

https://github.com/danarrib/BulletGCSS/commit/3b094b4176ccd34585c644d4a9a47a0531088548

stronnag, 2021-02-10:
Thanks for letting me know,  at 1400 UTC `fl2mqtt` should run on the new protocol; broker.emqx.io org/mwptools/mqtt/otxplayer

And I was expecting something radical (protocol buffers for example).

danarrib, 2021-02-10:
That's nice. I'm following the flight here.

[image]

I noticed that fl2mqtt doesn't send the waypoint 0 to the MQTT. This caused a minor bug on the Elevation data retrieval routine, that I already fixed.

Also noticed that fl2mqtt doesn't send the `3df` field (a flag indicating if gps has 3d fix or not). That's why the GPS sat counter field is red all the time.

stronnag, 2021-02-10:
`fl2mqtt` sends 3df:, from my mqtt monitor just now:

```
Message: flt:2456,ont:2516,ran:-40,pan:20,hea:305,ggc:305,alt:0,asl:32,gsp:2,bpv:1450,acv:362,cad:4254,cud:0,rsi:78,gla:357609060,glo:1403789130,gsc:21,ghp:150,3df:3,hds:52,hdr:282,trp:0,fs:0,arm:1
```

stronnag, 2021-02-10:
I see, i should be 0 or 1; I missed that in the protocol spec :).

danarrib, 2021-02-10:
Yeah, I just noticed...  I was just editing the message. :-)

The problem is that it's a flag (0 or 1), and 3 is not a valid value for it.
I changed the logic. Instead of checking == 1, It'll check !=0.

I also noticed that you send all the fields, on every message. 
It's not a problem for the UI since usually it runs on a more capable internet, it just will consume more data on the network and on the MQTT Broker.

The on time and Flight time will be tracked by UI automatically if not sent by the aircraft. On Bullet GCSS I send this information only once per minute. The same with Home coordinates and Callsign.

You can also send only the fields that changed since the last message. This is the way that aircraft sends the data.

stronnag, 2021-02-10:
I know how the a/c sends the data, however for `fl2mqtt` 

* It's a simulation, for visualising blackbox and openTX logs
* The assumption is that you have a reasonable network connection (or use a local broker)
* I'm lazy, and a couple of 100 bytes / sec doesn't seem excessive.

stronnag, 2021-02-10:
addressed by 830acd2 (support and protocol auto detection) - dev branch for now.

closed 2021-02-10 by stronnag