stronnag/mwptools #90: inav radar snoop mode does not work

stronnag, 2021-05-03:
Due to either changes in the inav radar modus-operandi or a general misunderstanding, this no longer works.

I guess I'll have to fix it ...

stronnag, 2021-05-05:
@snaewe, this should be fixed by development commit 7290c9f  (see https://github.com/stronnag/mwptools/wiki/Accessing-development-and-other-branches if you don't have the dev branch).
```
git checkout development
make && sudo make install
```
Then:
```
mwp  --radar-device <device_name>
```
If you also have a FC with a GPS fix connected to mwp, then mwp should report the **home position**, otherwise it will report the centre of the map as the base station location.

If you build the simulator
```
cd mwptools/samples/radar_sim
make
```
then you can use a UDP connection to at least see something is happening:

Start mwp as:

```
mwp  --radar-device :3000 --no-poll
```
And the simulator:
```
./mwp-inav-radar-sim -d udp://localhost:3000
```
The simulator should report the startup messages from mwp, masquerading as aFC; and mwp should display four simulated radar contacts.

snaewe, 2021-05-05:
I'll test that soon and let you know.
Thanks for working on this!

snaewe, 2021-05-05:
Ok, when using mwp-inav-radar I see 4 'objects' flying around.
Now, how would a 'real' scenario be? What would I need to connect where?
I have:
- one inav FC with a inav-Radar module connected that has a GPS fix
- another inav-Radar module connected to my PC via a UART-TTL adapter (appears as /dev/USB0)
- the inav FC is connected to my PCas well  (/dev/ttyACM0) running mwp --radar-device /dev/USB0

Now, when I connect press 'connect' in MWP, I see all data from the FC as usual, but nothing else.
Radar View is empty.
What am I missing?

stronnag, 2021-05-05:
> Ok, when using mwp-inav-radar I see 4 'objects' flying around.
> Now, how would a 'real' scenario be? What would I need to connect where?
> I have:
>
> * one inav FC with a inav-Radar module connected that [...]

The radar device started automatically, you do not (should not) use Connect; I'll prevent that. You will need to set the baud rate as part of the device name, so for 115200 it would be:

```
mwp  --radar-device /dev/ttyUSB0@115200
```

I don't have the hardware, so can't advise more, maybe I can the simualor / mwp on either side of a BT / TTL device to try.

snaewe, 2021-05-05:
This is what I see on stdout:

2021-05-05T17:14:17+0200 Set up radar device /dev/ttyUSB1@115200
2021-05-05T17:14:17+0200 start radar reader /dev/ttyUSB1@115200
2021-05-05T17:14:17+0200 Host power: 74%, state: Discharging (2), warn: None (1)
2021-05-05T17:14:18+0200 Comm error count 1
2021-05-05T17:14:18+0200 expected header0 (ff)

After that, the radar-module's screen turns black! and it reboots

snaewe, 2021-05-05:
https://youtu.be/Uwh8q-gLHCU

stronnag, 2021-05-05:
So some of it is working; the 42 satellites in mwp's signature. So it's got as far as requesting and mwp providing it's GPS position. 
Let me have another read of the s̶p̶e̶c̶i̶f̶i̶c̶a̶t̶i̶o̶n̶   source code and see what might be happening. All mwp does is response to the devices's requests for data. I guess someone is not validating the their input .... bad bad.

stronnag, 2021-05-05:
otoh, I was lying about the size of one of the packets, which it would not like, and I'm not responding to another of MSPs I discovered in another file .....

Did I mention how much I dislike working around hardware I don't have and with no specification ....
 
Expect another commit soon.

stronnag, 2021-05-05:
044e429 might do it. No more lying about 2K byte message lengths, I guess it doesn't like that. 
See I even pen-test other people's software for free.

snaewe, 2021-05-05:
Better! But Lat/Lon doesn't get displayed correctly in the Radar View. Both almost 0 (0.00001). ID Altitude, Course (and apparently Speed, which *is* 0) are displayed correctly.

stronnag, 2021-05-05:
I guess there's two possibilities:
* mwp is sending a bad home position and inav-radar just bounces it back; or
* inav-radar decides mwp is not a FC and is doesn't want to play, sending back 0,0 -ish.

To see which is it:
* I've altered the way the radar reply decides whether the home position is set
* Enabled some debug to see what's happening.

If you run mwp as

```
mwp --debug-flags 16 --radar-device $RADAR_DEVICE
```
then it'll spew out raw and lat,lon values for the raw gps (rgps) sent to the `$RADAR_DEVICE` and (recv) the raw and lat/lon data from the device.

If it's still not getting valid lat/lon, please let me have a few samples of each (rgps, recv) from the latest dev commit (probably 5967db6)

```
2021-05-05T20:09:07+0100 RDR-rgps: Lat, Lon 50.910389 -1.534325
2021-05-05T20:09:07+0100 RDR-rgps: 02 2a 10 4f 58 1e 6c e1 15 ff 00 00 00 00 00 00 63 00

2021-05-05T20:09:07+0100 RDR-recv: Lat, Lon 50.910862 -1.533468
2021-05-05T20:09:07+0100 RDR-recv: 00 01 8e 61 58 1e ea 02 16 ff 21 13 00 00 35 00 06 07 04
```

stronnag, 2021-05-05:
Looks like a bug in the firmware  to me

main.cpp, line 938 et al.
```
  peers[i].gps_comp.lat = peers[i].gps.alt;
  peers[i].gps_comp.lon = peers[i].gps.alt;
  peers[i].gps_comp.alt = peers[i].gps.alt; 
```

Would explain the 0,0 (for a slow / stationary node).

Paging @OlivierC-FR ...

OlivierC-FR, 2021-05-05:
Good catch ! Never had time to fly this code version, you saved me an afternoon wondering why it would not compensate, while last week-end it compensated in kilometers instead of meters...

snaewe, 2021-05-06:
Yes! Looks good now. One more Q, though: Why is the status "Hidden/4" ?

snaewe, 2021-05-06:
Hmmm...well....
Now, when I quit mwp, the ESP32 module dies (black display, needs hard reset).
@OlivierC-FR

stronnag, 2021-05-06:
> Yes! Looks good now. One more Q, though: Why is the status "Hidden/4" ?

Because the radar node is returning a state of 2 (which I was once led to understand means "hidden") and the LQ is 4. I may misinterpreted state 2 or conflated it with "MAVLInk Trafifc Report" (ADS-B), which is handled by the same dialogue.

stronnag, 2021-05-06:
* mwp now advertises "GCS" for inav-radar MSP_FC_VARIANT
* Merged into master

Closing as complete; will revisit (including the use of the main serial device vice mandatory `--radar-device`) should inav-radar become less chatty.

closed 2021-05-06 by stronnag

stronnag, 2021-05-06:
For completeness, via 0fe9fcc (dev branch only as of 2021-05-06) it is possible to use the main msp port for inav-radar (vice using `--radar-device`). However, it is still necessary to add a command option to mwp; it needs to told to relax the default inbound msp direction check.

This is enabled as 
```
mwp --relaxed-msp
```
which should be 'mainly harmless' for normal operations. It's somewhat more acceptable to put this in `~/config/mwp/cmdopts` to make it the default.