stronnag/mwptools #92: BUG? pitch on AHI reversed?

wx4cb, 2021-06-21:
issue: Using mavlink either via crossfire or via a direct serial cable from the FC, roll is correct, but the pitch is reversed.

IE, the AHI shows a nose down attitude when the airframe is pitched up and vice versa.

```
uname -a: 
Linux toshtop 5.4.0-74-generic #83-Ubuntu SMP Sat May 8 02:35:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

charlesb@toshtop:~$ mwp --version
4.168.682
```

Not a big deal but as someone who is used to pitch up being in the blue :D 

(side note: love the new way to install :dancers: - so much better)

[logs.zip](https://github.com/stronnag/mwptools/files/6689512/logs.zip)

stronnag, 2021-06-21:
possibly a bug in the mavlink decoder; iirc LTM is correct. I'll look at it, thanks.

wx4cb, 2021-06-21:
yea, i just loaded a bblog from my flight this morning of the FC and it's correct

stronnag, 2021-06-22:
Should be fixed in c47eb75, please check and close if so.

wx4cb, 2021-06-25:
yup.. works fine... sorry for the delay.. bad week....

closed 2021-06-25 by wx4cb