stronnag/mwptools #93: Feature Request: Add current and mAh to battery monitor

wx4cb, 2021-06-21:
is it possible to add the current draw and possibly the amount consumed to the battery monitor?

wx4cb, 2021-06-21:
could it be that i have the window too small maybe? I thought it was in there, just didnt see it... the two logs in the other "issue" are both on inav 3 and i did notice there was a few crc fails, but that could also be the crossfire :D 

never thought to do a log when it was directly connected

wx4cb, 2021-06-21:
dammit wrong button

wx4cb, 2021-06-21:
yea i see it in the bblog i'm playing

stronnag, 2021-06-22:
Thanks for checking. Neither issue is surprising; Mavlink has not been used much since LTM was introduced --- until CRSF came along.

stronnag, 2021-06-22:
Should be fixed in c47eb75, please check and close if so.

wx4cb, 2021-06-25:
yup works great....

closed 2021-06-25 by wx4cb