stronnag/mwptools #103: (nothing urgent) WSL-G moving windows issue

wx4cb, 2021-10-30:
using the latest deb from the releases page, under Win 11 and WSL-G (btw not a fan of windows but this helps a lot with my work too :D) when you try and move the telemetry data windows on the right hand side, the blue bit where you are moving the window to is on the opposite side of the screen.

This is most likely a WSL issue obviously, but just thought it should be noted.

Great work again johnathan as always.

[image]

stronnag, 2021-10-30:
> using the latest deb from the releases page, under Win 11 and WSL-G (btw not a fan of windows but this helps a lot with my work too :D) 

Having a `.deb` is another nice (but unexpected) side-effect of the `meson + ninja` migration. I think these will become a regular (monthly) event. 

> when you try and move the telemetry data windows on the right hand side, the blue bit where you are moving the window to is on the opposite side of the screen.
> 
> This is most likely a WSL issue obviously, but just thought it should be noted.

Not specifically WSL. It's a Wayland (the new Linux graphics API) issue. When using X Window the blue indicator moves correctly;  with Wayland it does not. You'll see the same thing on native Linux with Wayland. WSLG defaults to  emulating Wayland. It's either a low level API issue or a problem in `gdl` (gnome dock library). Neither of which I can directly influence. 

On native Linux (or a VM), you can force correct behaviour by forcing XWayland:
```
$ GDK_BACKEND=x11 mwp
```

To my immense surprise, this works in WSLG as well. So if you want to move the dock widgets around you can force x11 mode (which will be less slower) and otherwise let it use Wayland.

> Great work again johnathan as always.
> 
I didn't do anything much (other than `ser2udp`); Microsoft did the hard work. 

Closing, as I can't do anything about it, and there is a workaround. Thanks for mentioning it, at least it's now documented.

closed 2021-10-30 by stronnag

wx4cb, 2021-10-30:
@stronnag  of course, but still.... that's perfectly fine. so far Win11 isnt too bad but i've only just finished installing it and the first thing i installed was chrome, the 2nd MWPtools lol