stronnag/mwptools #104: Desktop Icon (win 11 WSL-G)

wx4cb, 2021-11-05:
question.. i created a shortcut using wslusc, but i thought there was an icon (.ICO) file somewhere but i can't find it in the source tree?

does one actually already exist? if not i'll create one

```
 wslusc  /usr/local/bin/mwp 
```

stronnag, 2021-11-05:
`src/common/mwp_icon.ico`

Stupid location, I'll move it sometime later.

stronnag, 2021-11-05:
See also Marc's (@b14ckyy) BAT file in the [wiki](https://github.com/stronnag/mwptools/wiki/mwp-in-Windows-11---WSL-G#launch-ser2udp-and-mwp-in-one-go) which provides a laucher (and starts `ser2udp`).

wx4cb, 2021-11-05:
lol makes sense (icon location) lol....

As for the shortcut, i use the same "startmwp" bash file that I use on my normal linux machine (it's just an mwp -N -r -l default) script file, but if you use:

```
wslusc -i <icon path> <command> 
```

it creates the shortcut on the desktop for you :D

stronnag, 2021-11-11:
1. The Windows icon is now in the `docs ` directory
2. The WSLG page on the wiki has its own copy for download

closed 2021-11-11 by stronnag