stronnag/mwptools #105: [Feature Request] mwp should handle CRSF telemetry

stronnag, 2021-11-13:
From Marc Dornan via https://www.rcgroups.com/forums/showpost.php?p=48150643&postcount=519

WIP. Currently I've reverse engineered enough to get IDs from CRSF frames.

stronnag, 2021-11-13:
Gradually decoding the fields from some random raw data logs off the internet. What would really help would be a validated (i.e. with a Blackbox log) raw data capture, from a real flight, preferably with different flight modes (ACRO, RTH etc.)

* A raw data capture of the CRSF telemetry stream;
* A blackbox log of the same event

Failing the BBL, some indication of characteristics (location, altitude ranges, speed ranges, flight modes exercised) would suffice.

The raw capture can be just a captured stream, or one with context from e.g. [mwp-serial-cap](https://github.com/stronnag/mwptools/blob/master/src/samples/mwp-serial-cap/README.md)

Should any volunteer need [mwp-serial-cap](https://github.com/stronnag/mwptools/blob/master/src/samples/mwp-serial-cap/README.md) building for any reasonable platform (Linux, MacOS, Windows, FreeBSD), please ask.

wx4cb, 2021-11-23:
gotcha covered, my setup is TX16S with full module, and an hc05 attached to the right uart using telemetry mirror.

do you want the telemetry mirror or the mavemu from the module (can also do the wifi too if you want).

hoepfully if the wind isn't too bad in the morning will do it then, if not probably thanksgiving

stronnag, 2021-11-23:
The telemetry mirror please. The transport (BT, wifi) doesn't matter; it's the CRSF format telemetry that's of interest.

wx4cb, 2021-11-23:
ok will do

wx4cb, 2021-11-23:
[crsfidletest.zip](https://github.com/stronnag/mwptools/files/7591218/crsfidletest.zip)

here's a quick idle test cap just to make sure it's capturing data... It was just a power on the plane and radio wait for sats and then arm and run the the modes and disarm

Is this the right data? if so i'll go try and work on it this week, now that i have an SD card for the BB

stronnag, 2021-11-23:
Looks good!

wx4cb, 2021-11-24:
[killerwhale-cap-11.24.21.zip](https://github.com/stronnag/mwptools/files/7597562/killerwhale-cap-11.24.21.zip)

hopefully this helps.... did acro, cruise, rth, a little too windy today so dont worry too much about the route.

the killerwhale1.cap should be the crsf, the other one is via bluetooth to the module and mavemu

enjoy

wx4cb, 2021-11-24:
ill sort out a BBL when i get home

stronnag, 2021-11-24:
Thank you very much.

wx4cb, 2021-11-24:
no problem, i have a feeling the crossfire module bluetooth (not the hc05 telemetry mirror) dropped out as it originally connected and i was getting data, but when i landed it was showing "connect" not connected if that makes sense, but everything from today should be there just in case.

wx4cb, 2021-11-24:
ill do another one at home with my quad that i can hover around the house when i get home

stronnag, 2021-11-24:
killerwhale.cap is perfect. All I need.
Have to sort out range / bearing (which I need to calculate) and see why I'm missing altitiude. Minor details, looks great.
[image]

wx4cb, 2021-11-24:
Yea.. winds were terrible.. have been all week...

If i can go out the next coiple days i will get more data for you. As i
say, ill get u the bbl when i get home

wx4cb, 2021-11-24:
[LOG00033.TXT](https://github.com/stronnag/mwptools/files/7598164/LOG00033.TXT)

 LOG33 should be the one corresponding to this flight

stronnag, 2021-11-24:
Thanks. It is. The results from the CRSF recording and the BBL are pretty much identical. 
Already pushed to development.

wx4cb, 2021-11-26:
@stronnag here's some more data from the bench. it seemed to recognise it ok no problem

My modes are manual/horizon-angle/cruise/(either wp or poshold)/acro/rth
[crsf-bench.zip](https://github.com/stronnag/mwptools/files/7610299/crsf-bench.zip)

wx4cb, 2021-11-30:
@stronnag not sure how happy you're gong to be, but i made 2 flights today, the first one with the grand tundra and the second one with the killer whale.

the KW has an inav radar onboard the tundra doesn't as of yet. but i had mwp running with the CRSF telemetry from the TX16s mirror.

I did some mode changes, some tuning (well tried :rofl: ) but hopefully this gives you a ton of info.

the device rfcomm0 is the telemetry mirror, the pingRX is obvious :D

[MWP-11.30.21.zip](https://github.com/stronnag/mwptools/files/7628753/MWP-11.30.21.zip)

stronnag, 2021-11-30:
Other than the MAVCRC stuff, anything I should be especially unhappy about?
Two radar devices, camera pushing the bleeding edge here.

wx4cb, 2021-11-30:
didnt get chance to look at the samera side of things as I couldn't see shit in the bright florida sun :rofl:

the MAVCRC is coming from the pingRX I beleive, but other than that... no it worked fine.

wx4cb, 2021-11-30:
maybe you can do it so that you have a uri type of scheme, such as:

MSP://xxxxx@yyyyy
MAV://xxxxx@yyyyy

wx4cb, 2021-11-30:
and then if that's not there just auto detect ?

stronnag, 2021-11-30:
Alas no, testing with `mwp --radar-device udp://:40001`.  I could add another hint parameter, however I think that's not necessary yet.

The  Mav CRC errors are on two messages I don't need `MAVLINK_MSG_ID_DATA_REQUEST` and `MAVLINK_MSG_ID_OWNSHIP`. Maybe I'm not setting the CRC seed to the correct value. Anyway, it's not affecting the actual ADS-B info, which is always parsed correctly. Just ugly in the logs.

btw have you updated the firmware on the PingRX; don't recall ever seeing these messages before.

wx4cb, 2021-11-30:
@stronnag no i havent updated anything on the pingRX i don't even think that's possible. as for the udp stuff, i haven't had chance to do that. i'm trying to tune these 3 planes before i head off to the cold white north for christmas with the missus....

I can try that with the crossfire wifi later this week see what happens.

stronnag, 2021-11-30:
When I run `mwp._dev_ttyPINGRX_57600.2021-11-30T081731.raw` though my mavlink conformity tester, I get:
```
Mav1: mavlink_traffic_report 38 : ICAO 11351520, callsign AAL2245
V1 OK 30485, fail 1305
```
So even though it's a lot of noise in the log, it's 0.4% of the total messages. Hardly a big fail.

wx4cb, 2021-11-30:
yea it's annoying, but not a consequence i don't think to be honest. maybe just silently drop them?

stronnag, 2021-11-30:
I'll maybe wrap them up under a `DEBUG_FLAG`

stronnag, 2021-11-30:
So the CRC seeds for 202, 203 where not in mwp's mavlink crc seed table (they are PingRX specific messages). 
Fixing that gives:
```
V1 OK 31787, fail 3
```
0.009% error rate; I think we live with that.

wx4cb, 2021-11-30:
oh i dunno about that..... i mean.... we ARE aiming for multi-wii perfection right :rofl:

stronnag, 2021-12-05:
closing, committed to master

closed 2021-12-05 by stronnag

marianomd, 2023-04-11:
Hello!! Is there a way to open a binary file with the raw CRSF telemetry dump directly from mwp? I tried to use the file path in the "Device" textbox, and in fact it seems it decoded part of it, a siren sounded, but the application froze.

This is the log:

```
2023-04-10T23:00:02-0300 mwp
2023-04-10T23:00:02-0300 buildinfo: 2023-03-05T11:17:11Z Linux 6.2.2-zen1-1-zen x86_64 buster
2023-04-10T23:00:02-0300 toolinfo: gcc version 8.3.0 (Debian 8.3.0-6)  / GNU ld (GNU Binutils for Debian) 2.31.1 / Vala 0.42.5
2023-04-10T23:00:02-0300 version: mwp 4860c967 () / 2023-03-05T11:15:38+00:00 6.064.403
2023-04-10T23:00:02-0300 "Ubuntu 20.04.2 LTS" on Mariano-F15 running Linux 5.15.90.1-microsoft-standard-WSL2 x86_64
2023-04-10T23:00:02-0300 WM: Wayland / Unknown DE
2023-04-10T23:00:02-0300 GL: Mesa/X.org / llvmpipe (LLVM 11.0.0, 256 bits)
2023-04-10T23:00:02-0300 hypervisor: wsl
2023-04-10T23:00:02-0300 GLib2 2.64
2023-04-10T23:00:02-0300 MQTT enabled via the "mosquitto" library
2023-04-10T23:00:02-0300 Failed to find "mavlogdump.py" on PATH
2023-04-10T23:00:02-0300 "blackbox_decode" too old, replay disabled
2023-04-10T23:00:02-0300 Using fl2ltm 1.0.9 commit:6d7ec1f
2023-04-10T23:00:02-0300 Using speech api 1 [espeak]
Could not connect: No such file or directory
2023-04-10T23:00:02-0300 Get map preference
2023-04-10T23:00:04-0300 Starting Bing proxy
2023-04-10T23:00:04-0300 BB load async map zoom : true
2023-04-10T23:00:04-0300 Loaded /usr/share/mwp/vcols.css
2023-04-10T23:00:04-0300 Fixed font true
2023-04-10T23:00:04-0300 gsettings blank default-map, using osm-mapnik
2023-04-10T23:00:04-0300 Delta speed 0.000000
2023-04-10T23:00:04-0300 Show main window

(org.stronnag.mwp:2184): Gdk-WARNING **: 23:00:04.586: Native Windows taller than 65535 pixels are not supported
2023-04-10T23:00:04-0300 UPower: Could not connect: No such file or directory
2023-04-10T23:04:23-0300 Try connect /mnt/c/Users/USER/Desktop/FPV/crsf.log
2023-04-10T23:04:23-0300 Connected /mnt/c/Users/USER/Desktop/FPV/crsf.log true
2023-04-10T23:04:23-0300 Serial ready
2023-04-10T23:04:23-0300 New CRSF Mode Manual (0) 0 -1s 0.000000 0.000000 0 0
2023-04-10T23:04:23-0300 init icon 0 0 0 (true)
2023-04-10T23:04:23-0300 Craft is armed, special=0
2023-04-10T23:04:23-0300 Enable nav speak
2023-04-10T23:04:23-0300 Initialised "none" for speech
2023-04-10T23:04:23-0300 New CRSF Mode Acro (1) 1 0s 0.000000 0.000000 0 0
2023-04-10T23:04:23-0300 null speech say 9 satellites
2023-04-10T23:04:23-0300 null speech say Heading 174
2023-04-10T23:04:23-0300 null speech say Voltage 17
2023-04-10T23:04:23-0300 Set home -34.657067 -61.039130 (CRSF true)
2023-04-10T23:04:23-0300 New CRSF Mode Angle (2) 1 0s -34.657111 -61.039146 0 0
2023-04-10T23:04:23-0300 New CRSF Mode GPS Hold (9) 1 0s -34.657127 -61.039139 0 2
2023-04-10T23:04:23-0300 Set poshold -34.657127 -61.039139
2023-04-10T23:04:23-0300 New CRSF Mode Angle (2) 1 0s -34.657143 -61.039136 0 0
2023-04-10T23:04:23-0300 Comms error CRSF/CRC 1
2023-04-10T23:04:23-0300 New CRSF Mode GPS Hold (9) 1 0s -34.657407 -61.039046 0 2
2023-04-10T23:04:23-0300 Set poshold -34.657407 -61.039046
2023-04-10T23:04:23-0300 New CRSF Mode Acro (1) 1 0s -34.659797 -61.036768 0 0
2023-04-10T23:04:23-0300 Comms error CRSF/CRC 2
2023-04-10T23:04:23-0300 Comms error CRSF/CRC 3
2023-04-10T23:04:23-0300 New CRSF Mode Return Home (13) 1 0s -34.657511 -61.038911 0 4
2023-04-10T23:04:23-0300 New CRSF Mode Acro (1) 1 0s -34.657361 -61.039047 0 0
2023-04-10T23:04:23-0300 Disarmed CRSF
2023-04-10T23:04:23-0300 New CRSF Mode Manual (0) 0 -1s 0.000000 0.000000 0 0
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 4
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 5
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 6
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 7
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 8
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 9
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 10
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 11
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 12
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 13
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 14
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 15
2023-04-10T23:04:24-0300 Craft is armed, special=0
2023-04-10T23:04:24-0300 Enable nav speak
2023-04-10T23:04:24-0300 New CRSF Mode Acro (1) 1 0s 0.000000 0.000000 0 0
2023-04-10T23:04:24-0300 null speech say 8 satellites
2023-04-10T23:04:24-0300 null speech say Heading 175
2023-04-10T23:04:24-0300 null speech say Range 13, bearing 202
2023-04-10T23:04:24-0300 null speech say Voltage 16.4
2023-04-10T23:04:24-0300 Set home -34.657109 -61.038927 (CRSF true)
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 16
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 17
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 18
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 19
2023-04-10T23:04:24-0300 Disarmed CRSF
2023-04-10T23:04:24-0300 New CRSF Mode Manual (0) 0 -1s 0.000000 0.000000 0 0
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 20
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 21
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 22
2023-04-10T23:04:24-0300 Comms error CRSF/CRC 23
2023-04-10T23:04:28-0300 message => No data for 5s
```

Thanks in advance.

stronnag, 2023-04-11:
You can use [mwp-log-replay](https://github.com/stronnag/mwptools/blob/master/src/samples/mwp-log-replay/README.md) to replay the log to a "serial" device. The easiest way is to use a UDP port; for example:
```
# Start mwp as a UDP listener, port is arbitrary, here 40001 is chosen
## -a connect immediately without user intervention
## -d serial-device. No host part means it listens for remote connections
## listen on UDP port 40001

mwp  -a -d udp://:40001

# In another  terminal  (even other machine if you replace localhost with the machine running mwp)

mwp-log-replay -d udp://localhost:40001 /path/to/my/logfile.raw
``` 

I should at least document this (and maybe automate it .....)

stronnag, 2023-04-11:
See #143