stronnag/mwptools #106: Multiple "radar" devices

wx4cb, 2021-11-23:
is it possible to have more than one "--radar-device" int he command line options?

It doesn't specifgically say it in the wiki that I could see....

basically i wanna do --radar-device=/dev/inav-radar@115200 --radar-device=/dev/pingrx@115200 

👯‍♂️

stronnag, 2021-11-23:
Not possible now, I will make it happen .....

stronnag, 2021-11-23:
OK, so that was an easy, 15 minute job. There is now a `multi-radar` branch. You can specify multiple `,` separated device names. Testing with the simulators and multiple udp sockets.
```
mwp --radar-device=udp://:3000,udp://:3001
```

Then:
```
mwp-inav-radar-sim -d udp://localhost:3000
# and
mwp-mavlink-traffic-sim -d udp://localhost:3001
```
Result:

[image]

stronnag, 2021-11-23:
Note I'm waiting on a CRSF data capture to merge  #105 then it'll be a big master merge up.

wx4cb, 2021-11-23:
I can catch that for you if you don't have a crossfire module let me know how you want me to capture it and I can do that this afternoon

stronnag, 2021-11-23:
> I can catch that for you if you don't have a crossfire module let me know how you want me to capture it and I can do that this afternoon

What would be ideal would be a capture of the raw BT telemetry output, preferably with `mwp-serial-cap`. For a non-trivial flight (i.e. some nav modes in the mix). If there was a matching BBL it would be perfect. https://github.com/stronnag/mwptools/issues/105#issuecomment-968079478

No problem is it's not possible.

wx4cb, 2021-11-25:
so doing more work on this (multi-radars branch), bear in mind this is on the bench havent tested in flight yet

i'm getting a ton of mavcrc fails from the reader on the pingRX (ttyUSB), other thing i noticed is that if you have the LORA radar module running and a new node joins it doesn't seem to get the new node in the list until you restart the lora module. not sure if that was because it didnt have a GPS value, but it immediately popped in when the module reset even with a 0/0 coordinate.

[image]

radar_adsb1.* is the first test
the mwp_ is the second test with multiple lora modules.

[radar-data.zip](https://github.com/stronnag/mwptools/files/7604630/radar-data.zip)

wx4cb, 2021-11-25:
what is the altitude stored as in the RadarPlot struct? 

looking at slightly changing the list view to only show things up to a certain altitude... no point cluttering the list with things at 40k feet when we're only at 400ft lol

I thought it may have been internally stored as meters but i'm still showing things > FL330 when it's set to 10k meters (FL 328)

stronnag, 2021-11-25:
The altitude is normalised as metres (inav-radar reports cm, pingRX reports mm!). The pingRX reports will be AMSL, which is not too much of an issue in Florida (as you'll soon all be BMSL).

stronnag, 2021-11-25:
And you've kindly discovered a new bug. Each device channel has its own raw log (good). However they were time-stamped with 1 second resolution, so with multiple radar devices, the second would clobber the first.
Fixed in development.

wx4cb, 2021-11-25:
well at least i'm useful for some things lol

wx4cb, 2021-11-25:
@stronnag oko so the wrapping in the IF statement works for the list, still shows all in the map (for now), what would be the best place to put a config variable that can set the "max altitude" preferences i would guess?

stronnag, 2021-11-25:
The preferences. You need to change things in two places:
* `src/mwp/setttings.vala` : There's a structure that holds all the configuration data. Add a new `uint` to the structure. Then in the code that follows, for each one there's a stanza that accesses  it. Search for `max_wps` and copy that. Note that in the settings text name, `_` becomes `-`.
* `src/mwp/org.mwptools.planner.gschema.xml` defines the settings schema. There is a section here that defines how the setting appears. Search for `max-wps` and duplicate that.

wx4cb, 2021-11-25:
ok and then in the radar_list.vala i can just do settings.max_radar_altitude to reference it ?

stronnag, 2021-11-25:
Now you need to pass the settings to the `update()` function. It already has a configuration value (`dms`). So you'd need to pass the new parameter as well, so the signature might become 
```
    public void update (RadarPlot r, bool dms, uint max_radar_alt)
```
and the caller:
```
 radarv.update(r, conf.dms, conf.max_radar_alt);
```
where `max_radar_alt`  `"max-radar-alt"` is the new setting.

stronnag, 2021-11-25:
Or what ever you choose to call it.

wx4cb, 2021-11-25:
if i want to bring in MWPLog and DEBUG_FLAGS, what do i need to "include" ?

stronnag, 2021-11-25:
It knows about MWPLog. Just use it.
You'd have to pass the debug_flags variable to update() as well and use it like:
```
        if((debug_flags & DEBUG_FLAGS.RADAR) != DEBUG_FLAGS.NONE)
            MWPLog.message(sb.str);
```
OTOH, there's probably no need to do this in the `update()` function. 
As you don't want to do anything in `update()` do you?, then in `mwp.vala` at the end of ` process_mavlink_radar()`, can't you do filter there without changing the `update()` signature. Something like:

```
/** not even compile tested **/
if (r.altitude < conf.max_radar_alt) {
 radarv.update(r, conf.dms)
} else if((debug_flags & DEBUG_FLAGS.RADAR) != DEBUG_FLAGS.NONE) {
       MWPLog.message("Not listing %s at %dm\n", callsign, r.altitude);
}
```

wx4cb, 2021-11-25:
that works too, although i was trying to only do the code once as there's a number of "radarv.update" lines in there and that requires dupe code for each..

however that said, see what you thing and i can PR this havent quite figured out whether it's the MWPLog %d conversion, but the alt in the log message is always zero (unless i typod something):

https://github.com/wx4cb/mwptools/commit/ac6167026601bbc75da162c2cb726ba0d04e3372

stronnag, 2021-11-25:
> that works too, although i was trying to only do the code once as there's a number of "radarv.update" lines in there and that requires dupe code for each..

* However, there's only one place you need the debug, the others are either in inav-radar part (or the stale / hidden part). Just having the test in these places without the debug is cheap and goo enough. Having a single debug in the mavlink-radar parsing will demonstrate what's happening.
* There's probably no need for a separate debug category. 
* r.altitude is a double, change the format specifier to `"%.1f"`.

wx4cb, 2021-11-25:
that worked... i added the debug cat to just not have the extra "radar" stuff, removed that now.

added another commit (and then another one because i forgot to build in the right directory lol, i don't know how to merge those if at all possible)

https://github.com/wx4cb/mwptools/commit/f75577c01b759bfea3c8e599502cf6d0a4a247a0

wx4cb, 2021-11-30:
@stronnag  this seemed to work good, i think it is good to go, but i'll let you make the final determination after looking at the logs from in the CRSF thread.

stronnag, 2021-12-01:
Next up, proximity alerts.
* If there is a live home, use that; else
* if there is a planning home, use that (as example); else
* Don't show range.
[image]
* Auto switch between live / planned home as necessary.

wx4cb, 2021-12-01:
Haha. Great minds think alike. I was already thinking of this, but this week is a cluster for work so have been preoccupied

wx4cb, 2021-12-02:
looks like someone has been having too much fun lol

wx4cb, 2021-12-02:
btw, i was going to add to the radar wiki page about the schema.xml setting for the max altitude for the list today, but i see you been editing it for the other stuff... do you still want me to do that ?

https://github.com/stronnag/mwptools/wiki/mwp-Radar-View

stronnag, 2021-12-02:
No thanks, I have a script somewhere that generates that page; it's on my todo list prior to weekend master merge.

wx4cb, 2021-12-03:
something borked in the latest dev release.... it doesn't handle multiple --radar-devices now... at least in the /dev/ttyUSB range :D

if i run --radar-device=/dev/ttyUSB1@115200 --radar-device=/dev/ttyUSB0@57600, it only ever pickes up the USB0 device.

If i run them singularly it'll pick it up no problem

```
charlesb@ToshTop:~$ mwptools
2021-12-03T18:12:05-0500 mwp startup version: mwp fd40163 / 2021-12-03T21:08:51+00:00 4.337.760 on Xlib
2021-12-03T18:12:05-0500 "Ubuntu 21.04" on ToshTop running Linux 5.11.0-40-generic x86_64
2021-12-03T18:12:05-0500 hypervisor: none
2021-12-03T18:12:05-0500 Failed to find "blackbox_decode" on PATH
2021-12-03T18:12:05-0500 Failed to find "fl2ltm" on PATH
2021-12-03T18:12:05-0500 Failed to find "mavlogdump.py" on PATH
2021-12-03T18:12:05-0500 Using speech api 1 [espeak]
2021-12-03T18:12:06-0500 Starting Bing proxy 
2021-12-03T18:12:06-0500 BB load async map zoom : true
2021-12-03T18:12:06-0500 Loaded /usr/local/share/mwp/vcols.css
2021-12-03T18:12:06-0500 Fixed font true
2021-12-03T18:12:06-0500 Delta speed 0.000000
2021-12-03T18:12:06-0500 Set up radar device /dev/ttyUSB0@57600
2021-12-03T18:12:06-0500 start radar reader /dev/ttyUSB0@57600
2021-12-03T18:12:06-0500 raw log for /dev/ttyUSB0@57600 mwp._dev_ttyUSB0_57600.2021-12-03T181206.raw
2021-12-03T18:12:06-0500 Registered USB device: /dev/ttyUSB0 [0403:6001], Vendor: FTDI, Model: USB__-__Serial, Serial: (null), Driver: ftdi_sio
2021-12-03T18:12:06-0500 Registered USB device: /dev/ttyUSB1 [1a86:7523], Vendor: 1a86, Model: USB2.0-Serial, Serial: (null), Driver: ch341
2021-12-03T18:12:06-0500 Host power: 100%, state: Fully Charged (4), warn: None (1)
2021-12-03T18:12:06-0500 Comms error Detect 1
2021-12-03T18:12:06-0500 GST: "add" <TOSHIBA Web Camera - HD: TOSHIB> </dev/video0>
^C
charlesb@ToshTop:~$ !vi
vi .local/bin/mwptools 
charlesb@ToshTop:~$ mwp -N -4 -r --layout default -d /dev/rfcomm1@115200 --radar-device=/dev/ttyUSB1@115200
2021-12-03T18:13:07-0500 mwp startup version: mwp fd40163 / 2021-12-03T21:08:51+00:00 4.337.760 on Xlib
2021-12-03T18:13:07-0500 "Ubuntu 21.04" on ToshTop running Linux 5.11.0-40-generic x86_64
2021-12-03T18:13:07-0500 hypervisor: none
2021-12-03T18:13:07-0500 Failed to find "blackbox_decode" on PATH
2021-12-03T18:13:07-0500 Failed to find "fl2ltm" on PATH
2021-12-03T18:13:07-0500 Failed to find "mavlogdump.py" on PATH
2021-12-03T18:13:07-0500 Using speech api 1 [espeak]
2021-12-03T18:13:08-0500 Starting Bing proxy 
2021-12-03T18:13:08-0500 BB load async map zoom : true
2021-12-03T18:13:08-0500 Loaded /usr/local/share/mwp/vcols.css
2021-12-03T18:13:08-0500 Fixed font true
2021-12-03T18:13:08-0500 Delta speed 0.000000
2021-12-03T18:13:08-0500 Set up radar device /dev/ttyUSB1@115200
2021-12-03T18:13:08-0500 start radar reader /dev/ttyUSB1@115200
2021-12-03T18:13:08-0500 raw log for /dev/ttyUSB1@115200 mwp._dev_ttyUSB1_115200.2021-12-03T181308.raw
2021-12-03T18:13:08-0500 Registered USB device: /dev/ttyUSB0 [0403:6001], Vendor: FTDI, Model: USB__-__Serial, Serial: (null), Driver: ftdi_sio
2021-12-03T18:13:08-0500 Registered USB device: /dev/ttyUSB1 [1a86:7523], Vendor: 1a86, Model: USB2.0-Serial, Serial: (null), Driver: ch341
2021-12-03T18:13:08-0500 Host power: 100%, state: Fully Charged (4), warn: None (1)
2021-12-03T18:13:08-0500 Comms error Detect 1
2021-12-03T18:13:08-0500 GST: "add" <TOSHIBA Web Camera - HD: TOSHIB> </dev/video0>
```

stronnag, 2021-12-04:
> something borked in the latest dev release.... it doesn't handle multiple --radar-devices now... at least in the /dev/ttyUSB range :D
> 
> if i run --radar-device=/dev/ttyUSB1@115200 --radar-device=/dev/ttyUSB0@57600, it only ever pickes up the USB0 device.

Nothing has changed. You probably meant 
...
--radar-device=/dev/ttyUSB1@115200,/dev/ttyUSB0@57600
...
i.e. the documented comma separated syntax.

But I'll support multiple invocations of the `--radar-device` option as well.

stronnag, 2021-12-04:
Thusly:
```
$ mwp --radar-device=udp://:30001 --radar-device=udp://:30002,udp://:30003
2021-12-04T09:24:59+0000 mwp startup version: mwp local / 0000-00-00T00:00:00 4.337.804 on Wayland
2021-12-04T09:24:59+0000 "Arch Linux" on eeyore running Linux 5.15.6-arch2-1 x86_64
2021-12-04T09:24:59+0000 hypervisor: none
2021-12-04T09:24:59+0000 MQTT enabled via the "paho" library
2021-12-04T09:24:59+0000 Using speech api 3 [flite]
2021-12-04T09:25:00+0000 Starting Bing proxy 
2021-12-04T09:25:00+0000 External proxy "gmproxy" listening on :38983
2021-12-04T09:25:00+0000 External proxy "bproxy" listening on :46355
2021-12-04T09:25:00+0000 BB load async map zoom : true
2021-12-04T09:25:00+0000 Loaded /home/jrh/.config/mwp/vcols.css
2021-12-04T09:25:00+0000 Fixed font true
2021-12-04T09:25:00+0000 Delta speed 2.000000
2021-12-04T09:25:00+0000 Set up radar device udp://:30001
2021-12-04T09:25:00+0000 start radar reader udp://:30001
2021-12-04T09:25:00+0000 Set up radar device udp://:30002
2021-12-04T09:25:00+0000 Set up radar device udp://:30003
2021-12-04T09:25:00+0000 start radar reader udp://:30002
2021-12-04T09:25:00+0000 start radar reader udp://:30003
2021-12-04T09:25:00+0000 Host power: 0%, state: Unknown (0), warn: None (1)
2021-12-04T09:25:00+0000 mwp will manage power and screen saver / idle
```

```

wx4cb, 2021-12-04:
you know that was weird, i never ran the comma separated line, it was always multiple options LOL.. i'll give that a try :D

stronnag, 2021-12-04:
Sorry, but It could never have worked with multiple options (vice `,`). I guess I'd better add the provided command line to the log ...

stronnag, 2021-12-05:
Closing, committed to master

closed 2021-12-05 by stronnag

wx4cb, 2021-12-05:
@stronnag just noticed something i never thought of...... if the device doesn't exist on first opening, then we need a way to "ignore it" otherwise it keeps coming up in the logs as "no such file or device" (ENOENT).

stronnag, 2021-12-05:
Intended / documented behaviour. If the device does not exist, mwp retries it.

stronnag, 2021-12-05:
It would be more annoying (IMO); if it failed to connect due to operator error and then you had to restart mwp once you'd physically fixed the connection.

wx4cb, 2022-01-11:
how easy would it be to incorporate SBS messages ? this would be helful because then you can use anything online that has an SBS format output so that you don't actually need to have an ADSB receiver. You can even use dump1090 to pull it in from any ADSB receiver or even an SDR

http://woodair.net/sbs/article/barebones42_socket_data.htm

(will attach a data dump when i have it)