stronnag/mwptools #107: FR: Set home position of MWP outside of planes arming location

wx4cb, 2021-11-23:
currently the arming location of the airframe is used as the home location for the GCS.

(bear in mind i could be completely misinterpreting how it works)
Assuming it doesn't already exist and im blind, Is it possible to add a "right click -> set home" feature similar to mission planner so that if you're using a  lora module for inav radar directly connected to mwp via a serial port that the "location" for that particular lora node is sent to the other lora nodes so that the home point shows up on the OSD and also on the map ?

wx4cb, 2021-11-23:
bear in mind, i may be misinterpreting what hooking a lora module up to the gcs actually does, but in my limited testing with an airframe with a GPS lock, i'm getting nothing in the radar view

stronnag, 2021-11-23:
It would be possible to have the location on the map; however I'm pretty sure that it is a **feature** of inav-radar nodes are entirely passive and can't broadcast their position to other nodes. I had a long discussion with Olivier on this.

When I add the Radar home to the mwp, then you could also have proximity alerts, at least for ADS-B tracks.

wx4cb, 2021-11-23:
oh i see, even though the plane is sitting right next to me i should "see" the planes in the radar view table. i have data as i'm getting it from the crossfire module... i know it's kinda redundent, but i'm just trying to figure out why it's not showing anything... probably something i missed.

stronnag, 2021-11-23:
And you do have recent firmware on the radar modules; at least from 2021-05-06?
I'll have a think about how we can debug this (I don't any radar modules).

wx4cb, 2021-11-23:
it's 2.2 something from memory, doyou want a couple of radar modules? I can send you a couple if you want  ?

stronnag, 2021-11-24:
> it's 2.2 something from memory, doyou want a couple of radar modules? I can send you a couple if you want ?

Should be recent enough. Thanks but I'm trying really hard not to increase the amount of "stuff" I collect.

I'll up the debug in the radar section and see what's happening. In theory:
* The inav-radar polls us for some config information
* mwp admits to being a ground station.
* The inav-radar sends unsolicited location data for the other nodes.

The existing code will output the received locations when invoked with `--debug-flags 16`.
There is now `rdr-debug` branch that outputs information for the message exchanges. In theory,  you should see in the console / status log stuff like:

```
2021-11-24T19:34:04+0000 RDR-msg: MSP_CMDS_NAME
2021-11-24T19:34:04+0000 RDR-msg: MSP_CMDS_FC_VARIANT
2021-11-24T19:34:04+0000 RDR-msg: MSP_CMDS_FC_VERSION
2021-11-24T19:34:04+0000 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2021-11-24T19:34:04+0000 RDR-recv 0: Lat, Lon 54.138514 -4.702034
2021-11-24T19:34:04+0000 RDR-recv: 00 01 b0 e1 44 20 c7 86 32 fd 9f 13 00 00 2c 00 87 05 01
2021-11-24T19:34:04+0000 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2021-11-24T19:34:04+0000 RDR-recv 1: Lat, Lon 54.138415 -4.702023
2021-11-24T19:34:04+0000 RDR-recv: 01 01 d9 dd 44 20 3d 87 32 fd 9c 13 00 00 87 00 5f 06 01
```
The first 3 are the radar polling mwp for information, the `RDR-recv N:` tags is the locations provided for beacon 'N'.

i.e. the rdr-debug branch, something like:
```
mwp --radar-device /dev/myradar@115200 --debug-flags 16
```

wx4cb, 2021-11-24:
```
[11/95] Compiling Vala source ../src/fc-cli/fc-get.vala ../src/common/ser...ommon/mwplog.vala ../src/common/devman.vala ../src/common/mwptermcap.vala
FAILED: fc-get.p/src/fc-cli/fc-get.c fc-get.p/src/common/serial-device.c fc-get.p/src/common/mspmsg.c fc-get.p/src/common/sede.c fc-get.p/src/common/mspcmd.c fc-get.p/src/common/mwplog.c fc-get.p/src/common/devman.c fc-get.p/src/common/mwptermcap.c 
valac -C --debug --debug --pkg linux --pkg posix --pkg gio-2.0 --pkg gudev-1.0 --color=always --directory fc-get.p --basedir ../ --define=LINUX --define=OLDTVI --vapidir /usr/src/mwptools/src/common --pkg mwpfuncs -D USE_TERMCAP ../src/fc-cli/fc-get.vala ../src/common/serial-device.vala ../src/common/mspmsg.vala ../src/common/sede.vala ../src/common/mspcmd.vala ../src/common/mwplog.vala ../src/common/devman.vala ../src/common/mwptermcap.vala
../src/common/serial-device.vala:1130.46-1130.46: error: syntax error, expected identifier
										fr_publish(SportDev.get_buffer()[1:]);
										                                   ^
../src/common/serial-device.vala:1152.44-1152.44: error: syntax error, expected identifier
								fr_publish(SportDev.get_buffer()[1:]);
								                                   ^
Compilation failed: 2 error(s), 0 warning(s)
[12/95] Compiling Vala source ../src/samples/cliterm/cliterm.vala ../src/...rc/common/mspcmd.vala ../src/common/mwplog.vala ../src/common/devman.vala
FAILED: cliterm.p/src/samples/cliterm/cliterm.c cliterm.p/src/common/serial-device.c cliterm.p/src/common/mspmsg.c cliterm.p/src/common/sede.c cliterm.p/src/common/mspcmd.c cliterm.p/src/common/mwplog.c cliterm.p/src/common/devman.c 
valac -C --debug --debug --pkg linux --pkg posix --pkg gio-2.0 --pkg gudev-1.0 --color=always --directory cliterm.p --basedir ../ --define=LINUX --define=OLDTVI --vapidir /usr/src/mwptools/src/common --pkg mwpfuncs ../src/samples/cliterm/cliterm.vala ../src/common/serial-device.vala ../src/common/mspmsg.vala ../src/common/sede.vala ../src/common/mspcmd.vala ../src/common/mwplog.vala ../src/common/devman.vala
../src/common/serial-device.vala:1130.46-1130.46: error: syntax error, expected identifier
										fr_publish(SportDev.get_buffer()[1:]);
										                                   ^
../src/common/serial-device.vala:1152.44-1152.44: error: syntax error, expected identifier
								fr_publish(SportDev.get_buffer()[1:]);
								                                   ^
Compilation failed: 2 error(s), 0 warning(s)
[15/95] Compiling Vala source ../src/mwp/mwp.vala ../src/mwp/liststore.va...rc/common/mlock.vala ../src/common/json_io.vala ../src/common/devman.vala
FAILED: mwp.p/src/mwp/mwp.c mwp.p/src/mwp/liststore.c mwp.p/src/mwp/mwp_markers.c mwp.p/src/mwp/mwp_craft.c mwp.p/src/mwp/settings.c mwp.p/src/mwp/dialogs.c mwp.p/src/mwp/mapsources.c mwp.p/src/mwp/local_geo.c mwp.p/src/mwp/logger.c mwp.p/src/mwp/mwp-replay.c mwp.p/src/mwp/map_seeder.c mwp.p/src/mwp/mwp-layout.c mwp.p/src/mwp/iparser.c mwp.p/src/mwp/mwp_bbox.c mwp.p/src/mwp/ah_widget.c mwp.p/src/mwp/get_places.c mwp.p/src/mwp/mission.c mwp.p/src/mwp/mwp_dbus.c mwp.p/src/mwp/name_type_map.c mwp.p/src/mwp/mwp_pe_support.c mwp.p/src/mwp/kmlovly.c mwp.p/src/mwp/mwp_upower.c mwp.p/src/mwp/replicate_wp.c mwp.p/src/mwp/mwp-mission-preview.c mwp.p/src/mwp/mwp-radar_list.c mwp.p/src/mwp/dbus-mwp-client.c mwp.p/src/mwp/mwp_safehomes.c mwp.p/src/mwp/mwp_otx.c mwp.p/src/mwp/mwp_xdg_notify.c mwp.p/src/mwp/mwp_elevations.c mwp.p/src/mwp/mdialog.c mwp.p/src/mwp/mm_support.c mwp.p/src/mwp/flysky.c mwp.p/src/common/xmlio.c mwp.p/src/common/serial-device.c mwp.p/src/common/geocalc.c mwp.p/src/common/utils.c mwp.p/src/common/mspmsg.c mwp.p/src/common/sede.c mwp.p/src/common/mspcmd.c mwp.p/src/common/mwc.c mwp.p/src/common/mwplog.c mwp.p/src/common/mavcmd.c mwp.p/src/common/cliterm.c mwp.p/src/common/mlock.c mwp.p/src/common/json_io.c mwp.p/src/common/devman.c 
valac -C --debug --debug --pkg gmodule-2.0 --pkg linux --pkg posix --pkg vte-2.91 --pkg gstreamer-1.0 --pkg libsoup-2.4 --pkg cairo --pkg pango --pkg gdl-3.0 --pkg clutter-gtk-1.0 --pkg gio-2.0 --pkg libxml-2.0 --pkg champlain-0.12 --pkg champlain-gtk-0.12 --pkg clutter-1.0 --pkg atk --pkg gtk+-3.0 --pkg gudev-1.0 --color=always --directory mwp.p --basedir ../ --define=LINUX --define=OLDTVI --vapidir /usr/src/mwptools/src/common --pkg mwpfuncs --define=USE_ESPEAK ../src/mwp/mwp.vala ../src/mwp/liststore.vala ../src/mwp/mwp_markers.vala ../src/mwp/mwp_craft.vala ../src/mwp/settings.vala ../src/mwp/dialogs.vala ../src/mwp/mapsources.vala ../src/mwp/local_geo.vala ../src/mwp/logger.vala ../src/mwp/mwp-replay.vala ../src/mwp/map_seeder.vala ../src/mwp/mwp-layout.vala ../src/mwp/iparser.vala ../src/mwp/mwp_bbox.vala ../src/mwp/ah_widget.vala ../src/mwp/get_places.vala ../src/mwp/mission.vala ../src/mwp/mwp_dbus.vala ../src/mwp/name_type_map.vala ../src/mwp/mwp_pe_support.vala ../src/mwp/kmlovly.vala ../src/mwp/mwp_upower.vala ../src/mwp/replicate_wp.vala ../src/mwp/mwp-mission-preview.vala ../src/mwp/mwp-radar_list.vala ../src/mwp/dbus-mwp-client.vala ../src/mwp/mwp_safehomes.vala ../src/mwp/mwp_otx.vala ../src/mwp/mwp_xdg_notify.vala ../src/mwp/mwp_elevations.vala ../src/mwp/mdialog.vala ../src/mwp/mm_support.vala ../src/mwp/flysky.vala ../src/common/xmlio.vala ../src/common/serial-device.vala ../src/common/geocalc.vala ../src/common/utils.vala ../src/common/mspmsg.vala ../src/common/sede.vala ../src/common/mspcmd.vala ../src/common/mwc.vala ../src/common/mwplog.vala ../src/common/mavcmd.v [... cut ...]
										fr_publish(SportDev.get_buffer()[1:]);
										                                   ^
../src/common/serial-device.vala:1152.44-1152.44: error: syntax error, expected identifier
								fr_publish(SportDev.get_buffer()[1:]);
								                                   ^
Compilation failed: 2 error(s), 0 warning(s)
[16/95] Compiling Vala source ../src/samples/area-tool/survey.vala ../src...rc/common/xmlio.vala ../src/common/geocalc.vala ../src/common/mspcmd.vala
ninja: build stopped: subcommand failed.
```

stronnag, 2021-11-24:
Thanks, I always forget about the ancient Ubuntu compiler. Fix on its way.

wx4cb, 2021-11-24:
lol... considering im riding the hippo nowadays lol

stronnag, 2021-11-24:
Fixed. Pull again please.

wx4cb, 2021-11-24:
ok, update....

```
TL;DR Have the lora device plugged in, startup mwp and then hit the reset button on the device... if mwp crashes, disconnect the module, then reinsert the lora device after about 30 seconds (to give the TTY device node time to die), then try again
```

if for some reason the system was up and running with the module in GCS mode (due to close of mwp), mwp will hang at "setting up radar device".

if for some reason it is sitting there booted up at NoFC on the lora device, mwp will never initialise it.If i then hit reset on the LORA device then it will initialise it (assume because it sees the MSP messages)

```
2021-11-24T15:35:12-0500 RDR-msg: MSP_CMDS_FC_VARIANT
2021-11-24T15:35:12-0500 RDR-msg: MSP_CMDS_FC_VERSION
2021-11-24T15:35:12-0500 RDR-msg: MSP_CMDS_NAME
2021-11-24T15:35:15-0500 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2021-11-24T15:35:15-0500 RDR-recv 0: Lat, Lon 0.000000 0.000000
```

On all occasions i get the following at some point during the startup of the reader process

```2021-11-24T15:34:41-0500 Comm error count 1
2021-11-24T15:34:41-0500 expected header0 (0xf1)```

wx4cb, 2021-11-24:
is this all mwps fault? no i don't think so, but some form of timeout should be added to the radar module setup to stop it from hanging. also, hav to make sure the module is not in GCS mode already before starting mwp.

I wonder if you can just read the port and see if it's already squirting out messages (as it would be if already in gcs mode) ???

stronnag, 2021-11-24:
You should be able to use any serial terminal:

```
cat < /dev/name
```
or maybe
```
cliterm -d /dev/name -n
```

stronnag, 2021-11-24:
I have a theory that the enabling of multiple radar devices may have upset USB detection. 
May take a little while to get to the bottom of it.

Did it work prior to multiple radars (the master branch is still pre-merge).

wx4cb, 2021-11-24:
ok - problem may not have been MWP.... I just completely wiped the  binary from /usr/loca/bin and refetched the multi-radars branch, recompiled and it's working.

startup MWP, hit the reset button on the lora module and it kicks in. I wonder if it was just something funky with this laptop.

stronnag, 2021-11-24:
OK, thanks for that. And there is a minor issue that the device will now be auto-detected and  added to the Devices drop down which might be confusing, so I need to fix that anyway.

wx4cb, 2021-11-24:
i did notice that, but that's a minor thing

wx4cb, 2021-12-05:
how about being able to use a gpsd instance if one exists ? might be easier as you can then use it via localhost ?

stronnag, 2021-12-05:
Interesting idea. 
I'm still not sure I really understand the issue, you can tell mwp where the unarmed fake home is with Ctrl-Shift-H and position the icon.

wx4cb, 2021-12-05:
cus then you can use it while moving :D

stronnag, 2021-12-05:
You need to explain the use case to me in slow, coherent sentences.
_"For I am a bear of very little brain, and long words bother me."_

wx4cb, 2021-12-05:
well. for example... lets say that you're at one end of the park but the safehome of the plane is 500 feet away, would be nice to be able to see where mwp is in relation to the plane is on the map

stronnag, 2021-12-05:
Ok, gpsd is trival. If it's present, and has a fix, show an icon.

wx4cb, 2021-12-05:
yup. pretty much

stronnag, 2021-12-06:
Ok, so there will be a `Location Icon` menu option (which I may changes to `GCS Location`, as that more expresses the intent).
[image]

By default, it will display a tasteful gold star which one may drag around. It has no purpose other than showing some random location.
[image]
If you don't like the icon, you can override it by creating your own icon in `~/.config/mwp/pixmaps/` (as one can for _any_ mwp specific icon), with the same file name as the default in `$prefix/share/mwp/pixmaps/`. So below, I override  `$prefix/share/mwp/pixmaps/gcs.svg` with `~/.config/mwp/pixmaps/gcs.svg` (which happens to be: (a) a stronnag and (b) a PNG, not SVG, but we're a real OS and file "extensions" are an advisory illusion).
[image]

If `gpsd` is detected (on `localhost`), then the position will be driven by `gpsd`, as long as it has  a 3D fix.

wx4cb, 2021-12-06:
is this in the development version or you havent pushed it yet?

stronnag, 2021-12-07:
It's the "concept" at the moment. I'll do some testing on it and push it later today.

stronnag, 2021-12-07:
Ok it's pushed to development, now you see stuff like (with safehome):
[image]
`gpsd` will drive the GCS location icon if available and valid.

stronnag, 2021-12-17:
merged into master

closed 2021-12-17 by stronnag

stronnag, 2021-12-24:
@wx4cb how about:
* if the GCS icon is displayed (or driven by `gpsd`)
then, for inav radar,
* instead of being a passive 'GCS' node; instead
* **mwp** advertises itself as `INAV` with the GCS location
In the latter case, any inav-radar aircraft would see the GCS as a node

wx4cb, 2021-12-24:
so inav-radar has multiple modes for a gcs? yea. it would be a good thing to advertise itself as a gcs so that it shows up on the OSD. the only issue i can see is you may have to increase the number of nodes in I.R if you have  4 planes running right ?
but yea, that's a great idea.

stronnag, 2021-12-24:
The way inav-radar works is, for a node:

* If you declare yourself as `GCS`, then the node is passive, does not take up a lora slot, does not advertise itself to any other nodes.
* If you declare as `INAV` then you take up a node, and must respond to additional MSP queries from the radar board, including `MSP_RAW_GPS', which **mwp** would do if the GCS icon is enabled. You then become visible to other nodes.

As noted, you need sufficient radar slots for this to work.