stronnag/mwptools #108: Add option to have fpv feed in app

wx4cb, 2021-11-24:
Not sure its possible, but maybe there is a way to have a pop up window that includes the video feed from something like /dev/video0

Would be a nice addition for when you are showing people.

There are multiple devices that can be connected via usb that show up as a "camera".

Hardware can be provided if requested:)

wx4cb, 2021-11-24:
Looks to have a v4l2 api in the docs. doing more investigation:

https://valadoc.org/v4l2/index.htm

stronnag, 2021-11-26:
Adding media playback via GStreamer is  pretty easy (mwp already does it for audio).
So please have a think about exactly what you want and the relevant media sources, for example:

| When | Media source |
| ------- |  ------------------ |
| BBL replay | MP4 video file |
| Receiving telemetry | Webcam |
| Receiving telemetry | Streaming video (**please  specify format and source / interface**) |

Also, how do you think it should be displayed, for example:
* fixed window in a corner of the map
* "always on top" floating window (with as minimal decoration as possible)

wx4cb, 2021-11-26:
ok cool.

not sure much about the streaming video part of it, would have to think about that one... maybe an RTSP/MP4 stream.

As far as the webcam, if you do it for a video4Linux (V4L) device that should prety much encompass 99% of things like webcams, and also the USB FPV receivers for example this one here which is the one I use myself and works great with linux/VLC already (shows up as a regular camera device eg: /dev/video0). 

Hardware for the latter as shown below can be provided if requested.

https://smile.amazon.com/Makerfire-FPV-Receiver-Android-SKYDROID/dp/B09DCB997S/ref=sr_1_3?keywords=usb+fpv+receiver&qid=1637943649&sr=8-3

as for how, right now mission planner for AP afaik shows it "underneath" the AHI which i think is silly, but i would suggest maybe  just being able to have a regular window over the top would also allow it to be moved to a secondary monitor if one is used so that would be the ideal solution unless you can think of anything better.

As for how, i would suggest running unde ra separate child process (assuming that you can fork in vala), heck, you could even use the VLC library and have VLC handle all the decoding and displaying, that would reduce a lot of the work imho as well.

stronnag, 2021-11-26:
if you just want VLC, you can do that now.

No need for any hardware, I have some webcams working fine with GStreamer / v4l2

wx4cb, 2021-11-26:
i just suggested VLC as that's the only one i've really ever used (not really used gstreamer).

stronnag, 2021-11-28:
Ok, you asked for it, you got it.
Please try the `video_star` branch. You will find a new option `Video Stream` on the `View` menu. This will give you a dialogue to select the video source. V4L2 video devices are auto-detected and populate the combo box.

[image]

You can also use streaming video (http/https or any many other scheme supported by Gstreamer), or local files, e.g:
[image]

For a webcam, the stream starts immediately and there is no slider; for media where the size is fixed and known, there is a slider and the user must start the stream. This is mainly to support 'post-event' video in blackbox replays (which is to be implemented), where it might be necessary to position the video to match the replay.

Anyway, please give it a try.

stronnag, 2021-11-28:
On Debian / Ubuntu, it will be necessary to:
```
sudo apt install libgstreamer-plugins-base1.0-dev
```
Prior to building.

stronnag, 2021-11-28:
Some streaming examples for test:

* https://www.freedesktop.org/software/gstreamer-sdk/data/media/sintel_trailer-480p.webm
* rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4

wx4cb, 2021-11-28:
i see it pickup the webcam on the laptop, but i don't see a view option in the menu unless i'm blin
[image]
d

wx4cb, 2021-11-28:
```
charlesb@ToshTop:/usr/src/stronnag/mwptools/build$ ./mwp
2021-11-28T16:24:26-0500 mwp startup version: mwp 50f6ad6 / 2021-11-28T18:22:41+00:00 4.329.723 on Xlib
2021-11-28T16:24:26-0500 "Ubuntu 21.04" on ToshTop running Linux 5.11.0-40-generic x86_64
2021-11-28T16:24:26-0500 hypervisor: none
2021-11-28T16:24:26-0500 Failed to find "blackbox_decode" on PATH
2021-11-28T16:24:26-0500 Failed to find "fl2ltm" on PATH
2021-11-28T16:24:26-0500 Failed to find "mavlogdump.py" on PATH
2021-11-28T16:24:26-0500 Using speech api 1 [espeak]
2021-11-28T16:24:26-0500 Starting Bing proxy 
2021-11-28T16:24:26-0500 BB load async map zoom : true
2021-11-28T16:24:26-0500 Loaded /usr/local/share/mwp/vcols.css
2021-11-28T16:24:26-0500 Fixed font true
2021-11-28T16:24:26-0500 Delta speed 0.000000
2021-11-28T16:24:26-0500 Host power: 100%, state: Fully Charged (4), warn: None (1)
2021-11-28T16:24:27-0500 Gst: add TOSHIBA_Web_Camera_-_HD - SunplusIT_Inc </dev/video0>
```

stronnag, 2021-11-28:
Pleased to see if found the webcam. Seems like either
* You didn't install everything `ninja install` ; the updated `mwp.ui` is not installed; or
* You have done multiple installs with different `--prefix` without first removing  an older install; mwp is finding the old `mwp.ui` rather the newer version.

stronnag, 2021-11-28:
Thusly:
[image]

wx4cb, 2021-11-28:
oops completely forgot about that. :D

wx4cb, 2021-11-28:
[image]

Works with USB FPV Receiver :D heh

stronnag, 2021-11-28:
Nice. It was fun doing it, learned a lot.

snaewe, 2021-12-02:
I'm trying to use an Eachine ROTG01 (https://www.banggood.com/Eachine-ROTG01-UVC-OTG-5_8G-150CH-Full-Channel-FPV-Receiver-For-Android-Mobile-Phone-Tablet-Smartphone-Non-original-p-1147692.html). It works in VLC and 'guvcview' but not in mwp. I don't see it in the 'Webcams' selection box, and when I enter 'v4l2:///dev/video0' in the URI box (That's the URI VLC seems to use) I only get a green window.
This what I see in 'dmesg' when I plug it in:

```
usb 2-1.4: New USB device found, idVendor=18ec, idProduct=5850, bcdDevice= 1.00
usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 2-1.4: Product: USB2.0 PC CAMERA
usb 2-1.4: Manufacturer: ARKMICRO
uvcvideo: Found UVC 1.00 device USB2.0 PC CAMERA (18ec:5850)
uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
input: USB2.0 PC CAMERA as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.0/input/input31
```

Any ideas? Thx.

stronnag, 2021-12-02:
Can we start with:
```
gst-launch-1.0 playbin uri=v4l2:///dev/video0
```
Does that work?

wx4cb, 2021-12-02:
I believe from memory that a green video would be a codec ssue check to see
in check to see in vlc watch codec the video screen is using in media
information window

stronnag, 2021-12-02:
Next, please unpack the attached archive,  then:
```
cd gst-test
make
./udev-vid
# after output, if none, please plug / unplug the device, then if nothing happens, ^C
^C
# Save the output if any, I'd like to see it.
# then
./gst-devmon
# same procedure, I'd like to see the outout
```
Here, with an old Mobuis, stuff like:
```
./udev-vid 
---- all video devices ---
>>>>>>>>>>>>>>>>> video0 /dev/video0 
  COLORD_DEVICE => 1
  COLORD_KIND => camera
  CURRENT_TAGS => :uaccess:seat:
  DEVLINKS => /dev/v4l/by-id/usb-C-DUTEK_Mobius-video-index0 /dev/v4l/by-path/pci-0000:00:14.0-usb-0:4.2:1.0-video-index0
  DEVNAME => /dev/video0
  DEVPATH => /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4.2/3-4.2:1.0/video4linux/video0
  ID_BUS => usb
  ID_FOR_SEAT => video4linux-pci-0000_00_14_0-usb-0_4_2_1_0
  ID_MODEL => Mobius
  ID_MODEL_ENC => Mobius
  ID_MODEL_ID => 1002
  ID_PATH => pci-0000:00:14.0-usb-0:4.2:1.0
  ID_PATH_TAG => pci-0000_00_14_0-usb-0_4_2_1_0
  ID_REVISION => 0010
  ID_SERIAL => C-DUTEK_Mobius
  ID_TYPE => video
  ID_USB_DRIVER => uvcvideo
  ID_USB_INTERFACES => :0e0100:0e0200:010100:010200:
  ID_USB_INTERFACE_NUM => 00
  ID_V4L_CAPABILITIES => :capture:
  ID_V4L_PRODUCT => Video Capture 2
  ID_V4L_VERSION => 2
  ID_VENDOR => C-DUTEK
  ID_VENDOR_ENC => C-DUTEK\x20
  ID_VENDOR_ID => 0603
  MAJOR => 81
  MINOR => 0
  SUBSYSTEM => video4linux
  TAGS => :uaccess:seat:
  USEC_INITIALIZED => 6496732941
/dev/video0 Mobius C-DUTEK
>>>>>>>>>>>>>>>>> video1 /dev/video1 
  COLORD_DEVICE => 1
  COLORD_KIND => camera
  CURRENT_TAGS => :uaccess:seat:
  DEVLINKS => /dev/v4l/by-path/pci-0000:00:14.0-usb-0:4.2:1.0-video-index1 /dev/v4l/by-id/usb-C-DUTEK_Mobius-video-index1
  DEVNAME => /dev/video1
  DEVPATH => /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4.2/3-4.2:1.0/video4linux/video1
  ID_BUS => usb
  ID_FOR_SEAT => video4linux-pci-0000_00_14_0-usb-0_4_2_1_0
  ID_MODEL => Mobius
  ID_MODEL_ENC => Mobius
  ID_MODEL_ID => 1002
  ID_PATH => pci-0000:00:14.0-usb-0:4.2:1.0
  ID_PATH_TAG => pci-0000_00_14_0-usb-0_4_2_1_0
  ID_REVISION => 0010
  ID_SERIAL => C-DUTEK_Mobius
  ID_TYPE => video
  ID_USB_DRIVER => uvcvideo
  ID_USB_INTERFACES => :0e0100:0e0200:010100:010200:
  ID_USB_INTERFACE_NUM => 00
  ID_V4L_CAPABILITIES => :
  ID_V4L_PRODUCT => Metadata 2
  ID_V4L_VERSION => 2
  ID_VENDOR => C-DUTEK
  ID_VENDOR_ENC => C-DUTEK\x20
  ID_VENDOR_ID => 0603
  MAJOR => 81
  MINOR => 1
  SUBSYSTEM => video4linux
  TAGS => :uaccess:seat:
  USEC_INITIALIZED => 6496733511
^C
 ###############################################
./gst-devmon 
Props pipewire-proplist, object.path=(string)v4l2:/dev/video0, device.api=(string)v4l2, media.class=(string)Video/Source, api.v4l2.path=(string)/dev/video0, api.v4l2.cap.driver=(string)uvcvideo, api.v4l2.cap.card=(string)"Video\ Capture\ 2", api.v4l2.cap.bus_info=(string)usb-0000:00:14.0-4.2, api.v4l2.cap.version=(string)5.15.6, api.v4l2.cap.capabilities=(string)84a00001, api.v4l2.cap.device-caps=(string)04200001, device.id=(string)52, node.name=(string)v4l2_input.pci-0000_00_14.0-usb-0_4.2_1.0, node.description=(string)"Video\ Capture\ 2", factory.name=(string)api.v4l2.source, node.pause-on-idle=(string)false, factory.id=(string)10, client.id=(string)32, media.role=(string)Camera, node.driver=(string)true, object.id=(string)53;
GST: add Mobius - C-DUTEK </dev/video0>
^C
```

stronnag, 2021-12-02:
And you do have an (undocumented) dependency `gst-plugins-good`  (Arch) / `gstreamer1.0-plugins-good` (Debian / Ubuntu) installed.

Lack of that would probably explain both lack of detection and green-screen.

stronnag, 2021-12-02:
* [mwp wiki Video Article](https://github.com/stronnag/mwptools/wiki/mwp_video_player) (new).

* [mwp wiki Recent Changes](https://github.com/stronnag/mwptools/wiki/Recent-Changes) (updated).

wx4cb, 2021-12-02:
here's mine for the Eachine ROT-G02 that i have a screenshot above working as designed
```
charlesb@ToshTop:/usr/src/wx4cb/gst-test$ ./udev-vid 2>&1 > udev-vid.txt
>>>>>>>>>>>>>>>>> video0 /dev/video0 
  COLORD_DEVICE => 1
  COLORD_KIND => camera
  CURRENT_TAGS => :uaccess:seat:
  DEVLINKS => /dev/v4l/by-path/pci-0000:00:14.0-usb-0:1:1.0-video-index0 /dev/v4l/by-id/usb-ARKMICRO_USB2.0_PC_CAMERA-video-index0
  DEVNAME => /dev/video0
  DEVPATH => /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
  ID_BUS => usb
  ID_FOR_SEAT => video4linux-pci-0000_00_14_0-usb-0_1_1_0
  ID_MODEL => USB2.0_PC_CAMERA
  ID_MODEL_ENC => USB2.0\x20PC\x20CAMERA
  ID_MODEL_ID => 3399
  ID_PATH => pci-0000:00:14.0-usb-0:1:1.0
  ID_PATH_TAG => pci-0000_00_14_0-usb-0_1_1_0
  ID_REVISION => 0100
  ID_SERIAL => ARKMICRO_USB2.0_PC_CAMERA
  ID_TYPE => video
  ID_USB_DRIVER => uvcvideo
  ID_USB_INTERFACES => :0e0100:0e0200:010100:010200:
  ID_USB_INTERFACE_NUM => 00
  ID_V4L_CAPABILITIES => :capture:
  ID_V4L_PRODUCT => USB2.0 PC CAMERA
  ID_V4L_VERSION => 2
  ID_VENDOR => ARKMICRO
  ID_VENDOR_ENC => ARKMICRO
  ID_VENDOR_ID => 18ec
  MAJOR => 81
  MINOR => 0
  SUBSYSTEM => video4linux
  TAGS => :uaccess:seat:
  USEC_INITIALIZED => 5118596
>>>>>>>>>>>>>>>>> video1 /dev/video1 
  COLORD_DEVICE => 1
  COLORD_KIND => camera
  CURRENT_TAGS => :uaccess:seat:
  DEVLINKS => /dev/v4l/by-id/usb-ARKMICRO_USB2.0_PC_CAMERA-video-index1 /dev/v4l/by-path/pci-0000:00:14.0-usb-0:1:1.0-video-index1
  DEVNAME => /dev/video1
  DEVPATH => /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video1
  ID_BUS => usb
  ID_FOR_SEAT => video4linux-pci-0000_00_14_0-usb-0_1_1_0
  ID_MODEL => USB2.0_PC_CAMERA
  ID_MODEL_ENC => USB2.0\x20PC\x20CAMERA
  ID_MODEL_ID => 3399
  ID_PATH => pci-0000:00:14.0-usb-0:1:1.0
  ID_PATH_TAG => pci-0000_00_14_0-usb-0_1_1_0
  ID_REVISION => 0100
  ID_SERIAL => ARKMICRO_USB2.0_PC_CAMERA
  ID_TYPE => video
  ID_USB_DRIVER => uvcvideo
  ID_USB_INTERFACES => :0e0100:0e0200:010100:010200:
  ID_USB_INTERFACE_NUM => 00
  ID_V4L_CAPABILITIES => :
  ID_V4L_PRODUCT => USB2.0 PC CAMERA
  ID_V4L_VERSION => 2
  ID_VENDOR => ARKMICRO
  ID_VENDOR_ENC => ARKMICRO
  ID_VENDOR_ID => 18ec
  MAJOR => 81
  MINOR => 1
  SUBSYSTEM => video4linux
  TAGS => :uaccess:seat:
  USEC_INITIALIZED => 5112885
>>>>>>>>>>>>>>>>> video2 /dev/video2 
  COLORD_DEVICE => 1
  COLORD_KIND => camera
  CURRENT_TAGS => :uaccess:seat:
  DEVLINKS => /dev/v4l/by-id/usb-SunplusIT_Inc_TOSHIBA_Web_Camera_-_HD-video-index0 /dev/v4l/by-path/pci-0000:00:14.0-usb-0:7:1.0-video-index0
[... 46 lines cut ...]
  ID_V4L_CAPABILITIES => :
  ID_V4L_PRODUCT => TOSHIBA Web Camera - HD: TOSHIB
  ID_V4L_VERSION => 2
  ID_VENDOR => SunplusIT_Inc
  ID_VENDOR_ENC => SunplusIT\x20Inc
  ID_VENDOR_ID => 04f2
  MAJOR => 81
  MINOR => 3
  SUBSYSTEM => video4linux
  TAGS => :uaccess:seat:
  USEC_INITIALIZED => 5165132
^C
```

gst-devmon
```
charlesb@ToshTop:/usr/src/wx4cb/gst-test$ ./gst-devmon 
Props pipewire-proplist, object.path=(string)v4l2:/dev/video0, device.api=(string)v4l2, media.class=(string)Video/Source, api.v4l2.path=(string)/dev/video0, api.v4l2.cap.driver=(string)uvcvideo, api.v4l2.cap.card=(string)"USB2.0\ PC\ CAMERA", api.v4l2.cap.bus_info=(string)usb-0000:00:14.0-1, api.v4l2.cap.version=(string)5.11.22, api.v4l2.cap.capabilities=(string)84a00001, api.v4l2.cap.device-caps=(string)04200001, device.id=(string)31, node.name=(string)v4l2_input.pci-0000_00_14.0-usb-0_1_1.0, node.description=(string)"USB2.0\ PC\ CAMERA", factory.name=(string)api.v4l2.source, node.pause-on-idle=(string)false, factory.id=(string)10, client.id=(string)30, media.role=(string)Camera, node.driver=(string)true, object.id=(string)35;
GST: add USB2.0_PC_CAMERA - ARKMICRO </dev/video0>
Props pipewire-proplist, object.path=(string)v4l2:/dev/video2, device.api=(string)v4l2, media.class=(string)Video/Source, api.v4l2.path=(string)/dev/video2, api.v4l2.cap.driver=(string)uvcvideo, api.v4l2.cap.card=(string)"TOSHIBA\ Web\ Camera\ -\ HD:\ TOSHIB", api.v4l2.cap.bus_info=(string)usb-0000:00:14.0-7, api.v4l2.cap.version=(string)5.11.22, api.v4l2.cap.capabilities=(string)84a00001, api.v4l2.cap.device-caps=(string)04200001, device.id=(string)33, node.name=(string)v4l2_input.pci-0000_00_14.0-usb-0_7_1.0, node.description=(string)"TOSHIBA\ Web\ Camera\ -\ HD\ \ TOSHIB", factory.name=(string)api.v4l2.source, node.pause-on-idle=(string)false, factory.id=(string)10, client.id=(string)30, media.role=(string)Camera, node.driver=(string)true, object.id=(string)37;
GST: add TOSHIBA_Web_Camera_-_HD - SunplusIT_Inc </dev/video2>
GST: remove  </dev/video0>
Props pipewire-proplist, object.path=(string)v4l2:/dev/video0, device.api=(string)v4l2, media.class=(string)Video/Source, api.v4l2.path=(string)/dev/video0, api.v4l2.cap.driver=(string)uvcvideo, api.v4l2.cap.card=(string)"USB2.0\ PC\ CAMERA", api.v4l2.cap.bus_info=(string)usb-0000:00:14.0-1, api.v4l2.cap.version=(string)5.11.22, api.v4l2.cap.capabilities=(string)84a00001, api.v4l2.cap.device-caps=(string)04200001, device.id=(string)32, node.name=(string)v4l2_input.pci-0000_00_14.0-usb-0_1_1.0, node.description=(string)"USB2.0\ PC\ CAMERA", factory.name=(string)api.v4l2.source, node.pause-on-idle=(string)false, factory.id=(string)10, client.id=(string)30, media.role=(string)Camera, node.driver=(string)true, object.id=(string)35;
GST: add USB2.0_PC_CAMERA - ARKMICRO </dev/video0>
```

wx4cb, 2021-12-02:
i don't know why there are 2 devices for each physical device (this and the laptop webcam)

stronnag, 2021-12-02:
There always are, some other function other than capture. 
Anyway, I suspect it's the `gstreamer1.0-plugins-good` (driver) issue.

wx4cb, 2021-12-02:
here's a list of my installed gstreamer packages i have installed on here (not all are needed i have other apps that use the others plus i'm a plugin whore :dancing_women: 

as an aside, yes i would beleive it's the plugins missing as the plugins also provide the codecs

```
root@ToshTop:~# dpkg-query -l |grep -i gst
ii  gir1.2-clutter-gst-3.0:amd64               3.0.27-2                                                             amd64        Gobject introspection data for Clutter GStreamer elements
ii  gir1.2-gst-plugins-base-1.0:amd64          1.18.4-1                                                             amd64        GObject introspection data for the GStreamer Plugins Base library
ii  gir1.2-gstreamer-1.0:amd64                 1.18.4-1                                                             amd64        GObject introspection data for the GStreamer library
ii  gnome-video-effects                        0.5.0-1ubuntu1                                                       all          Collection of GStreamer effects
ii  gstreamer1.0-alsa:amd64                    1.18.4-1                                                             amd64        GStreamer plugin for ALSA
ii  gstreamer1.0-clutter-3.0:amd64             3.0.27-2                                                             amd64        Clutter PLugin for GStreamer 1.0
ii  gstreamer1.0-gl:amd64                      1.18.4-1                                                             amd64        GStreamer plugins for GL
ii  gstreamer1.0-gtk3:amd64                    1.18.4-1ubuntu1                                                      amd64        GStreamer plugin for GTK+3
ii  gstreamer1.0-packagekit                    1.2.2-2ubuntu1                                                       amd64        GStreamer plugin to install codecs using PackageKit
ii  gstreamer1.0-pipewire:amd64                0.3.24-3                                                             amd64        GStreamer 1.0 plugin for the PipeWire multimedia server
ii  gstreamer1.0-plugins-bad:amd64             1.18.4-1ubuntu1                                                      amd64        GStreamer plugins from the "bad" set
ii  gstreamer1.0-plugins-base:amd64            1.18.4-1                                                             amd64        GStreamer plugins from the "base" set
ii  gstreamer1.0-plugins-base-apps             1.18.4-1                                                             amd64        GStreamer helper programs from the "base" set
ii  gstreamer1.0-plugins-good:amd64            1.18.4-1ubuntu1                                                      amd64        GStreamer plugins from the "good" set
ii  gstreamer1.0-pulseaudio:amd64              1.18.4-1ubuntu1                                                      amd64        GStreamer plugin for PulseAudio (transitional package)
ii  gstreamer1.0-rtsp:amd64                    1.18.4-1                                                             amd64        RTSP plugin for GStreamer
ii  gstreamer1.0-tools                         1.18.4-1                                                             amd64        Tools for use with GStreamer
ii  gstreamer1.0-x:amd64                       1.18.4-1                                                             amd64        GStreamer plugins for X11 and Pango
ii  libclutter-gst-3.0-0:amd64                 3.0.27-2                                                             amd64        Open GL based interactive canvas library GStreamer elements
ii  libgstreamer-gl1.0-0:amd64                 1.18.4-1                                                             amd64        GStreamer GL libraries
ii  libgstreamer-plugins-bad1.0-0:amd64        1.18.4-1ubuntu1                                                      amd64        GStreamer libraries from the "bad" set
ii  libgstreamer-plugins-base1.0-0:amd64       1.18.4-1                                                             amd64        GStreamer libraries from the "base" set
ii  libgstreamer-plugins-base1.0-dev:amd64     1.18.4-1                                                             amd64        GStreamer development files for libraries from the "base" set
ii  libgstreamer-plugins-good1.0-0:amd64       1.18.4-1ubuntu1                                                      amd64        GStreamer development files for libraries from the "good" set
ii  libgstreamer1.0-0:amd64                    1.18.4-1                                                             amd64        Core GStreamer libraries and elements
ii  libgstreamer1.0-dev:amd64                  1.18.4-1                                                             amd64        GStreamer core development files
ii  libgstrtspserver-1.0-0                     1.18.4-1                                                             amd64        GStreamer RTSP Server (shared library)
```

snaewe, 2021-12-03:
Ok. Here are the results more or less in the order you requested. I hope it's readable. Thx.

[mwp-video-test.txt](https://github.com/stronnag/mwptools/files/7650246/mwp-video-test.txt)

stronnag, 2021-12-03:
> Ok. Here are the results more or less in the order you requested. I hope it's readable. Thx.
> 
> [mwp-video-test.txt](https://github.com/stronnag/mwptools/files/7650246/mwp-video-test.txt)

Thanks, and what distro / version are you running?

wx4cb, 2021-12-03:
That’s weird as I have the exact same device. I’m sitting in the airport right now so no laptop, but what I don’t see is a “capture” capability unless I’m blind. If you look at the one I posted  you’ll see this, … I didn’t see it on your list

```
ID_V4L_CAPABILITIES => :capture:
  ID_V4L_PRODUCT => USB2.0 PC CAMERA
  ID_V4L_VERSION => 2
  ```

wx4cb, 2021-12-03:
Open up vac, go into the media info window and screenshot each tab plz

snaewe, 2021-12-03:
```
$ cat /etc/lsb-release 
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=20.2
DISTRIB_CODENAME=uma
DISTRIB_DESCRIPTION="Linux Mint 20.2 Uma"

```

snaewe, 2021-12-03:
@wx4cb There's only something on the "Codec" page (Yes, I'm dschörmän...):
[image]

wx4cb, 2021-12-03:
Ok so mpg should work, when I get home later, I’ll check mine

stronnag, 2021-12-03:
> ```
> $ cat /etc/lsb-release 
> DISTRIB_ID=LinuxMint
> DISTRIB_RELEASE=20.2
> DISTRIB_CODENAME=uma
> DISTRIB_DESCRIPTION="Linux Mint 20.2 Uma"
> ```

Unsupported OS. udev fails to report a device node; under Mint the device fails to offer the capture capability. Not looking very hopeful.

snaewe, 2021-12-03:
Hhmmm...why do other applications (qv4l2 as well) work, then?

stronnag, 2021-12-03:
> Hhmmm...why do other applications (qv4l2 as well) work, then?

No idea. Don't care; irrelevant. mwp uses gstreamer. If your OS is so broken that gstreamer doesn't work and udev, a fundamental OS component doesn't recognise the device node, it's pretty much game over.
You might try this pipeline; please provide all the output and whether there is a green display.

```
gst-launch-1.0 -v v4l2src ! videoconvert ! videoscale ! xvimagesink
```
If it displays an image, then we can perhaps work around the udev (mis)behaviour.

wx4cb, 2021-12-03:
Vlc rrads the stream directly, most probably anything that uses v4l or
gstreamer probably wont work, although i obviously dont know for definite

snaewe, 2021-12-03:
```
$ gst-launch-1.0 -v v4l2src ! videoconvert ! videoscale ! xvimagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000470153
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

```

No image displayed.
If it doesn't work, it doesn't work.
Would be nice, though...
Thanks.

wx4cb, 2021-12-03:
interesting... @stronnag looks to be similar to this maybe ? </straw clutching>

https://github.com/dusty-nv/jetson-inference/issues/461

stronnag, 2021-12-03:
I know why mwp does not 'see' Stefan's camera, that's due to the fact the the gstreamer monitor API returns different tags depending on the underlying "media" api. We've (probably) both been using pipewire. Older APIs return different tags. 
I've push a development commit which should mean that @snaewe 's camera is detected and appears in mwp's "Video Stream" dialogue (albeit the device names are now the from the less helpful gst API `(vice udev)). However this doesn't mitigate the fact that Mint is an unsupported disaster area where gstreamer can't play video from the camera.

snaewe, 2021-12-03:
@stronnag You're the best! I'll try it ASAP.

stronnag, 2021-12-03:
Well ... until you can persuade Mint to play video from the camera, it's still green screen time.

snaewe, 2021-12-03:
I tried 9910c9ed8c . Still green...

snaewe, 2021-12-03:
Ok. I started mwp, disconnected the device, re-connected it (mwp told me 
```
2021-12-03T20:49:52+0100 GST: "add" <USB2.0 PC CAMERA> </dev/video0>
2021-12-03T20:49:52+0100 VideoDevs </dev/video0> <USB2.0 PC CAMERA>)

```
and then my device was listed in the video source selection dialog, but the video is still green.
[image]

stronnag, 2021-12-03:
Could have told you that ... a Mint bonus, everything is green, even videos that other distros play ... I can't fix Mint.

snaewe, 2021-12-03:
> Could have told you that ... a Mint bonus, everything is green, even videos that other distros play ... I can't fix Mint.

That's Ok for me....
As I said before, it would be nice if it worked, but hey.
Thank you very much for you effort, Jonathan.

stronnag, 2021-12-04:
FYI, I have a camera here (an old Mobius) that worked on some computers and not others, including, annoyingly, my main development box. Seems the issue was an old  USB2.0 (extension) hub that didn't provide enough bandwidth; so I just got a black screen.

Fixed by setting uvcvideo quirk 640:
`UVC_QUIRK_FIX_BANDWIDTH` (0x80, 128)
`UVC_QUIRK_RESTRICT_FRAME_RATE` (0x200, 512)

Test fix
```
sudo rmmod uvcvideo
modprobe uvcvideo quirks=64
```
Permanent solution, add a file e.g. `/etc/modprobe.d/v4l2.conf` containing the line:
```
options uvcvideo quirks=640
```
or to any other `.conf` file under `/etc/modprobe.d/`

stronnag, 2021-12-05:
closing, committed to master

closed 2021-12-05 by stronnag