stronnag/mwptools #112: Unable to use  Blackbox log replay

Torchi, 2021-12-11:
Hello. I tried to install MWP on a Mint Linux computer and I have fiddled a bit and I would be willing to continue (in any case I get confidence with Linux in trying) but I am now stuck in a stable error case. Maybe you can spot the problem at glance. I get to the point you can see in the attached figure but... once configured the Blackbox log and given OK... nothing happens. The program MWP stderr file is attached. I tried several situations and files and it never proceed any further. What is the problem ? please
[mwp_stderr_2021-12-09.txt](https://github.com/stronnag/mwptools/files/7696553/mwp_stderr_2021-12-09.txt)
[mwp](https://user-images.githubusercontent.com/18173871/145666808-04e79dd0-51a1-48c8-b8de-e5ecc8f7ecef.png)

stronnag, 2021-12-11:
> Hello. I tried to install MWP on a Mint Linux computer and I have fiddled a bit and I would be willing to continue (in any case I get confidence with Linux in trying) but I am now stuck in a stable [...]

So a couple of things are happening:

* The geonames timezone lookup failed; that's because mwp is using the localised number format in the geonames web service (`?lat=44,801167&lng=8,762139`), vice POSIX C locale style `?lat=44.801167&lng=8.762139`; mea culpa. This is just an inconvenience (UTC v. local time stamps). Now fixed in the `geonames_locale` branch.

* The replayer terminated instantly. This should not happen. Can you  please:
  * run `mwp --debug-flags 32 -b /home/augusto/Scrivania/LOG00001.TXT`, try to play the log, psot the output here.
  * run `fl2ltm --help`, post the output here.
  * run `blackbox-decode --help`,  post the output here.
  * run `fl2ltm -device udp://localhost:42699 --index 1 --type -1 -interval 100 /home/augusto/Scrivania/LOG00001.TXT`,  post the output here.

I think you'll see that the version of `blackbox-decode` is too old. If so, please try the a [later release](https://github.com/iNavFlight/blackbox-tools/releases), either last stable (0.4.6) or the later 4.0.0-rc2 release candidate.

Torchi, 2021-12-11:
I did the tests 1 2 3 4.  Then I realized that blackbox_decode was not giving a version so I replaced it.  I then run tests 1 and 2 again (labeled .. after ..)   with no apparent difference.
[test1.TXT](https://github.com/stronnag/mwptools/files/7696875/test1.TXT)
[test1 after.TXT](https://github.com/stronnag/mwptools/files/7696876/test1.after.TXT)
[test1 after -  mwp_stderr_2021-12-11.txt](https://github.com/stronnag/mwptools/files/7696877/test1.after.-.mwp_stderr_2021-12-11.txt)
[test1 - mwp_stderr_2021-12-11.TXT](https://github.com/stronnag/mwptools/files/7696878/test1.-.mwp_stderr_2021-12-11.TXT)
[test2.TXT](https://github.com/stro
[test3 after.TXT](https://github.com/stronnag/mwptools/files/7696886/test3.after.TXT)
[test4.TXT](https://github.com/stronnag/mwptools/files/7696887/test4.TXT)
nnag/mwptools/files/7696879/test2.TXT)
[test3.TXT](https://github.com/stronnag/mwptools/files/7696880/test3.TXT)

stronnag, 2021-12-11:
The output of fl2ltm is incomplete. Please include the complete outout so there is the version and commit information.

Torchi, 2021-12-11:
Right. 
[test2 bis.TXT](https://github.com/stronnag/mwptools/files/7696982/test2.bis.TXT)

stronnag, 2021-12-11:
Thanks. That provides the information needed. Please use the latest [fl2ltm](https://github.com/stronnag/bbl2kml/releases/latest).
On some OS there was an IPv4 / IPv6 mismatch between `mwp` and `fl2ltm`. The (unsupported) Mint seemly being one of the affected distros).

Torchi, 2021-12-11:
Better but  still NOK.

fl2ltm is now 0.12.0
[test 2 ter.TXT](https://github.com/stronnag/mwptools/files/7697098/test.2.ter.TXT)

https://user-images.githubusercontent.com/18173871/145678318-ef6fbe86-8623-4d46-93a2-1bf6649bffea.mp4

[test 1 bis.TXT](https://github.com/stronnag/mwptools/files/7697102/test.1.bis.TXT)
[test 1 bis - mwp_stderr_2021-12-11.txt](https://github.com/stronnag/mwptools/files/7697103/test.1.bis.-.mwp_stderr_2021-12-11.txt)

mwp (still same as at beginning) now starts drawing but immediately crashes.  Can see mp4 and of course the .TXT files.

stronnag, 2021-12-11:
Please supply the BBL for me to test.

Torchi, 2021-12-11:
I understand  you ask for blackbox.
[LOG00001.TXT](https://github.com/stronnag/mwptools/files/7697116/LOG00001.TXT)

Torchi, 2021-12-11:
I have just now found that this one (same day same model same place) works !!
[LOG00002.TXT](https://github.com/stronnag/mwptools/files/7697120/LOG00002.TXT)

stronnag, 2021-12-11:
LOG00001.TXT, Ubuntu 20.04 LTS (a **supported** OS). `,` decimal locale, Xlib, works fine.
[image]

stronnag, 2021-12-11:
You could try the `geonames_locale` branch; maybe the "abundance of caution" change to the BBL zoom to start function will help, or maybe it's just another reason why I don't support Mint.

Torchi, 2021-12-11:
Today is late but tomorrow I shall come with a solid tests result.   Anyway, for the time being:  the mwp compiled from the geoname_locale branch seems definitely better.  I selected 10 various blackboxes  and the  "normal release" mwp crashes, when starting replay, about  half of the times. I have  done, by now,  several tests with the "geoname_locale" mwp and it never crashed.

closed 2021-12-11 by stronnag