stronnag/mwptools #113: what did i do wrong?

wx4cb, 2022-01-05:
fresh pull of the main dev branch on a new machine

meson buils, ninja and ninja install all apeared to run ok but when i try and run mwp i get the following

```
charlesb@odyssey:/usr/src/mwptools/build$ mwp
2022-01-05T12:46:31-0500 mwp 
2022-01-05T12:46:31-0500 version: mwp 755768f / 2022-01-05T17:25:07+00:00 5.005.626 on Xlib
2022-01-05T12:46:31-0500 "Ubuntu 20.04.3 LTS" on odyssey running Linux 5.4.0-90-generic x86_64
2022-01-05T12:46:31-0500 hypervisor: none

(mwp:2487): GLib-GIO-ERROR **: 12:46:31.323: Settings schema 'org.mwptools.planner' is not installed
Trace/breakpoint trap
charlesb@odyssey:/usr/src/mwptools/build$ ls /usr/local/share/glib-2.0/schemas
gschemas.compiled  org.mwptools.planner.gschema.xml
charlesb@odyssey:/usr/src/mwptools/build$ 

```

Build/install log
```
charlesb@odyssey:/usr/src/mwptools$ meson build
The Meson build system
Version: 0.53.2
Source dir: /usr/src/mwptools
Build dir: /usr/src/mwptools/build
Build type: native build
Project name: mwptools
Project version: 5.005.626
C compiler for the host machine: cc (gcc 9.3.0 "cc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0")
C linker for the host machine: cc ld.bfd 2.34
Vala compiler for the host machine: valac (valac 0.48.6)
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program meson/post_install.sh found: YES (/usr/src/mwptools/meson/post_install.sh)
Configuring mwp-config.h using configuration
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Run-time dependency gudev-1.0 found: YES 233
Run-time dependency ncurses found: YES 6.2.20200212
Library m found: YES
Program go found: YES (/usr/bin/go)
Run-time dependency gtk+-3.0 found: YES 3.24.20
Run-time dependency atk found: YES 2.35.1
Run-time dependency clutter-1.0 found: YES 1.26.4
Run-time dependency champlain-gtk-0.12 found: YES 0.12.20
Run-time dependency champlain-0.12 found: YES 0.12.20
Run-time dependency libxml-2.0 found: YES 2.9.10
Run-time dependency gio-2.0 found: YES 2.64.6
Run-time dependency clutter-gtk-1.0 found: YES 1.8.4
Run-time dependency gdl-3.0 found: YES 3.34.0
Run-time dependency pango found: YES 1.44.7
Run-time dependency cairo found: YES 1.16.0
Run-time dependency libsoup-2.4 found: YES 2.70.0
Run-time dependency gstreamer-1.0 found: YES 1.16.2
Run-time dependency gstreamer-pbutils-1.0 found: YES 1.16.2
Run-time dependency vte-2.91 found: YES 0.60.3
Library posix found: YES
Library linux found: YES
Library paho-mqtt3cs found: NO
Library mosquitto found: NO
Library espeak-ng found: YES
Library speechd found: NO
Library flite found: YES
Run-time dependency gmodule-2.0 found: YES 2.64.6
Run-time dependency bash-completion found: YES 2.10
Dependency gio-2.0 found: YES 2.64.6 (cached)
Library posix found: YES
Library linux found: YES
Library ncurses found: YES
Dependency gio-2.0 found: YES 2.64.6 (cached)
[... 74 lines cut ...]
Installing /usr/src/mwptools/src/common/mwp_icon.svg to /usr/local/share/icons/hicolor/scalable/apps
Installing /usr/src/mwptools/src/mwp/org.mwptools.planner.gschema.xml to /usr/local/share/glib-2.0/schemas
Installing /usr/src/mwptools/src/mwp/vcols.css to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/default.layout to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/beep-sound.ogg to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/bleet.ogg to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/menubar.ui to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/mwp.ui to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/orange.ogg to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/sat_alert.ogg to /usr/local/share/mwp
Installing /usr/src/mwptools/src/mwp/mwp.desktop to /usr/local/share/applications
Installing /usr/src/mwptools/src/mwp/mwp_complete.sh to /usr/local/share/bash-completion/completions
Installing /usr/src/mwptools/src/mwp/blackbox_decode_complete.sh to /usr/local/share/bash-completion/completions
Installing /usr/src/mwptools/src/samples/area-tool/mwp_area_icon.svg to /usr/local/share/icons/hicolor/scalable/apps
Installing /usr/src/mwptools/src/samples/area-tool/mwp-area-planner.desktop to /usr/local/share/applications
Installing /usr/src/mwptools/src/samples/flash.sh to /usr/local/bin
Installing /usr/src/mwptools/docs/mwptools.pdf to /usr/local/share/doc/mwp
Installing /usr/src/mwptools/docs/debian-ubuntu-dependencies.txt to /usr/local/share/doc/mwp
Installing /usr/src/mwptools/docs/fedora.txt to /usr/local/share/doc/mwp
Installing /usr/src/mwptools/src/bbox-replay/inav_states.rb to /usr/local/bin
Installing /usr/src/mwptools/src/bbox-replay/inav_states_data.rb to /usr/local/bin
Running custom install script '/usr/src/mwptools/meson/post_install.sh'
Compiling gsettings schemas...
Updating desktop icon cache...
```

stronnag, 2022-01-05:
I recommend either `--prefix=/usr` or `--prefix=~/.local`; here you have (by meson default) `--prefix=/usr/local`. 

This means that the mwp specific settings are under `/usr/local/share` which is may not be in data search path.

* remove what you've done and install in a known location; either `--prefix=/usr` or `--prefix=~/.local`, e.g.
    ```
    sudo ninja uninstall
    meson --reconfigure --prefix=$HOME/.local
    ninja install
    ```
 or
* `XDG_DATA_DIR=/usr/loca/share mwp` should also work with the current install.     
 
I thought you were "Mr Local Install", anyway.

wx4cb, 2022-01-05:
i always install "my stuff" ie.. non system apps in /usr/local..... that's just a hang over from the old school days :D not to mention that it's worked fine on other systems as this is how i do all of them :heh:

i'll try the XDG thing, but as far as "mr local install" - i thought you said you was removing that :D

stronnag, 2022-01-05:
> i'll try the XDG thing, but as far as "mr local install" - i thought you said you was removing that :D

It was such a good idea it's the preferred state (mainly due to meson making it sane).

stronnag, 2022-01-05:
btw it's XDG_DATA_DIRS (with a S). mea culpa. Which by default should include `/usr/local/share` so I'm confused too.

wx4cb, 2022-01-11:
weird i reinstalled the OS and it was fine... i dunno...

closed 2022-01-11 by wx4cb