stronnag/mwptools #114: FR: Add SBS Basestation (dump1090) Networked ADSB Support

wx4cb, 2022-01-11:
how easy would it be to incorporate SBS messages ? this would be helful because then you can use anything online that has an SBS format output so that you don't actually need to have an ADSB receiver. You can even use dump1090 to pull it in from any ADSB receiver or even an SDR

http://woodair.net/sbs/article/barebones42_socket_data.htm

(will attach a data dump when i have it)

wx4cb, 2022-01-11:
[dump1090_sbs.csv](https://github.com/stronnag/mwptools/files/7850298/dump1090_sbs.csv)

stronnag, 2022-01-12:
OK, I'll do it.

stronnag, 2022-01-12:
@wx4cb  Could you please record a longer log; it would be useful to have some STA stanzas when aircraft are dropped from the picture.

wx4cb, 2022-01-12:
Sure.. ill do it when i get home

stronnag, 2022-01-12:
[image]

wx4cb, 2022-01-12:
@stronnag  I just got home and are recording some logs now, but if you can find any dump1090 SBS output on the internet you can get them yourself :D 

for example:https://www.adsbhub.org/howtogetdata.php

stronnag, 2022-01-12:
Well, if I had nothing better to do than trawl the internet for files ...

wx4cb, 2022-01-12:
> Well, if I had nothing better to do than trawl the internet for files ...

you mean to say youtube is not your full time job? LOL

wx4cb, 2022-01-12:
[dump1090_sbs.csv](https://github.com/stronnag/mwptools/files/7857144/dump1090_sbs.csv)

This enough data? :dancing_men:

stronnag, 2022-01-12:
Thanks. I'll see how that goes. Strangely, no STA messages. OTOH, the whole thing is a mess. Air position reports with no positions, no speed or course (so I compute that).

Anyway, initial attempt in `development` 
`--radar-device sbs://[host[:port]]`

wx4cb, 2022-01-12:
@stronnag  that could be because right now it's just sitting on my tripod in the office with a crappy antenna.. there's a high likelihood that i'm missing packets and data

stronnag, 2022-01-12:
* Will retry to connect to device every 60 seconds
* Replay tool `src/samples/sbs-test/sbs-player.rb`

stronnag, 2022-01-14:
> @stronnag that could be because right now it's just sitting on my tripod in the office with a crappy antenna.. there's a high likelihood that i'm missing packets and data

More likely that these messages are only emitted by the original Basestation software; we don't need them as contacts will timeout just as PingRX and inav radar does.

wx4cb, 2022-01-14:
[GlobsOfADSBDataForStronnag.zip](https://github.com/stronnag/mwptools/files/7871638/GlobsOfADSBDataForStronnag.zip)

Is this enough data ? :D FYI, it's worldwide data so not sure where it's coming from specifically, but this was just a 20 minute dump from adsbhub.

stronnag, 2022-01-14:
I have enough data thanks

wx4cb, 2022-01-14:
can confirm this appears to be working fine on a live ADSB stream

stronnag, 2022-01-14:
Thanks for the confirmation.

stronnag, 2022-01-14:
btw, ended up with c. 1100 targets logged in the big global file (more in total, as some had timed out).

wx4cb, 2022-01-14:
lol... i dunno i started it, got caught up in a phone call and sat back down and was like... oh crap.... well.... :D

closed 2022-01-14 by stronnag