stronnag/mwptools #115: Make error on ser2udp

ramiss, 2022-01-27:
I'm not used to Linux, anyone able to help me figure out why I can't build ser2udp?  I followed the Windows 11 directions to the letter on a new Windows 11 and WSL install.

Or... can anyone just share the exe?

gcs@RichardsGCS:~/mwptools/mwptools/src/samples/s2n$ make ser2udp.exe
GOOS=windows go build -ldflags "-w -s" -o ser2udp.exe
/bin/sh: 1: go: not found
make: *** [Makefile:9: ser2udp.exe] Error 127

stronnag, 2022-01-27:
You do not have the Go language installed.
`sudo apt install  golang-go `
Anyway, here's a binary.
[ser2udp-x86_64.zip](https://github.com/stronnag/mwptools/files/7948201/ser2udp-x86_64.zip)

stronnag, 2022-01-27:
Archive also  added to release area.

closed 2022-01-27 by stronnag

ramiss, 2022-01-27:
Thanks for this.  Please note I was able to build the exe once I installed Go.
However, I also downloaded the binary provided and defender flagged it as malware.  Defender did not flag the exe I built. Just a note to beware of.

stronnag, 2022-01-27:
Whatever Defender is, it's lying to you.

ramiss, 2022-01-30:
Windows Defender is the built in antivirus for Windows. Even if defender is wrong, why does it flag your exe and not the one I just built?

stronnag, 2022-01-30:
I have no idea. Ask Microsoft.

stronnag, 2022-01-30:
and while you're at it, ask them to explain why their voodoo fails to flag the file when copied to either my win10 or win11 VMs.