stronnag/mwptools #116: TBS Crossfire and rc by MAVLink

ltwin8, 2022-02-15:
following issue:

Inav: ebable MAVLink @115200baud on uart1
CRSF: nano rx: set ch1 and ch2 to MAVLink
Inav: set receiver to MAVLink

now, if no set "rc by MAVLink" in the crossfire, then I can get voltage gps and so on values in mwptools (connected via tcp://192.168.188.132:5760) *BUT* no rc in Inav.

turn on "rc by MAVLink": rc commands in Inav *BUT* no values in mwptools, however still activity on "RX bytes" and "TX bytes"

is this a bug?

background: I would like to use a single uart (forced to, 3 UARTS only, use GPS and DJI) for RC and uploading waypoints from ground station.

thanks for any help

ltwin8, 2022-02-15:
[image]
if this helps, this is the serial terminal, top right one can see data is indeed coming

ltwin8, 2022-02-15:
[image]
here one can see the serial terminal output and top right that data is send

stronnag, 2022-02-15:
Please don't assume that I have any knowledge about what you're trying to do; I don't have any CRSF equipment. So you are going to have to explain very clearly what you wish to achieve.

Some clarifications about mwp:

* It is designed to process a subset of the MAVLink (and the native CRSF) protocols **relevant to the display of INAV telemetry**.
* It will accept any valid MAVLink (to avoid spamming its logs with bogus errors), but only processes the messages it needs to display INAV telemetry data.
* mwp only processes MAVLink for telemetry; uploading missions is MSP.

The screenshot of non-printable data tells me nothing useful. If you wish to capture the serial data seen by mwp, the easiest way is to invoke mwp as:

```
mwp --raw-log
```

You will then find (in the current directory), files named like `mwp._dev_rfcomm0_115200.2021-11-30T081744.raw` (i.e. encoding the capture device and a time stamp). These can be analysed with protocol specific tools under `mwptools/src/samples`. If you post the files here I can look at them.

Please also note that mwp can also [process native CRSF telemetry](https://stronnag.github.io/mwptools/mwp-multi-procotol/#multi-protocol-selection), you don't necessarily have to use MAVLink.

If you have an interesting / compelling use case then I may be interested in supporting it, however I'm also aware of a number of users successfully using mwp with both MAVLink and CRSF without problem.

ltwin8, 2022-02-15:
Thanks for clarification.

first of all: I use MWP with success, but need 2 UARTs every time (MSP and Serial RX) that works as expected.

however the usecase is to use an FC with only one free UART, others occupied by DJI and GPS, but if Waypoints are only possible by MSP and not mavlink i guess it’s not possible.

simplified I would like to use one UART for RC input from Transmitter and to communicate via the TX module (TBS TX WLAN or BT connected to PC) with the groundstation application and have the possibility to set waypoints 

thanks for your help

stronnag, 2022-02-15:
This is then a physical limitation of the FC hardware (unless CRSF has a single UART mode, which I thought it might). mwp only needs one uart (for telemetry, it has nothing to do with serial RX).
In any case:
* mwp only supports MSP for WP transfer
* inav only accepts WPs when unarmed (apart from NAV_GCS aka 'follow me'), which is not supported by mwp.
So this is not something which mwp can solve, closing.

closed 2022-02-15 by stronnag