stronnag/mwptools #117: mwp crashes when window is closed

snaewe, 2022-03-28:
mwp crashes when I just close the window using the "x" icon. Ending via menu or shortcut CTRL-q works as expected.
I'm on Linux Mint 20.3. Here's the terminal output if that helps:

> 2022-03-28T13:27:03+0200 mwp 
2022-03-28T13:27:03+0200 version: mwp 30bd5f4 / 2022-03-27T17:19:02+01:00 5.086.586 on Xlib
2022-03-28T13:27:03+0200 "Linux Mint 20.3" on elitebook running Linux 5.4.0-104-generic x86_64
2022-03-28T13:27:03+0200 hypervisor: none
2022-03-28T13:27:03+0200 MQTT enabled via the "mosquitto" library
2022-03-28T13:27:03+0200 Failed to find "mavlogdump.py" on PATH
2022-03-28T13:27:03+0200 "fl2ltm 0.0.0, commit: local" may be too old, upgrade recommended
2022-03-28T13:27:03+0200 Using speech api 1 [espeak]
2022-03-28T13:27:04+0200 Starting Bing proxy 
2022-03-28T13:27:04+0200 BB load async map zoom : true
2022-03-28T13:27:04+0200 Loaded /usr/local/share/mwp/vcols.css
2022-03-28T13:27:04+0200 Fixed font true
2022-03-28T13:27:04+0200 Delta speed 0.000000
2022-03-28T13:27:04+0200 Host power: 96%, state: Fully Charged (4), warn: None (1)
2022-03-28T13:27:04+0200 mwp will manage power and screen saver / idle
Segmentation fault (core dumped)

stronnag, 2022-03-28:
Perhaps predictably, this cannot be reproduced on a supported Debian derivative OS (Ubuntu LTS, Debian Stable).
If you wish to take this further:
* Run it in `gdb`, provide a backtrace (`bt`) when it crashes. 
```
    meson reconfigure --buildtype=debug
    meson install -C build
    cd build/
    gdb ./mwp
```

snaewe, 2022-03-28:
> #0  0x00007ffff10d04b0 in  () at /lib/x86_64-linux-gnu/libGLX.so.0
#1  0x00007ffff665cfe0 in  () at /lib/x86_64-linux-gnu/libX11.so.6
#2  0x00007ffff665d095 in  () at /lib/x86_64-linux-gnu/libX11.so.6
#3  0x00007ffff665e01d in _XReply () at /lib/x86_64-linux-gnu/libX11.so.6
#4  0x00007ffff66597f1 in XSync () at /lib/x86_64-linux-gnu/libX11.so.6
#5  0x00007ffff101cd8c in  () at /lib/x86_64-linux-gnu/libGLX_mesa.so.0
#6  0x00007ffff101dd47 in  () at /lib/x86_64-linux-gnu/libGLX_mesa.so.0
#7  0x00007ffff1012979 in  () at /lib/x86_64-linux-gnu/libGLX_mesa.so.0
#8  0x00007ffff5cc88a0 in  () at /lib/x86_64-linux-gnu/libcogl.so.20
#9  0x00007ffff5cb7c40 in  () at /lib/x86_64-linux-gnu/libcogl.so.20
#10 0x00007ffff6fb80fb in  () at /lib/x86_64-linux-gnu/libclutter-1.0.so.0
#11 0x00007ffff7028849 in  () at /lib/x86_64-linux-gnu/libclutter-1.0.so.0
#12 0x00007ffff7f50802 in g_closure_invoke () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff7f63f96 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007ffff7f6fbbe in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#15 0x00007ffff7f700f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#16 0x00007ffff6fc67bd in  () at /lib/x86_64-linux-gnu/libclutter-1.0.so.0
#17 0x00007ffff6fc5984 in  () at /lib/x86_64-linux-gnu/libclutter-1.0.so.0
#18 0x00007ffff6f2dbe1 in  () at /lib/x86_64-linux-gnu/libclutter-gtk-1.0.so.0
#19 0x00007ffff7f50965 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007ffff7f6fb48 in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007ffff7f700f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#22 0x00007ffff78abc53 in gtk_widget_unrealize () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#23 0x00007ffff78b02e6 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007ffff7f50965 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#25 0x00007ffff7f6fb48 in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#26 0x00007ffff7f700f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#27 0x00007ffff78abc53 in gtk_widget_unrealize () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#28 0x00007ffff7796b06 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#29 0x00007ffff78b02e6 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#30 0x00007ffff7f50965 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#31 0x00007ffff7f6fb48 in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#32 0x00007ffff7f700f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#33 0x00007ffff78abc53 in gtk_widget_unrealize () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#34 0x00007ffff7643550 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#35 0x00007ffff78b02e6 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#36 0x00007ffff7f50965 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#37 0x00007ffff7f6fb48 in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#38 0x00007ffff7f700f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#39 0x00007ffff78abc53 in gtk_widget_unrealize () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#40 0x00007ffff78baa79 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#41 0x00007ffff78b02e6 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#42 0x00007ffff78bd61e in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#43 0x00007ffff7f50802 in g_closure_invoke () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#44 0x00007ffff7f63f96 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#45 0x00007ffff7f6fbbe in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#46 0x00007ffff7f700f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#47 0x00007ffff78abc53 in gtk_widget_unrealize () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#48 0x00007ffff78ae5d8 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#49 0x00007ffff78c227c in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#50 0x00007ffff763591f in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#51 0x00007ffff7f574d1 in g_object_run_dispose () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#52 0x00007ffff7762149 in gtk_main_do_event () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#53 0x00007ffff78bbecb in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#54 0x00007ffff743ef4d in  () at /lib/x86_64-linux-gnu/libgdk-3.so.0
#55 0x00007ffff7e6504e in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#56 0x00007ffff7e65400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#57 0x00007ffff7e654a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#58 0x00007ffff71c2fe5 in g_application_run () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#59 0x000055555560079f in mwp_main (args=0x7fffffffe338, args_length1=1) at ../src/mwp/mwp.vala:10987
        _tmp0_ = CLUTTER_INIT_SUCCESS
        sb = 0x55555596f960
        _tmp1_ = 0x55555596f960
        _tmp6_ = 0x55555596f960
        _tmp7_ = 0x555555a3c780 "/home/stefan/src/mwptools/build/mwp "
        _tmp8_ = 0x5555558976b0 "\001"
        app = 0x555555838080
        _tmp9_ = 0x555555838080
        _tmp10_ = 0x555555838080
        result = 0
#60 0x0000555555600806 in main (argc=1, argv=0x7fffffffe338) at ../src/mwp/mwp.vala:10974

stronnag, 2022-03-28:
So it dies deep in Mint's OpenGL libraries. 
Mint is not a supported environment; so it's a WONTFIX.

closed 2022-03-28 by stronnag