stronnag/mwptools #120: Cygwin go and ncurses not found

PWN-GH, 2022-04-24:
Followed https://github.com/stronnag/mwptools/wiki/mwp-on-cygwin exactly command by command, got stuck on the last command of https://github.com/stronnag/mwptools/wiki/mwp-on-cygwin#mwp-on-cygwin.
log:
[meson-log.txt](https://github.com/stronnag/mwptools/files/8549761/meson-log.txt)

User@DESKTOP-NK2MISL ~/mwptools
$ meson build --buildtype=release --strip --prefix /usr
The Meson build system
Version: 0.60.3
Source dir: /home/User/mwptools
Build dir: /home/User/mwptools/build
Build type: native build
Project name: mwptools
Project version: 5.101.449
C compiler for the host machine: cc (gcc 11.2.0 "cc (GCC) 11.2.0")
C linker for the host machine: cc ld.bfd 2.37
Vala compiler for the host machine: valac (valac 0.38.8)
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program meson/post_install.sh found: YES (/home/User/mwptools/meson/post_install.sh)
Configuring mwp-config.h using configuration
Found pkg-config: /usr/bin/pkg-config (1.8.0)
Found CMake: /cygdrive/c/Program Files/CMake/bin/cmake (3.23.0)
WARNING: CMake Toolchain: Failed to determine CMake compilers state
Run-time dependency ncurses found: NO (tried pkgconfig)
Library m found: YES
Program go found: NO

meson/meson.build:51:0: ERROR: Program 'go' not found or not executable

A full log can be found at /home/User/mwptools/build/meson-logs/meson-log.txt

stronnag, 2022-04-24:
Kudos on even trying to build on (unsupported) cygwin. 
I'll see what I can do. It built recently on my cygwin.

stronnag, 2022-04-24:
I remember now; meson 0.60 changed the default for `find_program` from silently ignore fail to fatal error.
Fixed in `development` branch for now.

PWN-GH, 2022-04-24:
Thank you for your help. I got past that one.

Now I am stuck on here though:

User@DESKTOP-NK2MISL ~/mwptools/build
$ ninja && ninja install
ninja: error: loading 'build.ninja': No such file or directory

User@DESKTOP-NK2MISL ~/mwptools/build
$ ls
meson  meson-info  meson-logs  meson-private  mwp-config.h  src

New log: 
User@DESKTOP-NK2MISL ~/mwptools
$ meson build --buildtype=release --strip --prefix /usr
The Meson build system
Version: 0.60.3
Source dir: /home/User/mwptools
Build dir: /home/User/mwptools/build
Build type: native build
Project name: mwptools
Project version: 5.114.486
C compiler for the host machine: cc (gcc 11.2.0 "cc (GCC) 11.2.0")
C linker for the host machine: cc ld.bfd 2.37
Vala compiler for the host machine: valac (valac 0.38.8)
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program meson/post_install.sh found: YES (/home/User/mwptools/meson/post_install.sh)
Configuring mwp-config.h using configuration
Found pkg-config: /usr/bin/pkg-config (1.8.0)
Dependency glib-2.0 found: NO found 2.54.3 but need: '>=2.66'
Found CMake: /cygdrive/c/Program Files/CMake/bin/cmake (3.23.0)
WARNING: CMake Toolchain: Failed to determine CMake compilers state
Run-time dependency glib-2.0 found: NO (tried pkgconfig)
Message: using fallback URI parser
WARNING: CMake Toolchain: Failed to determine CMake compilers state
Run-time dependency ncurses found: NO (tried pkgconfig)
Library m found: YES
Program go found: NO
Run-time dependency gtk+-3.0 found: YES 3.22.28
Run-time dependency atk found: YES 2.26.1
Run-time dependency clutter-1.0 found: YES 1.26.2
Run-time dependency champlain-gtk-0.12 found: YES 0.12.16
Run-time dependency champlain-0.12 found: YES 0.12.16
Run-time dependency libxml-2.0 found: YES 2.9.12
Run-time dependency gio-2.0 found: YES 2.54.3
Run-time dependency glib-2.0 found: YES 2.54.3
Run-time dependency clutter-gtk-1.0 found: YES 1.8.4
Run-time dependency gdl-3.0 found: YES 3.26.0
Run-time dependency pango found: YES 1.40.14
Run-time dependency cairo found: YES 1.17.4
Run-time dependency libsoup-2.4 found: YES 2.60.3
Run-time dependency gstreamer-1.0 found: YES 1.12.5
WARNING: CMake Toolchain: Failed to determine CMake compilers state
Run-time dependency gstreamer-pbutils-1.0 found: NO (tried pkgconfig)

src/mwp/meson.build:21:0: ERROR: Dependency "gstreamer-pbutils-1.0" not found, tried pkgconfig

A full log can be found at /home/User/mwptools/build/meson-logs/meson-log.txt
[meson-log.txt](https://github.com/stronnag/mwptools/files/8549878/meson-log.txt)

if it will help other people: 
-wget needs to be installed in the cygwin installer by changing the View from "Pending" to "Full", searching for "get", then changing the New option from "Keep" to a version(or double clicking the word Keep).
-use --branch development in the git command to get the development branch instead of master

stronnag, 2022-04-24:
see meson setup in the INSTALL file in the mwptools root directory

eli-schwartz, 2022-04-24:
> I remember now; meson 0.60 changed the default for `find_program` from silently ignore fail to fatal error.

It has defaulted to `required: true` since version 0.4.0: https://github.com/mesonbuild/meson/commit/c4e58795e77562815b0204d000b774c716826e71

PWN-GH, 2022-04-24:
Found it, also needed "libgstinterfaces1.0-devel" from the cygwin installer to build

PWN-GH, 2022-04-24:
It runs now but several things aren't working so I will revert to linux.
[image]

PWN-GH, 2022-04-24:
Last log for more info if someone wants to take a look at it:

User@DESKTOP-NK2MISL ~/mwptools/build
$ ./mwp.exe
libGL error: Windows-DRI extension not available
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
libGL error: Windows-DRI extension not available
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
2022-04-24T22:30:28+0700 ./mwp
2022-04-24T22:30:28+0700 version: mwp 292a2b84 / 2022-04-24T14:32:02+01:00 5.114.486
2022-04-24T22:30:28+0700 DESKTOP-NK2MISL running CYGWIN_NT-10.0-19044 3.3.4-341.x86_64 x86_64
2022-04-24T22:30:28+0700 WM: Xlib / Unknown DE
2022-04-24T22:30:28+0700 GL: Unknown / Unsupported
2022-04-24T22:30:28+0700 hypervisor: none
2022-04-24T22:30:28+0700 Failed to find "blackbox_decode" on PATH
2022-04-24T22:30:28+0700 Failed to find "mwp-plot-elevations" on PATH
2022-04-24T22:30:28+0700 Failed to find "fl2ltm" on PATH
2022-04-24T22:30:28+0700 Failed to find "mavlogdump.py" on PATH
2022-04-24T22:30:28+0700 Using external speech api [espeak -v en]
2022-04-24T22:30:31+0700 Starting Bing proxy
2022-04-24T22:30:31+0700 BB load async map zoom : true
2022-04-24T22:30:31+0700 Loaded /home/User/.local/share/mwp/vcols.css
2022-04-24T22:30:31+0700 Fixed font true
2022-04-24T22:30:31+0700 gsettings blank default-map, using osm-mapnik
2022-04-24T22:30:31+0700 Delta speed 0.000000
2022-04-24T22:30:32+0700 UPower: Could not connect: No such file or directory
2022-04-24T22:30:41+0700 Try connect /dev/ttyS22
2022-04-24T22:30:41+0700 Connected /dev/ttyS22 false
2022-04-24T22:30:41+0700 Serial ready
2022-04-24T22:30:41+0700 set mrtype=3 cap =3c
2022-04-24T22:30:41+0700 Using MSP v2 0204
2022-04-24T22:30:41+0700 Model name: ""
2022-04-24T22:30:41+0700 V2 mixer 3 0
2022-04-24T22:30:41+0700 Feature Mask [00480004] : telemetry false, gps false, current false
2022-04-24T22:30:42+0700 set TZ offset 420
2022-04-24T22:30:42+0700 INAV v4.1.0  REVO (611d26f4)
2022-04-24T22:30:42+0700 Generate navconf b
2022-04-24T22:30:42+0700 Masks arm 1 angle 4 horz 8 ph 0 rth 0 wp 0
2022-04-24T22:30:42+0700 WP_GETINFO: 120/0/0
2022-04-24T22:30:42+0700 Activeboxes 8 00008000 00000000
2022-04-24T22:30:42+0700 Requesting common settings
2022-04-24T22:30:42+0700 Received (raw) nav_wp_safe_distance 10000
2022-04-24T22:30:42+0700 Received (raw) inav_max_eph_epv 1000
2022-04-24T22:30:42+0700 Received gps_min_sats 6
2022-04-24T22:30:42+0700 MSP Error: MSP_CMDS_COMMON_SETTING[4099] nav_rth_home_offset_distance
2022-04-24T22:30:42+0700 Arming flags: RC Link (40000), load 9% MSP_CMDS_INAV_STATUS
2022-04-24T22:30:42+0700 Sensors: Acc  (0081)
2022-04-24T22:30:42+0700 INAV v4.1.0  REVO (611d26f4) API 2.4 Nav Pr 0 QUADX
2022-04-24T22:30:42+0700 switch val == 00008000 (0000000c)
2022-04-24T22:30:42+0700 message => No GPS/BARO/MAG detected

(mwp:1933): GStreamer-CRITICAL **: gst_element_get_bus: assertion 'GST_IS_ELEMENT (element)' failed

(mwp:1933): GStreamer-CRITICAL **: gst_bus_add_watch_full: assertion 'GST_IS_BUS (bus)' failed

(mwp:1933): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed

(mwp:1933): GStreamer-CRITICAL **: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
2022-04-24T22:30:42+0700 no GPS/BARO/MAG, sensor = 0x81
2022-04-24T22:30:42+0700 Timer cycle for 3 items, 19 => 85 bytes
2022-04-24T22:30:42+0700 Start poller

closed 2022-04-24 by PWN-GH

stronnag, 2022-04-24:
FWIW, working example (qemu/lkvm rather than bare metal)

* Audio OK
* Map working

But it's a pretty sad user experience, the only usable Windows way is wsl2g on Win11. Win10 wsl is possible, but you're still at the mercy 3rd party X server.

```
16:47:36 mwp --rings 50,20
16:47:36 version: mwp bab3c4c / 2022-04-24T15:14:41+01:00 5.114.512
16:47:36 win10box running CYGWIN_NT-10.0-19044 3.3.4-341.x86_64 x86_64
16:47:36 WM: Xlib / Unknown DE
16:47:36 GL: Unknown / Unsupported
16:47:37 hypervisor: none
16:47:37 Failed to find "mavlogdump.py" on PATH
16:47:37 Using fl2ltm.exe 1.0.1 commit:c83569f
16:47:37 Using external speech api [espeak -v en]
16:47:38 Starting Bing proxy
16:47:39 External proxy "gmproxy" listening on :49736
16:47:39 External proxy "bproxy" listening on :49737
16:47:39 BB load async map zoom : true
16:47:39 Loaded /home/win10/.config/mwp/vcols.css
16:47:39 Fixed font true
16:47:39 Delta speed 0.000000
16:47:40 UPower: Could not connect: No such file or directory
16:47:40 mwp will manage power and screen saver / idle
16:48:11 Getting base location from index 1 xx.xxxxxx -xx.xxxxxx true
16:48:12 Geonames xx.xxxxxx -xx.xxxxxx America/New_York
16:48:33 Replay "bbox" log /home/win10/LOG00031.TXT model -1
16:48:34 fl2ltm -device udp://localhost:60840 --index 1 --type -1 -interval 141 C:\cygwin64\home\win10\LOG00031.TXT  # pid=891
16:48:34 set mrtype=8 cap =0
16:48:34 Using MSP v2 0200
16:48:34 set TZ offset 60
16:48:34 INAV v2.6.1  OMNIBUSF4PRO (5510259)
16:48:34 Model name: ""
16:48:34 Arming flags: Ready to Arm (0000), load 0% MSP_CMDS_INAV_STATUS
16:48:35 Sensors: Acc Baro GPS  (000b)
16:48:35 INAV v2.6.1  OMNIBUSF4PRO (5510259) API 2.0 Nav Pr 0 FLYING_WING
16:48:35 init icon 8 0 8 (false)
16:48:35 Timer cycle for 9 items, 55 => 178 bytes
16:48:35 Craft is armed, special=0
16:48:35 Enable nav speak
16:48:35 LTM telemetry
16:48:35 Managing screen idle and suspend
16:48:35 New LTM Mode Launch (20) 1 0s 0.000000 0.000000 0 0
16:48:35 Set home xx.xxxxxx -xx.xxxxxx (LTM OFrame true)
16:48:46 New LTM Mode Horizon (3) 1 8s xx.xxxxxx -xx.xxxxxx 0 0
16:48:58 New LTM Mode Return Home (13) 1 18s xx.xxxxxx -xx.xxxxxx 0 4
16:49:24 New LTM Mode GPS Hold (9) 1 37s xx.xxxxxx -xx.xxxxxx 0 2
16:49:24 Set poshold xx.xxxxxx -xx.xxxxxx
16:49:24 New LTM Mode Horizon (3) 1 37s xx.xxxxxx -xx.xxxxxx 0 0
16:49:28 New LTM Mode Cruise (18) 1 39s xx.xxxxxx -xx.xxxxxx 0 20
16:49:28 New LTM Mode Waypoints (10) 1 40s xx.xxxxxx -xx.xxxxxx 0 8
16:49:39 ============== Replay complete ====================
```