stronnag/mwptools #123: Flash.sh usage

stuartofmt, 2022-05-28:
Hi --
Can you explain the usage of Flash.sh.  I've looked at the code and cannot see how the variable P is being initially set or the intended modifications to make the script work.  Specifically: 
`
For P
do
<snip>
done
`

Wondering if you can explain?
As an example BrainFPV uses hex files and shows up as a `[0483:df11]` device on /dev/ttyACM0

**Some background:**

I'm using iNav on Windows with a BrainFPV Radix Li.  It refuses to flash from Windows but flashes reliably from a Mac (which I have to borrow).  I've tried all the permutations of Zadig and ImpulseRC etc.

I have several Raspberry Pi's (but of course getting iNav Configurator to compile / work is problematic).  So I thought I'd just  use dfu-util and write a short bash or python script to do the flashing.  It's only occasional use so it does not have to be fancy.
In doing my homework - I stumbled on Flash.sh and thought that your efforts will save my effort :-)

stronnag, 2022-05-28:
I hoped it was adequately documented [in the manual](https://stronnag.github.io/mwptools/mwp-miscellaneous-tools/#flashsh-fcflash)

Let me know if the manual is unclear.

stronnag, 2022-05-28:
As to you other question, it's idiomatic bash. It reads the supplied arguments.
It is not my intention to teach basic Unix scripting, there are plenty of tutorials on the internet for that.

stuartofmt, 2022-05-28:
Many thanks -- The manual answers my questions.  I had looked but somehow the manual evaded me.
On the bash aspect - again thanks, I was unaware of the technique you used.  It was not my intent to request tutoring :-)

closed 2022-05-28 by stronnag

stronnag, 2022-05-28:
And just for the record, works fine on my Rpi4, as long as the required dependencies are installed (which in future the script should check)

```
[jrh@woozle]$ fcflash  /tmp/inav_5.0.0_WINGFC_no-lto.hex 
9600
DFU: dfu-util -d 0483:df11 --alt 0 -s 0x08000000:force:leave -D /tmp/inav_5.0.0_WINGFC_no-lto.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 537512
Erase   	[=========================] 100%       537512 bytes
Erase    done.
Download	[=========================] 100%       537512 bytes
Download done.
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state
[jrh@woozle]$ uname -a
Linux woozle 5.15.40-3-rpi-ARCH #1 SMP PREEMPT Fri May 20 17:25:45 UTC 2022 aarch64 GNU/Linux
```

stuartofmt, 2022-05-28:
Works perfectly on Pi 3B+

`Linux weatherdog 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`

using

`./flash.sh erase /dev/ttyACM0 ,<firmware>.hex `

THANKS !!!