stronnag/mwptools #129: Multi Mission planning is broken

b14ckyy, 2022-12-14:
If I plan a waypoint mission and then add a second one, the first mission stays on screen. 
Any further added waypoint is then just continuing the existing ones.

When I switch back to Mission 1, only the first mission is shown. Switching to 2 then shows all waypoints of both planned missions. 
After loading the missions to the FC they are also stored exactly that way. Mission 1 has 5 waypoints for example but Mission 2 then has 10 waypoints. 

I can delete the waypoints 1-5 in mission 2 initially but when I try to switch back to Mission 1, the application crashes. 

Seen on Versions: 5.295.580 4d6e772 and current master 5.311.575 c3008c0
OS: WSL2 Windows 11
FC Firmware independent

[image]
[image]

INAV Configurator shows them as expected. Either ALL or Single mission: 
[image]
[image]

stronnag, 2022-12-14:
Confirmed . Thanks

b14ckyy, 2022-12-14:
btw just one minor thing. 
When "Edit Waypoints" is active and I pan through the map, every time I release the left mouse button it creates a waypoint. 
its just a minor inconvenience but while you're at it, maybe this can be fixed as well?

stronnag, 2022-12-14:
Should be fixed in master (and development) branches. I'll maybe build a `.deb` later.

b14ckyy, 2022-12-14:
Great! Multi Mission switching works now properly and the WP creation during mouse panning is also fixed. 
One issue remains though: The crash when deleting multiple waypoints at the beginning of the mission. Does not seem to be related to multi mission. 

Creating 10 waypoints in a row. Then select WP 1-5 in the WP list. Right click and delete selection. Application crashes. 
It just says "Segmentation Fault" in the shell. Nothing in the mwp_stderr_2022-12-14.txt

stronnag, 2022-12-14:
I can't reproduce this crash on current master (but I did see it on an older build).

stronnag, 2022-12-14:
There is no more useful logging for this case. I'll try on a Win11 VM when I can evict Mrs Stonnag from the desktop computer.

b14ckyy, 2022-12-14:
No Rush. Right now I can work around that. Just needed multi mission with Useraction support for a test and INAV Mission Control does not support Useractions yet. So thanks a lot for your quick response.

stronnag, 2022-12-14:
Same crash here on win11/WSL-G. Next up native Debian / Ubuntu / FreeBSD.

stronnag, 2022-12-14:
> I place a bet on Ubuntu as well.

Hope so, then there is a possibility to see why

stronnag, 2022-12-14:
Reworked the way bulk delete works. Please try the latest master (or development). I could get it to crash on Ub 22.04 after multiple bulk-deletes. So it's now simpler ...

b14ckyy, 2022-12-14:
Perfect! All working now. crated 50 waypoints and deleted the first 20 with no issue. 
Thanks a lot mate!

closed 2022-12-14 by stronnag