stronnag/mwptools #131: Follow doesn't follow

timtuxworth, 2023-01-29:
I just installed mwptools on a fresh install of Ubuntu 22.04 and connected my TX16 S running Open TX 2.3.15. I enabled Telemetry forwarding and wmptools can see the telemetry and displays vehicle attitude int he HUD and  the GPS location in the applets, but the map doesn't move/reposition, even though the "follow" option is checked. I tried checking/checking the box.

stronnag, 2023-01-29:
`Follow` means that the craft location is updated with a valid GPS fix
`In View` means the map will be panned to keep `Follow` in view
`use-legacy-centre-on` keeps the vehicle centred and moves the map, which is more CPU intensive (and may induce motion sickness in the observer).

If you still can't set this to work, a dump of the settings (`gsettings list-recursively org.mwptools.planner`) and a recording of the telemetry (`mwp --raw-log`) would help.

timtuxworth, 2023-01-31:
When map starts up it's centered in Europe somewhere. My location is in Western Canada. I'm just doing this on the bench atm, but the map doesn't reposition to where I am, even though I have both Follow and In View checked. I tried mwp --use-legacy-center-on but there is no such option.
[image]

snaewe, 2023-01-31:
It's called 'use-legacy-centre-on' (yes, centre not center).

timtuxworth, 2023-01-31:
[image]

stronnag, 2023-01-31:
Please see the preferences for setting a default location.
`use-legacy-centre-on` is a setting not a option.
It is also spelled in English.

As you're not supplied the requested data, I shall close this.

timtuxworth, 2023-01-31:
These are the logs 
[mwp_2023-01-31_105342.log](https://github.com/stronnag/mwptools/files/10549401/mwp_2023-01-31_105342.log)

https://www.dropbox.com/s/xip7vicdcjikme3/mwp._dev_ttyACM0.2023-01-31T105323.raw?dl=0