stronnag/mwptools #133: [Feature Request] : Support AP flavour of CRSF telemetry

stronnag, 2023-01-31:
### Discussed in https://github.com/stronnag/mwptools/discussions/132

<div type='discussions-op-text'>

<sup>Originally posted by **timtuxworth** January 29, 2023</sup>
I just installed mwptools on a fresh install of Ubuntu 22.04 and connected my TX16 S running Open TX 2.3.15. I enabled Telemetry forwarding on the transmitter and wmptools can see the telemetry and displays vehicle attitude in the HUD and  the GPS location in the applets, but the map doesn't move/reposition, even though the "follow" option is checked. I tried checking/unchecking the box.
</div>

stronnag, 2023-01-31:
From #132, the user is using AP which is not supported firmware, however it might be interesting to support it. We'll see.

stronnag, 2023-01-31:
Mapping FBWA to Angle (and FBWB to Horizon) means we get a location on the map.
(via  [ap-crsf](https://github.com/stronnag/mwptools/tree/ap-crsf) branch).
More mappings to follow.

@timtuxworth , if you connect a battery it would be useful to know if the voltage is reported correctly. And if you move the model, does the AHI move?

The faint line to (0,0) is because we have (alas) ` 8.56s: GPS: 0.000000 0.000000 0m 0.0m/s 0.0° 5 sats`

[image]

stronnag, 2023-01-31:
all AP modes should now be handled (perhaps not correctly ....)

timtuxworth, 2023-01-31:
I'll rebuild and test in a bit. Not sure if you got the "Q" options for QuadPlane/VTOL planes - in full they are here:
https://ardupilot.org/plane/docs/quadplane-flight-modes.html
Specifically:
[QSTABILIZE (Mode 17)](https://ardupilot.org/plane/docs/qstabilize-mode.html)
[QHOVER (Mode 18)](https://ardupilot.org/plane/docs/qhover-mode.html)
[QLOITER (Mode 19)](https://ardupilot.org/plane/docs/qloiter-mode.html)
[QLAND (Mode 20)](https://ardupilot.org/plane/docs/qland-mode.html)
[QRTL (Mode 21)](https://ardupilot.org/plane/docs/qrtl-mode.html)
[QAUTOTUNE (Mode 22)](https://ardupilot.org/plane/docs/qautotune-mode.html)
[QACRO (Mode 23)](https://ardupilot.org/plane/docs/qacro-mode.html)
[LOITER to QLAND (mode 25)](https://ardupilot.org/plane/docs/loiter-to-qland-mode.html)

If its a multi-rotor (drone) - then there are a bunch of modes, but I don't have one of these to test:
https://ardupilot.org/copter/docs/flight-modes.html

stronnag, 2023-01-31:
That's not a huge amount of help, as abbreviations will be used for the CRSF `FM` parameter. So I probably have most of them, but not all the MR items. Unless someone want's to volunteer a list of all AP flight modes mapped to the INAV modes exposed over CRSF, that's as good as it's likely to get for now; AP is not a priority.

timtuxworth, 2023-01-31:
No worries. FYI - yes the artificial horizon moves. It worked great from the first time I connected. I'll do some more testing later, including the battery thing.

stronnag, 2023-01-31:
Cool. I'll do the MR stuff sometime.

timtuxworth, 2023-02-02:
I just connected a different plane with a battery monitor to get the battery. This works great, but the location doesn't display. Here is a raw-log.
[image]
https://www.dropbox.com/s/g16bbpezedsria9/mwp._dev_ttyACM0.2023-02-01T200833.raw?dl=0

stronnag, 2023-02-02:
Here, the log shows the location (and a "Home" icon with > 6 satellites).

* Location is shown
* At c.30s, The mode icon briefly flashs green (POSHOLD) due to `FM: CIRC`
* A bit later, the AHI moves, presumably the model was moved
* Accumulated current consumption is shown
* Elapsed time is shown (0:47s)
* The only modes in the supplied log are `FBWA`, `FBWB` and `CIRC`. Your image shows `Manual` (aka `FM: MAN` )

[image]

Do you still have the mwp console log `~/mwp_stderr_2023-02-01.txt`? If so, can I see it please?

stronnag, 2023-02-02:
OTOH, if I play your log into a build off `master` branch (vice `ap-crsf` branch); then guess what, I see the same as you, including "Manual".

timtuxworth, 2023-02-02:
Here's that file. I did build from ap-crsf. 
[mwp_stderr_2023-02-01.txt](https://github.com/stronnag/mwptools/files/10570206/mwp_stderr_2023-02-01.txt)

timtuxworth, 2023-02-02:
Does this help: 
```
mwp --version
libGL error: pci id for fd 11: 1ab8:0010, driver (null)
pci id for fd 12: 1ab8:0010, driver (null)
6.019.690
```

timtuxworth, 2023-02-02:
Just a suggestion - the flight mode is very tiny. On the field, I'd like to see it nice and big, maybe a dock item (or a field in one of the existing dock items).

stronnag, 2023-02-02:
The track colour  (and sometimes an icon) indicate the flight mode. For use in the field, the audio output is by far the most useful as one is either wearing googles or looking at the vehicle.
[image]

For the direction, this is reported correctly by INAV ...

timtuxworth, 2023-02-02:
Thanks Jonathan, this is looking great, I'm just about ready to try to fly it - if all goes well that might be on Saturday. Yes I usually connect a Bluetooth headset to my groundstation so I can listen to audio even if I get too far from it.

stronnag, 2023-02-02:
Good luck. Please record a raw log, I'd love to see what happens in the air with AP (in particular the direction).
If you haven't seen it the myriad of configuration settings is described in the [manual]( https://stronnag.github.io/mwptools/mwp-Configuration/)  . Bit of a legacy mis-mash of command  line options (which can be persisted in a file) and the `gsettings` registry.

stronnag, 2023-02-02:
And I'll merge the `ap-crsf` branch into master this evening.

closed 2023-04-29 by stronnag