stronnag/mwptools #150: Windows 11 (WSL-G) Installation

Jullibach, 2023-07-16:
Hi, I tried to install mwptools on a clean ubuntu installation (windows 11 subsystem).
I tried to install it by downloading the .deb to /home/user/mwp (I created the /mwp by mkdir).
The download appears to be working normally.
If I run [apt install] ubuntu wants to download 270mb new packages so I agreed with y.
The Download is fine until:

......
Setting up libchamplain-gtk-0.12-0:amd64 (0.12.20-1build1) ...
Setting up mwptools (6.159.538) ...
Processing triggers for libgdk-pixbuf-2.0-0:amd64 (2.42.8+dfsg-1ubuntu0.2) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
N: Download is performed unsandboxed as root as file '/home/user/mwp/mwptools_6.159.538_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
......

If I then try to run the GUI by typing [mwp] I dont get an error but the console goes to a new line without any advise:
.......
Setting up mwptools (6.159.538) ...
Processing triggers for libgdk-pixbuf-2.0-0:amd64 (2.42.8+dfsg-1ubuntu0.2) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
N: Download is performed unsandboxed as root as file '/home/user/mwp/mwptools_6.159.538_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
user@Jullibach-Laptop:~/mwp$ mwp
user@Jullibach-Laptop:~/mwp$
......

Now I dont know how to proceed with this error.
OS: Windows 11 Home 22621.1992  x64

Complete Log:
user@Jullibach-Laptop:~$ sudo apt update
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease
Get:3 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]
Get:4 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [572 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [108 kB]
Get:6 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [14.1 MB]
Get:7 http://security.ubuntu.com/ubuntu jammy-security/main Translation-en [140 kB]
Get:8 http://security.ubuntu.com/ubuntu jammy-security/main amd64 c-n-f Metadata [11.0 kB]
Get:9 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [558 kB]
Get:10 http://security.ubuntu.com/ubuntu jammy-security/restricted Translation-en [87.2 kB]
Get:11 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 c-n-f Metadata [532 B]
Get:12 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [748 kB]
Get:13 http://security.ubuntu.com/ubuntu jammy-security/universe Translation-en [133 kB]
Get:14 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 c-n-f Metadata [15.7 kB]
Get:15 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [36.5 kB]
Get:16 http://security.ubuntu.com/ubuntu jammy-security/multiverse Translation-en [7060 B]
Get:17 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 c-n-f Metadata [260 B]
Get:18 http://archive.ubuntu.com/ubuntu jammy/universe Translation-en [5652 kB]
Get:19 http://archive.ubuntu.com/ubuntu jammy/universe amd64 c-n-f Metadata [286 kB]
Get:20 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [217 kB]
Get:21 http://archive.ubuntu.com/ubuntu jammy/multiverse Translation-en [112 kB]
Get:22 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 c-n-f Metadata [8372 B]
Get:23 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [792 kB]
Get:24 http://archive.ubuntu.com/ubuntu jammy-updates/main Translation-en [200 kB]
Get:25 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 c-n-f Metadata [15.4 kB]
Get:26 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [567 kB]
Get:27 http://archive.ubuntu.com/ubuntu jammy-updates/restricted Translation-en [88.2 kB]
Get:28 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 c-n-f Metadata [528 B]
Get:29 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [943 kB]
Get:30 http://archive.ubuntu.com/ubuntu jammy-updates/universe Translation-en [202 kB]
Get:31 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 c-n-f Metadata [20.7 kB]
Get:32 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [41.6 kB]
Get:33 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse Translation-en [9764 B]
Get:34 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 c-n-f Metadata [476 B]
Get:35 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [40.9 kB]
Get:36 http://archive.ubuntu.com/ubuntu jammy-backports/main Translation-en [10.2 kB]
Get:37 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 c-n-f Metadata [388 B]
Get:38 http://archive.ubuntu.com/ubuntu jammy-backports/restricted amd64 c-n-f Metadata [116 B]
Get:39 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [22.2 kB]
Get:40 http://archive.ubuntu.com/ubuntu jammy-backports/universe Translation-en [15.4 kB]
Get:41 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 c-n-f Metadata [580 B]
[... 256 lines cut ...]
update-alternatives: using /usr/share/icons/Adwaita/cursor.theme to provide /usr/share/icons/default/index.theme (x-cursor-theme) in auto mode
Setting up humanity-icon-theme (0.6.16) ...
Setting up ubuntu-mono (20.10-0ubuntu2) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for shared-mime-info (2.1-2) ...
Processing triggers for udev (249.11-0ubuntu3.9) ...
Processing triggers for libglib2.0-0:amd64 (2.72.4-0ubuntu2.2) ...
Setting up libgtk-3-0:amd64 (3.24.33-1ubuntu2) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
Setting up libgtk-3-bin (3.24.33-1ubuntu2) ...
Setting up libvte-2.91-0:amd64 (0.68.0-1) ...
Setting up libgdl-3-5:amd64 (3.40.0-1) ...
Setting up at-spi2-core (2.44.0-3) ...
Setting up glib-networking:amd64 (2.72.0-1) ...
Setting up libclutter-1.0-0:amd64 (1.26.4+dfsg-4build1) ...
Setting up libclutter-gtk-1.0-0:amd64 (1.8.4-4build2) ...
Setting up libsoup2.4-1:amd64 (2.74.2-3) ...
Setting up libchamplain-0.12-0:amd64 (0.12.20-1build1) ...
Setting up libchamplain-gtk-0.12-0:amd64 (0.12.20-1build1) ...
Setting up mwptools (6.159.538) ...
Processing triggers for libgdk-pixbuf-2.0-0:amd64 (2.42.8+dfsg-1ubuntu0.2) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
N: Download is performed unsandboxed as root as file '/home/user/mwp/mwptools_6.159.538_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
user@Jullibach-Laptop:~/mwp$ mwp
user@Jullibach-Laptop:~/mwp$

stronnag, 2023-07-16:
Works for me (as well as anything works in Windows)

```
jrh@win11:~$ uname -a
Linux win11 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
jrh@win11:~$ sudo apt install /tmp/mwptools_6.159.538_amd64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'mwptools' instead of '/tmp/mwptools_6.159.538_amd64.deb'
The following packages will be upgraded:
  mwptools
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/3031 kB of archives.
After this operation, 3873 kB of additional disk space will be used.
Get:1 /tmp/mwptools_6.159.538_amd64.deb mwptools amd64 6.159.538 [3031 kB]
(Reading database ... 95729 files and directories currently installed.)
Preparing to unpack .../mwptools_6.159.538_amd64.deb ...
Unpacking mwptools (6.159.538) over (6.036.386) ...
Setting up mwptools (6.159.538) ...
jrh@win11:~$ mwp -v
6.159.538
jrh@win11:~$ mwp
2023-07-16T22:23:03+0100 mwp --rings 50,20 -S 8192 -t 8
2023-07-16T22:23:03+0100 buildinfo: 2023-06-10T10:59:13Z Linux 6.3.7-zen1-1-zen x86_64 bullseye
2023-07-16T22:23:03+0100 toolinfo: gcc version 10.2.1 20210110 (Debian 10.2.1-6)  / GNU ld (GNU Binutils for Debian) 2.35.2 / Vala 0.48.17
2023-07-16T22:23:03+0100 version: mwp fc19efa9 (development) / 2023-06-08T15:57:41+01:00 6.159.538
2023-07-16T22:23:03+0100 "Ubuntu 22.04.2 LTS" on win11 running Linux 5.15.90.1-microsoft-standard-WSL2 x86_64
2023-07-16T22:23:03+0100 WM: Wayland / Unknown DE
2023-07-16T22:23:03+0100 GL: Mesa/X.org / llvmpipe (LLVM 15.0.7, 256 bits)
2023-07-16T22:23:04+0100 hypervisor: wsl
2023-07-16T22:23:04+0100 GLib2 2.72
2023-07-16T22:23:04+0100 MQTT enabled via the "mosquitto" library
2023-07-16T22:23:04+0100 Failed to find "mavlogdump.py" on PATH
2023-07-16T22:23:05+0100 Using fl2ltm 1.0.3 commit:6ac4fcf
2023-07-16T22:23:05+0100 Using speech api 3 [flite]
2023-07-16T22:23:05+0100 Get map preference
2023-07-16T22:23:06+0100 Starting Bing proxy
2023-07-16T22:23:07+0100 External proxy "gmproxy" listening on :37637
2023-07-16T22:23:07+0100 External proxy "bproxy" listening on :38533
2023-07-16T22:23:07+0100 BB load async map zoom : true
2023-07-16T22:23:07+0100 Loaded /home/jrh/.config/mwp/vcols.css
2023-07-16T22:23:07+0100 Fixed font true
2023-07-16T22:23:08+0100 Delta speed 2.000000
2023-07-16T22:23:08+0100 Show main window
2023-07-16T22:23:10+0100 UPower: Could not connect: No such file or directory
2023-07-16T22:23:10+0100 mwp will manage power and screen saver / idle
```
* It probably helps to have the package file in a world read/writable directory, rather than `$HOME`.
* Windows (WSL) is not a supported platform, I have virtually no Windows knowledge, so I can't help you further (other than suggesting you reboot the Windows box and try again).

closed 2023-07-16 by stronnag

Jullibach, 2023-07-16:
The Solution was:
>Bios Update on Host System
>Graphics Driver Update
>Make the Installation In a read/write file for everyone (sudo chmod 777 -c -R /[Location])
now the GUI seems to work.