stronnag/mwptools #152: Feature Request - Connect Multiple USB Devices.

PetgreeRC, 2023-07-21:
Hello,
when I go flying with my friends, I always connect my remote control to the computer via USB to display the telemetry data in mwp.
Now my friends also want to see their models on the map and also connect their remote controls to the computer via USB. Unfortunately, we can only select one radio for telemetry at a time.
It would be great if there was a way, similar to the radar feature, to display multiple models on the map and use the connected remote controls as the "source". (Maybe 1 remote control to display the data on the right and the other data only as a model on the map.)
Unfortunately, the ESP32 radar is not an option because it only has a limited range, and when flying long range you are quickly over this range.

stronnag, 2023-07-21:
That's a really interesting idea. So, for telemetry, you are using what? CRSF? Smartport? Actually, it probably doesn't matter but it would be useful to know. 
 So how about:
* An "Add telemetry source" option, which would give you the option to associate an input (either USB or BT), with an identifer (e.g. model or pilot name)
* Telemetry from that input would be displayed / updated on the map, like INAV Radar does at the moment, as a symbol and an identifier (no textual displays would update, but there could be a mouse over like the INAV Radar function (range, bearing (either from home or the primary aircraft), altitude etc.).
* The identifier can be set either by typing, or selecting from a predefined list (to help pilots with more than one friend).
* This would only work with "push" telemetry (so  `LTM`, `CRSF`, `S-Port`,  or `MPM`); however it would not be necessary to define the protocol as it can be evinced. So no `MSP` conversations with multiple models, as that would be  confusing and require a lot of work to sort out.

It would be helpful if you could supply some test data:

* For the primary user, invoke mwp as `mwp --raw-log` (this will generate a raw log file of the telemetry data for the primary user)
* For one or more other _concurrent_ users, connect their controllers and record their telemetry at the same time:
   ```
   mwp-serial-cap [-b baudrate] -d /dev/ttyXXXX pilotN.raw
   ```
* Let me have all the above raw log files

PetgreeRC, 2023-07-21:
Hi,
Yes, seeing 3 remote controls would be perfect.
We are all using Edge tx 2.8.4 and Express LRS 3.2.1, Express LRS should use CRSF telemetry. The telemetry sensors come from inav 6.1.
In the TX16S we all use USB Serial VCP which is set to forward telemetry and can then select the remote control accordingly in mwp (ttyACMxx). I am attaching the relevant logs below.
I don't know why I have two logs for ttACM0 but I just send both.
[mwpLOGS2023-07-21.zip](https://github.com/stronnag/mwptools/files/12134343/mwpLOGS2023-07-21.zip)

stronnag, 2023-07-21:
Thanks, I'll have a look. May take a little while as I work out the best way to do it; maybe I just fake up some INAV Radar messages from the additional USB devices.

stronnag, 2023-07-23:
> I am attaching the relevant logs below. I don't know why I have two logs for ttACM0 but I just send both. Also I can't attach .raw files so I had zipped all of them. [mwpLOGS2023-07-21.zip](https://github.com/stronnag/mwptools/files/12134343/mwpLOGS2023-07-21.zip)

Thanks, but I know how to decode CRSF; I was looking for logs when you're all flying to assess the requirement / vehicle relationship in that environment.

PetgreeRC, 2023-07-23:
Okay, that makes sense.  I'll do the logs again next week, but for now the weather is supposed to be bad ): .  But the "way" how the logs were made is correct?  I wasn't sure because there were many errors in the console.

stronnag, 2023-07-23:
The logs are fine. There were a few CRC errors on one of the logs, but no problem at all.

stronnag, 2023-08-02:
So we may end up with something like : 

[image]

The "Primary" vehicle (a flying wing) has the standard mwp visualisation attributes. The other icons, `Replay 0` and `Replay 1` are "secondary" tracks from other pilot's CRSF telemetry (but it could be any of LTM, Mavlink, CRSF, SPort or Flysky 'AA' (INAV type 1).  Note also that `Replay 0` has not reported for  over 5 minutes ("stale"); maybe it's crashed? At least the pilot knows where to start looking.

PetgreeRC, 2023-08-03:
I´ll try it next week when the weather becomes better, still raining /: ...

stronnag, 2023-08-05:
Added LQ (if available) or RSSI to the "Radar View" for "telemetry tracked" targets.
[mwptools_6.217.345_amd64.deb.gz](https://github.com/stronnag/mwptools/files/12266761/mwptools_6.217.345_amd64.deb.gz)

```
gunzip mwptools_6.217.345_amd64.deb.gz && sudo dpkg -i mwptools_6.217.345_amd64.deb
```

closed 2023-09-12 by stronnag