stronnag/mwptools #154: MWP Not showing / detecting Radar tracked Crafts from ESP32 Radar

b14ckyy, 2023-10-20:
# Description
MWP is not showing any Radar tracked crafts when a TTGO LoRa V1.0 Module is connected to the GCS Hardware. Neither in Primary connect mode nor in Telemetry Tracking mode. 

# Used Hardware and Software
- Raspberry Pi 4, 4GB, Raspberry OS Bookworm
- TTGO Lora Hardware V1.0
- ESP32 Radar V3.0.2
- Raspberry On Board UART enabled and tested with direct FC connection on /dev/ttyS0
- USB TTL Adapter teste with direct FC Connection on /dev/ttyUSBx

# Issue 1
- /dev/ttyS0 is not shown in the Telemetry Tracker Manager as an option despite correctly defined in cmdopts, therefore using TTL-USB adapter 
`--radar-device=/dev/ttyUSB0
--radar-device=/dev/ttyUSB1
--radar-device=/dev/ttyS0
--full-screen`
[image]

# issue 2
- LoRa Module connected to TTL Adapter and powered over USB by the Pi
- Radar is set to GCS Mode via Bluetooth Config
- Radar lists active nodes (2 in my test)
- Opening MWP with `mwp --raw-log --debug-flags=20`
- Enabling Telemetry Tracker for both USB0 and USB1 (as I am not sure what is the TTL and what is USB Serial right now, tried with single USB earlier too)
- Powering on craft and node is detected, waiting for GPS Fix
- Result: No Craft is shown on the Map or in the Radar View window
[image]

- put Radar module into Standard Node operation
- Restarted MWP for fresh logs
- reset the module in Node mode 
- Other node is seen and has GPS fix
- Nothing shown on the map and Radar Tracking window

LOGS: [Logs.zip](https://github.com/stronnag/mwptools/files/13051999/Logs.zip)

stronnag, 2023-10-20:
Thanks. It looks like Telem Tracker breaks `--radar-device`. may take a little while to get to the bottom of this ...

stronnag, 2023-10-20:
Confirmed `--radar-device` is not recognised from `cmdopts`

stronnag, 2023-10-20:
@b14ckyy Can you please do a simple test for me:

* Only use one of the RADAR devices
* Do not configure any Telemetry Trackers
* Do not connect mwp to a flight controller
* Use the USB-TTL adaptor, not the on board serial 

i.e. I'd like to see the unambiguous effect of one radar device. 

```
# /dev/ttyUSBx represents the USB device connected to the RADAR 
mwp --radar-device=/dev/ttyUSBx --debug-flags=20 --raw-log
```

* If the USB device needs a baud rate other than 115200, append `@` baudrate e.g. `/dev/ttyUSBx@38400`

If it's working, as well as RADAR tracks, there will be log entries like:
```
13:07:11.801660 RDR-msg: MSP_CMDS_NAME
13:07:11.832943 RDR-msg: MSP_CMDS_FC_VARIANT
13:07:11.862969 RDR-msg: MSP_CMDS_FC_VERSION
13:07:11.870482 RDR-msg: MSP_CMDS_RAW_GPS
13:07:11.870540 RDR-rgps: Lat, Lon 54.353972 -4.523611
13:07:11.870560 RDR-rgps: 02 2a 08 c2 65 20 73 c0 4d fd 00 00 00 00 00 00 63 00
13:07:12.083557 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
13:07:12.084574 RDR-recv 0: Lat, Lon 54.354010 -4.523533
13:07:12.084649 RDR-recv: 00 01 85 c3 65 20 7b c3 4d fd a6 13 00 00 31 00 21 05 01
13:07:12.207116 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
13:07:12.207398 RDR-recv 1: Lat, Lon 54.353916 -4.523525
13:07:12.207411 RDR-recv: 01 01 dc bf 65 20 c9 c3 4d fd 6a 13 00 00 8a 00 7b 06 01
```

If it's working, we can look at generating  `udev` rules to disambiguate the radar devices, if it doesn't work, please let me have the `mwp_stderr` and raw logs.

If you wish, repeat from the top for the other radar device (especially if the doesn't work).

b14ckyy, 2023-10-20:
@stronnag For clarification: I just used the same radar device all the time. Just switched it between Node-Operation and GS-Operation modes. 

I assume you want me to test both operation modes then? No problem will do in a bit.

stronnag, 2023-10-20:
> @stronnag For clarification: I just used the same radar device all the time. Just switched it between Node-Operation and GS-Operation modes.

OK.

> I assume you want me to test both operation modes then? No problem will do in a bit.

That would be useful to test both in the simplest usage case. 

Thanks

stronnag, 2023-10-20:
Thanks, that looks good. 

From the logs, the location is never changed. The status is "undefined" (a value that comes from the radar module),
`["Undefined",` "Armed", "Hidden", "Stale"]`. The `Last` and fields to the right are only updated if the status is not "Undefined". 

From your stderr log:
```
2023-10-20T15:59:17+0200 RDR-msg: MSP_CMDS_FC_VARIANT
2023-10-20T15:59:17+0200 RDR-msg: MSP_CMDS_FC_VERSION
2023-10-20T15:59:17+0200 RDR-msg: MSP_CMDS_NAME
2023-10-20T15:59:23+0200 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2023-10-20T15:59:23+0200 RDR-recv 0: Lat, Lon 51.488640 11.977810
2023-10-20T15:59:23+0200 RDR-recv: 00 00 00 8b b0 1e 34 ab 23 07 28 3c 00 00 69 00 00 00 00
2023-10-20T15:59:23+0200 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2023-10-20T15:59:23+0200 RDR-recv 0: Lat, Lon 51.488640 11.977810
2023-10-20T15:59:23+0200 RDR-recv: 00 00 00 8b b0 1e 34 ab 23 07 28 3c 00 00 69 00 00 00 00
2023-10-20T15:59:23+0200 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2023-10-20T15:59:23+0200 RDR-recv 0: Lat, Lon 51.488640 11.977810
2023-10-20T15:59:23+0200 RDR-recv: 00 00 00 8b b0 1e 34 ab 23 07 28 3c 00 00 69 00 00 00 00
2023-10-20T15:59:23+0200 RDR-msg: MSP_CMDS_COMMON_SET_RADAR_POS
2023-10-20T15:59:23+0200 RDR-recv 0: Lat, Lon 51.488640 11.977810
2023-10-20T15:59:23+0200 RDR-recv: 00 00 00 8b b0 1e 34 ab 23 07 28 3c 00 00 69 00 00 00 00
```
Note: I've replayed this as:

### One terminal

```
mwp --radar-device udp://:30001 --debug-flags 20
```
### Second terminal

```
mwp-log-replay -d udp://localhost:30001 mwp._dev_ttyUSB0.2023-10-20T155826.raw
```

I'm fixing the failure to read the predefined `--radar-device` from  the `cmdopts` --- I'll let you when that's done (properly, I have an ugly workaround).

b14ckyy, 2023-10-20:
Maybe undefined is because its not armed. I will play around a bit to see how it behaves. At least one issue is solved and I am not crazy or utterly stupid :P 
The internal UART of the Pi should then also not be a problem anymore. I will also doublecheck if MAYBE the USB works as well instead as I prefer USB connection to keep the UART for the ERLS Airport.

stronnag, 2023-10-20:
The parsing of `cmdopts` `--radar-device` is fixed by e0ae7b3a.
As I think that was the only real (code) issue, I'll close this later unless you have anything else to add on radar.

b14ckyy, 2023-10-20:
Not more on the MWP side then. I think there is a issue with the Radar Firmware probably. So I will test other versions now. Even when armed the status does not change and no updates coming in but as you said, the module does not send more 
[image]

it seems like very rarely I get an update in. Maybe the planes have to actually move I will see. 

Just one question: When I set the "Enabled" box for a radar device, it will switch into radar tracking on demand. But when I launch MWP with the radar device cli option this port is completely blocked for telemetry radar and cannot be disabled afterwards in the gui, right? Just for my understanding.

b14ckyy, 2023-10-20:
@stronnag I think the pasing is still not working. 
It works when I use command line but not with cmdopts
[image]

stronnag, 2023-10-20:
> Just one question: When I set the "Enabled" box for a radar device, it will switch into radar tracking on demand. But when I launch MWP with the radar device cli option this port is completely [...]

In order to define a radar service you **must** use `--radar-device`, either in `cmdopts` (how about a system wide one?) or on the command line. 

The devices listed in the "Telemetry Tracker" are not radar devices, they are for telemetry tracking.  So another pilot (without the radar module) could forward telemtry from her transmitter (`CRSF`, `S.Port`, `IBus`), for example in ETX/OTX "Telemetry Mirror" and a BT dongle and it would appear in mwp, both on the map and in the "Radar View" list. 

https://www.daria.co.uk/mwptools/mwp-Radar-View/

https://www.daria.co.uk/mwptools/mwp-telemetry-tracker/

> @stronnag I think the pasing is still not working. It works when I use command line but not with cmdopts 

Damn it. When I removed the debug messages, I must have borken it. Let me try again.

b14ckyy, 2023-10-20:
Doing another test right now. And yes it seems like clearly the GS mode is broken on Radar. 
When I define a GCS Location in MWP so it fakes itself as an INAV FC and radar is in normal Node-Operation I get perfect updates once a second while flying in the X-Plane simulator. This clearly works. but in GS mode it seems to output a new position just once every few minutes if at all.

stronnag, 2023-10-20:
> Damn it. When I removed the debug messages, I must have borken it. Let me try again.

Just failed to appreciate that `--radar-device 35:53:17:04:14:BA` (worked) and `--radar-device=35:53:17:04:14:BA`  (broken) needed to be handled specifically / separately. Should be fixed by a0cd8aa3

> This clearly works. but in GS mode it seems to output a new position just once every few minutes if at all.

This is indeed down to the firmware; once the initial conversation has happened, the radar device just pushes location updates to a passive mwp.

b14ckyy, 2023-10-20:
OK all working good now. In this case I will need to find a workaround to keep the module in Node-Operation without the need to manually set a GS location everytime. I hope this woll work better then with FormationFlight when its ready. Thanks a lot for the fix

closed 2023-10-20 by b14ckyy