stronnag/mwptools #156: Trying to open "Mission Editor" produces no visible response

mirov, 2023-11-04:
I've got mwptools installed (local install vs system, though I've tried both) and can enter missions, edit individual waypoints, and run the mission terrain analysis tool.  I can even replay BlackBox logs just fine.  My issue is that when I open a mission and then try to launch the mission editor, I don't see any effect.  What used to happen was that I'd get the popup window listing all the waypoints in tabular format.  There are no error messages that I can see in the stderr log, and nothing on stdout in response to the mission editor launch attempt.  I've tried with newly created missions created from scratch as well as missions that were loaded from my system's missions directory.  This is all running on Debian 12.2 that is up to date.

I've not been able to include the stderr.txt log (github complains that it is not a supported file type).  I've uploaded it to Google Drive however, and it is very short:  https://drive.google.com/file/d/1ACcshDtJqE9PqGUKZ1dYmxZ8jeKjR0mo/view?usp=drive_link

The issue also has me including a link to the stdout text file on drive (also very short): https://drive.google.com/file/d/1eXfJppFfDe67cHGH10NP8jMfbTJ3BZYK/view?usp=sharing

stronnag, 2023-11-04:
Thanks, I'll have a look tomorrow.

stronnag, 2023-11-05:
The files are useful, in particular:

```
023-11-04T14:41:22-0700 WM: Xlib / lightdm-xsession
2023-11-04T14:41:22-0700 GL: AMD / OLAND (, LLVM 15.0.6, DRM 2.50, 6.1.0-13-amd64)
```

So, Xlib and a fairly basic (no Window Manger?) X display.
And a 10 year old GPU.

I'm surprised you get as far as you do.

Even on fairly weak SOCs, mwp is fully functional, for example:
```
13:14:34.699052 "Arch Linux" on jagular running Linux 5.15.2-cwt-3.7.5-1 riscv64
13:14:34.699121 WM: Wayland / wlroots
13:14:34.699186 GL: Imagination Technologies / PowerVR B-Series BXE-4-32
13:14:34.730199 hypervisor: none
```
At least there's a WM there.

On my Debian 12 VM, without _any_ WM (just an unadorned X session)

```
2023-11-04T22:39:04+0000 version: mwp 704bfc79 (development) / 2023-10-28T14:43:11+01:00 6.301.588
2023-11-04T22:39:04+0000 "Debian GNU/Linux 12 (bookworm)" on debstab running Linux 6.1.0-13-amd64 x86_64
2023-11-04T22:39:04+0000 WM: Xlib / Unknown DE
2023-11-04T22:39:04+0000 GL: Mesa/X.org / llvmpipe (LLVM 15.0.6, 256 bits)
2023-11-04T22:39:04+0000 hypervisor: kvm
2023-11-04T22:39:04+0000 GLib2 2.74.6 (dock fallback)
2023-11-04T22:39:04+0000 MQTT enabled via the "mosquitto" library
2023-11-04T22:39:04+0000 BLE serial enabled
```

Everything, including the Mission Tote works just fine.

So I doubt there's anything I can do to workaround this. Have you tried a more recognised, lightweight WM (xfce, lxde)?

mirov, 2023-11-05:
Odd:
 $ wmctrl -m
Name: Metacity (Marco)
https://packages.debian.org/bookworm/metacity

Asking for the info a few different ways:
$ printf 'Desktop: %s\nSession: %s\n' "$XDG_CURRENT_DESKTOP" "$GDMSESSION"
Desktop: MATE
Session: lightdm-xsession

$ ls -l /etc/alternatives/x-window-manager
lrwxrwxrwx 1 root root 14 Dec  4  2022 /etc/alternatives/x-window-manager -> /usr/bin/marco

$ dpkg -l | grep -i 'window.*manager' | sed -ne '/^ii /!d;s/^ii *//;/^lib/d;p'
marco                                                       1.26.1-3+deb12u2                          amd64        lightweight GTK+ window manager for MATE
marco-common                                                1.26.1-3+deb12u2                          all          lightweight GTK+ window manager for MATE (common files)
wmctrl                                                      1.07-7+b1                                 amd64        control an EWMH/NetWM compatible X Window Manager

 I'm certainly not running "only" X11 (that's too minimalist even for me)...

stronnag, 2023-11-05:
OK, it doesn't work on MATE, I don't care ...

mirov, 2023-11-05:
I'd like to start off by saying that I'm really enjoying the capabilities that INAV and MWP are giving my planes.  It is just a lot of fun for me, but for the developers it is obviously a huge amount of work (and support).  I've been flying R/C since '71, and the combination of a flight controller and ELRS are really impressive.

Well I sure don't need to you like my current Desktop environment, and honestly I don't expect you to care.  But if you are pretty sure that's what's causing the trouble I can try something else or as you suggest even run in a VM.  It seems weird though that pretty much everything else I run on this desktop (Chrome mostly of course, Eagle, OpenScad, INAV & ELRS Configurators, even X-Plane and most of MWP all run fine).

May I ask how your code decided that there's no WM ?  Perhaps I can just set an ENV parameter that is unset or has an unsupported value.  Is the code execution falling out the bottom of a bunch of conditional statements (due to not finding what they are looking for) or are things actually so incompatible that the logic and dependencies just fail ?  If there was at least an error message printed I could grep for that and look into the code myself.

stronnag, 2023-11-05:
So, I just wasted some time installing "Mate Deskop" in my Debian Bookworm VM.

Guess what, it works, the Mission Tote appears when you select the bar label .... 

Nothing to see here.

stronnag, 2023-11-05:
Thusly:
[image]

closed 2023-11-05 by stronnag

mirov, 2023-11-05:
Ah, I see some of what's happening.  I've been launching the Mission Editor by right clicking on a waypoint and selecting "Mission Editor" (that still does not work for me).  However, I think you just opened up the window by clicking on the "bar label".  I assume that's the vertical text just to the right of the map.  When I click on that I get a cropped version of the editor (with my waypoints showing) in the upper left of the MWP window.  But I haven't been able to resize it in both X & Y.  Closing the editor popup (pressing the [X] in the window frame corner) does close it, but it no longer shows up in the docking bar.  But If I open up one of the other status sections (by clicking on the "DirectionView" or "Telemetry" labels) the editor label comes back.

stronnag, 2023-11-05:
Mate again, both sub-windows from the RMB menu of WP3 .... Mission Editor window enlarged by dragging the edges.

[image]

Your system seems somewhat broken ...

mirov, 2023-11-05:
It seems to me that the cropping/resizing issue on the mission editor window is probably some artifact of Mate/etc that MWP doesn't like.  Thanks for giving my issue more time than you wanted to or planned on, I'll explore window manager issues on my side.   MWP always opens up maximized, is that also an artifact or just the way it is ?

stronnag, 2023-11-05:
> It seems to me that the cropping/resizing issue on the mission editor window is probably some artifact of Mate/etc that MWP doesn't like. 

Doesn't happen here (Debian 12/ mate). Windows appears correctly and resizes sanely (it's done by the WM, not by mwp).

> Thanks for giving my issue more time than you wanted to or planned on, I'll explore window manager issues on my side. MWP always opens up maximized, is that also an artifact or just the way it is ?

Documented behaviour, specifically see  `--dont-maximise` (which may be [set persistently]( 
https://www.daria.co.uk/mwptools/mwp-Configuration/#cmdopts)).

mirov, 2023-11-05:
On my broken system --dont-maximise did cause it to open up a "not maximized" window that then jumped from one screen to the other, and entered that infinite zoom behavior we've chatted about before.  So I'll chalk that up to WM incompatibility.  Thanks for all your time this evening, I'll update you on what happens with another WM.

stronnag, 2023-11-05:
`<sigh>`

[image]

mirov, 2023-11-06:
Good morning.  I found a workaround that gives me full functionality despite whatever is going on with my WM.  Although for some reason I can't move or resize the Mission Editor window by dragging, I can type ALT-SPACE and then either an "m" or an "R" to move and resize.  It took me a while to find that for "r" to work the cursor must be near the edge of the window that you want to stretch.  So everything is working perfectly now, and I'm really sorry I took so much of your time and focus.  BTW, as I was editing all my missions I came up for an idea for a RFE, but I'll show that in a new thread.