stronnag/mwptools #162: Is it expected that a RTH and the end of a fixed-wing mission will by default return at zero altitude ?

mirov, 2023-11-27:
I flew a mission today that had the last WP meant to be a RTH (see attached screenshot, the RTH box is checked).
[image]

Anyway, I assumed it would loiter at the asked-for 76m and was quite surprised to have it spiral down (clipping a tree) as it apparently was headed to *zero* altitude.  The mission editor in fact shows it at zero, I am assuming that was actually the real value it used.  Or not ?  It obviously ignores the lat/lon which are also zeros for the RTH WP.

Is this a real issue or did something else probably cause it to try to land ?

stronnag, 2023-11-27:
Please include the mission file. Screen shots are rarely helpful.
In general, what happens with RTH is FC behaviour, not mission planner.

mirov, 2023-11-27:
[Rancho_Kaiser_Quary_initial_mission.mission.gz](https://github.com/stronnag/mwptools/files/13477842/Rancho_Kaiser_Quary_initial_mission.mission.gz)

I raise the issue here because in the mwptools mission editor it shows the RTH waypoint as having zero for altitude.  I just assumed it was falling back on the FC's idea of what altitude RTH should end up at.  And perhaps that is what happens and there was another issue..

stronnag, 2023-11-27:
The RTH WP has **NO** geographic information, because the mission planner is not psychic, it cannot foresee where you will actually arm. Established in the arming point is a matter for the FC. There is nothing wrong with the mission, it should not force a landing.

If the flight behaviour bothers you, please raise an FC issue; this will probably require a blackbox log and CLI diff for investigation.

closed 2023-11-27 by stronnag