stronnag/mwptools #163: mwp crashing few seconds after arm

ssoloshchak, 2024-01-25:
I'm trying to build and run mwp inside VM running on Mac with M2 processor.
My setup is next: Radiomaster boxer connected to laptop via usb cable, mirror telemetry via VCP is enabled: when drone is powered on (meteor65 with ELRS on board) telemetry appears on the screen (battery, horizon...). Few seconds after I armed the quad - app crashes with next error. 
 
`2024-01-25T14:17:07+0200 mwp 
2024-01-25T14:17:07+0200 buildinfo: 2024-01-25T12:16:37Z Linux 5.15.0-92-generic aarch64 ubuntu-linux-22-04-desktop
2024-01-25T14:17:07+0200 toolinfo: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)  / GNU ld (GNU Binutils for Ubuntu) 2.38 / Vala 0.56.0
2024-01-25T14:17:07+0200 version: mwp d96c289 (master) / 2024-01-22T16:11:24+00:00 7.022.582
2024-01-25T14:17:07+0200 "Ubuntu 22.04.3 LTS" on ubuntu-linux-22-04-desktop running Linux 5.15.0-92-generic aarch64
2024-01-25T14:17:07+0200 WM: Wayland / ubuntu
2024-01-25T14:17:07+0200 GL: Mesa / virgl (Apple M2 Pro (Compat))
2024-01-25T14:17:07+0200 hypervisor: parallels
2024-01-25T14:17:07+0200 GLib2 2.72
2024-01-25T14:17:07+0200 MQTT enabled via the "mosquitto" library
2024-01-25T14:17:07+0200 Failed to find "fl2ltm" on $PATH; see https://stronnag.github.io/mwptools/replay-tools/
2024-01-25T14:17:07+0200 Failed to find "mavlogdump.py" on $PATH; see https://stronnag.github.io/mwptools/replay-tools/
2024-01-25T14:17:07+0200 FYI, PATH is /home/parallels/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
2024-01-25T14:17:07+0200 Using speech api 1 [espeak]
2024-01-25T14:17:07+0200 Forcing proxy offline [NMSTATE_NM_STATE_DISCONNECTED]
2024-01-25T14:17:07+0200 Get map preference
2024-01-25T14:17:07+0200 Starting Bing proxy (offline)
2024-01-25T14:17:07+0200 Registered serial device: /dev/ttyACM0 [0483:5740], Vendor: EdgeTX, Model: Radiomaster_Boxer_Serial_Port, Serial: 00000000001B, Driver: cdc_acm
2024-01-25T14:17:07+0200 BB load async map zoom : true
2024-01-25T14:17:07+0200 Loaded /home/parallels/.local/share/mwp/vcols.css
2024-01-25T14:17:07+0200 Delta speed 0.000000
2024-01-25T14:17:07+0200 Show main window
2024-01-25T14:17:07+0200 Host power: 0%, state: Unknown (0), warn: None (1)
2024-01-25T14:17:07+0200 GST: "init" <FaceTime HD Camera: FaceTime HD> </dev/video0>
2024-01-25T14:17:10+0200 Trying OS open for /dev/ttyACM0 Radiomaster_Boxer_Serial_Port
2024-01-25T14:17:10+0200 Connected /dev/ttyACM0 Radiomaster_Boxer_Serial_Port (nopoll false)
2024-01-25T14:17:10+0200 Serial /dev/ttyACM0 Radiomaster_Boxer_Serial_Port ready
2024-01-25T14:17:10+0200 Comms error CRSF/CRC 1
2024-01-25T14:17:15+0200 message => No data for 5s
2024-01-25T14:17:30+0200 New CRSF Mode Manual (0) 0 -1s 0.000000 0.000000 0 0

(org.stronnag.mwp:14132): GLib-GIO-CRITICAL **: 14:17:38.071: g_file_new_for_path: assertion 'path != NULL' failed

(org.stronnag.mwp:14132): GLib-GIO-CRITICAL **: 14:17:38.071: g_file_query_exists: assertion 'G_IS_FILE(file)' failed

(org.stronnag.mwp:14132): GLib-GIO-CRITICAL **: 14:17:38.071: g_file_make_directory_with_parents: assertion 'G_IS_FILE (file)' failed
`

stronnag, 2024-01-25:
Well, I don't know where to start! Obviously not even close to a supported platform.  Somewhat impressed that (a) you've got this far and (b) you've found a (new) bug.

As a workaround, can you please do the following in a terminal:

```
gsettings set org.mwptools.planner log-save-path '/tmp'
```

(Any existing writable directory will do for `/tmp`).

Then try mwp again.

Please confirm if  this does fix  the problem.

ssoloshchak, 2024-01-25:
Ha! Yes it works now, thank you so much for such a quick response.
[image]

closed 2024-01-25 by ssoloshchak

cptX, 2024-01-28:
I can verify that in Linux Mint 21.2 the mwp program crashes after arming!
My mwp version is https://github.com/stronnag/mwptools/commit/7b357534bd8c1fb3e0d923db8503885e6418f57b (master) / 2023-12-24T20:26:35+00:00 6.358.735