stronnag/mwptools #164: Current and mAh not shown in Battery Monitor

cptX, 2024-01-28:
Hi, I just installed mwp and doing some initial tests on my desk.

Although, my telemetry (ELRS running Crossfire protocol) successfully transmits the battery percentage and the current draw the Battery monitor in mwp shows only the voltage, the current is always 0.00A and the mAh are not shown at all!

I have a jumper T16 Pro and mwp connected through USB VCP.

[image]

stronnag, 2024-01-28:
Works just fine here
[image]

If you provide a capture of the CRSF which exhibits this behaviour, it can be investigated further.
e.g. `mwp --raw-log`

closed 2024-01-28 by stronnag

cptX, 2024-01-29:
Hi @stronnag, please don't close the issues so fast, before even investigating if there is really a bug. Also, it's not very polite for the person that spent time reporting it. Personally, it pissed me off to be honest. I'll try to provide the logs.

stronnag, 2025-01-27:
The [User Guide](https://www.daria.co.uk/mwptools/mwp_support/) makes it quite clear that Issues that do not supply required / mandatory information may be closed immediately. 
I do not have time to waste with people who cannot be bothered to help me to help them.