stronnag/mwptools #165: Current position of aircraft not shown on map

cptX, 2024-01-28:
Hi, I just installed mwp and doing some initial tests on my desk.

I have successfully connected mwp through USB VCP to my Jumper T16 Pro running edgeTX.
I can see the coordinates, heading, altitude etc but on the map there is nothing shown and also the map is not automatically centering on my location. It's like the map is completely disconnected from the rest of the program and get's no input. Should a symbol be visible on my current location?
The aircraft is not armed!

My mwp version is 7b357534 (master) / 2023-12-24T20:26:35+00:00 6.358.735

[image]

cptX, 2024-01-28:
After hitting this bug too https://github.com/stronnag/mwptools/issues/163
I ran "gsettings set org.mwptools.planner log-save-path '/tmp'"
and this resulted in showing something on the map after all. So this is for sure related!

stronnag, 2024-01-28:
As CSRF does not transmit the model type, you will only get a symbol unless you tell mwp which symbol to use.
[user guide](https://www.daria.co.uk/mwptools/running/#command-line-options). Or set it permanently in [cmdopts](https://www.daria.co.uk/mwptools/mwp-Configuration/#configuration-files). The type codes are those defined by INAV (and multiwii).  From `src/mwp/mwp_craft.vala` :
```

   public enum Vehicles {
        ARROW = 0,
        TRI = 1,
        QUADP = 2,
        QUADX = 3,
        BICOPTER = 4,
        GIMBAL = 5,
        Y6 = 6,
        HEX6 = 7,
        FLYING_WING = 8,
        Y4 = 9,
        HEX6X = 10,
        OCTOX8 = 11,
        OCTOFLATP = 12,
        OCTOFLATX = 13,
        AIRPLANE = 14,
        HELI_120_CCPM = 15,
        HELI_90_DEG = 16,
        VTAIL4 = 17,
        HEX6H = 18,
        PPM_TO_SERVO = 19,
        DUALCOPTER = 20,
        SINGLECOPTER = 21,
        ATAIL4 = 22,
        CUSTOM = 23,
        CUSTOM_AIRPLANE = 24,
        CUSTOM_TRI = 25,
        PREVIEW = 26,
        FORCE_ARROW= 27,
        LAST
    }
```

closed 2024-01-28 by stronnag

cptX, 2024-01-29:
Hi @stronnag, please don't close the issues so fast, before even investigating if there is really a bug. Also, it's not very polite for the person that spent time reporting it.
In my opinion as it proved that this bug is related to https://github.com/stronnag/mwptools/issues/163 (log-save-path configuration) you should close it only when you solve this bug with a final commit and adding here the reference to that commit. I spent a couple of hours trying to understand why I don't see anything on the map and that was my first use of the program. Imagine my frustration...
Your answer about the icon although useful is not related to my bug report...

stronnag, 2024-01-29:
It has nothing to do with #163.

Please understand that if you report non-reproducible issues without supporting  data they will be closed. It just wastes my time.

cptX, 2024-01-30:
I don't think you understood my report here. Please read again the issue. I opened your program for the first time and I didn't see ANY mark on the map. I then ran  "gsettings set org.mwptools.planner log-save-path '/tmp'" before opening the app and it solved my problem so it has the same root cause with issue https://github.com/stronnag/mwptools/issues/163. Issue and bug reporting is for making your software better, right? Please read the reports before closing them, people are spending time to do this, as you spent your time to provide this software. And for this specific issue I think I provided enough data (eg. screenshot) to show you what was necessary...
Non-reproducible issue could be for you, but indeed for me. That's the concept of issue reporting, right? to track the issues that other people are having, except if you are not interested in this at all...