stronnag/mwptools #166: Failed to find "mwp-plot-elevations" on $PATH

tomaskir, 2024-02-04:
Hi,

I installed mwp to test around, as I want to setup an ELRS AirPort link to my aircraft and use mwp as a ground station.

Running on Arch, installed with `pamac build mwptools-git`. I changed the version to `7.028.536` in `PKGBUILD`, as the current one available on AUR is old. mwp installed and is running without issues after the install.

However, running mwp, I get `Failed to find "mwp-plot-elevations" on $PATH`, and as such terrain analysis doesn't work?

Any way to fix this? Thanks!

stronnag, 2024-02-05:
Maybe I need to update the `PKGBUILD`; especially if it failed to build `mwp-plot-elevations`.

stronnag, 2024-02-05:
Did you not get the `optdepends` displayed? That should have shown you that `go` is required for some components. 
Can you post the build log?

tomaskir, 2024-02-05:
> Did you not get the `optdepends` displayed? That should have shown you that `go` is required for some components. Can you post the build log?

I did get `optdepends`, and I did install `gnuplot` as it's listed as requirements for elevation plots. `go` is however listed as only a dependency for `Build otxlog (OpenTX log player)`.

Did not know `go` is a requirement for terrain analysis, will install `go`, reinstall mwp.

stronnag, 2024-02-05:
New PKGBUILD in the AUR

stronnag, 2024-02-05:
Automatic package rebuild after a trivial master push (using `paru` as the AUR manager).

closed 2024-02-05 by stronnag

tomaskir, 2024-02-05:
Could I suggest specifying in `optdepends` that `go` is required for Terrain Analysis / `mwp-plot-elevations`? Might make it easier for people to know `go` is needed for this feature to work.