stronnag/mwptools #168: WSL wont start

jasc76, 2024-03-03:
Sorry, just followed the directions to download the current release within W11 WSL (Ubuntu Standard Distri)

After all installation I get this
root@MF-6101:/mnt/c/Users/js# mwp
2024-03-03T20:52:39+0100 Cowardly refusing to run as root ... for your own safety

How can I fix this ?

stronnag, 2024-03-03:
Don't run as root. Or read the (searchable user guide to see how to too; if you really think i's a good idea).

closed 2024-03-03 by stronnag

jasc76, 2024-03-03:
Well, since I only have an online account I have no user name in WSL
Ok, I found your great option --really- and so on. 
Well, it has grahics issue. 
Starting via Icon in Start menu doesnt do anything. The area planner starts but also with graphics issue.
But I already get it, not to ask questions..