stronnag/mwptools #170: can't GtkClutter.init

Rothammel, 2024-03-25:
I am using Windows 11 and have installed ubuntu 22 LTS from the store.
mwp was installed with the sh script,
unfortunately I get the error at startup:

stan@surface:~$ mwp
Fatal: can't GtkClutter.init

stronnag, 2024-03-25:
* Uninstall the build from the script. The `INSTALL` file documents this.
* Install the [current release build]( https://github.com/stronnag/mwptools/releases/tag/7.057.554)
* If installing the current build either tells you that there are missing dependencies (or installs some other dependencies), please note them here, as they are missing from the script.

Strictly the error is because `mwp` cannot initialise the graphics subsystem (OpenGL) that it needs. In theory, WSL2 should supply this.

If the above steps still fail, please report back and we'll come up with some other diagnostics.

closed 2024-03-25 by Rothammel

stronnag, 2024-03-25:
well if you really thing that downloading stuff as **root** into your home directory is a good idea, you get what you deserve. The installation instructions do not tell you to do that.