stronnag/mwptools #176: mwp tools crashes on WSL when a blackbox log tries to replay

rickgode, 2024-05-31:
Fresh install of mwp tools and it seems to open up and run, but when I tried to load a blackbox file it crashes.  This seems to work fine in a Linux virtual box.

I think here are some logs?  I understand that WSL is not really supported, but I really just wanted to replay my blackbox logs in windows without having to run virtualbox.

Thanks!

ricky@DESKTOP-PC:~$ mwp

(org.stronnag.mwp:17945): Gtk-CRITICAL **: 20:26:46.794: gtk_scrolled_window_set_max_content_width: assertion 'width == -1 || priv->min_content_width == -1 || width >= priv->min_content_width' failed
2024-05-30T20:26:46-0500 mwp
2024-05-30T20:26:46-0500 buildinfo: 2024-05-31T00:31:27Z Linux 5.15.146.1-microsoft-standard-WSL2 x86_64 DESKTOP-PC
2024-05-30T20:26:46-0500 toolinfo: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)  / GNU ld (GNU Binutils for Ubuntu) 2.38 / Vala 0.56.0
2024-05-30T20:26:46-0500 version: mwp 2ffc486 (master) / 2024-05-25T13:44:36+01:00 7.146.457
2024-05-30T20:26:46-0500 "Ubuntu 22.04.4 LTS" on DESKTOP-PC running Linux 5.15.146.1-microsoft-standard-WSL2 x86_64
2024-05-30T20:26:46-0500 WM: Wayland / Unknown DE
2024-05-30T20:26:46-0500 GL: Microsoft Corporation / D3D12 (NVIDIA GeForce GTX 1070)
2024-05-30T20:26:46-0500 hypervisor: wsl
2024-05-30T20:26:46-0500 GLib2 2.72
2024-05-30T20:26:46-0500 MQTT enabled via the "mosquitto" library
2024-05-30T20:26:46-0500 Failed to find "mavlogdump.py" on $PATH; see https://stronnag.github.io/mwptools/replay-tools/
2024-05-30T20:26:46-0500 Using fl2ltm 1.0.20 commit:bee49b5
2024-05-30T20:26:46-0500 Using speech api 1 [espeak]
2024-05-30T20:26:46-0500 Get map preference
2024-05-30T20:26:46-0500 Starting Bing proxy
2024-05-30T20:26:46-0500 Loaded /home/ricky/.local/share/mwp/vcols.css
2024-05-30T20:26:46-0500 gsettings blank default-map, using osm-mapnik
2024-05-30T20:26:46-0500 Delta speed 0.000000
2024-05-30T20:26:47-0500 Show main window
2024-05-30T20:26:47-0500 UPower: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
Could not aquire name

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.124: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.154: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.210: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.259: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.283: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.299: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.323: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.343: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.360: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.376: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.393: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.410: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.427: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.445: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:06.462: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:16.357: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:16.411: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:16.444: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:16.461: failed to commit changes to dconf: Could not connect: No such file or directory

(org.stronnag.mwp:17945): dconf-WARNING **: 20:27:16.477: failed to commit changes to dconf: Could not connect: No such file or directory
[... 118 lines cut ...]
2024-05-30T20:08:00-0500 Get map preference
2024-05-30T20:08:00-0500 Starting Bing proxy
2024-05-30T20:08:00-0500 Loaded /home/ricky/.local/share/mwp/vcols.css
2024-05-30T20:08:00-0500 gsettings blank default-map, using osm-mapnik
2024-05-30T20:08:00-0500 Delta speed 0.000000
2024-05-30T20:08:00-0500 Show main window
2024-05-30T20:08:00-0500 UPower: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
2024-05-30T20:08:43-0500 Getting base location from index 1 32.881367 -97.349097 true
2024-05-30T20:08:43-0500 start DEM D/L https://s3.amazonaws.com/elevation-tiles-prod/skadi/N32/N32W098.hgt.gz => /tmp/N32W098.hgt.gz
2024-05-30T20:08:44-0500 Geonames 32.881367 -97.349097 America/Chicago
2024-05-30T20:08:45-0500 Finished DEM D/L /tmp/N32W098.hgt.gz
2024-05-30T20:08:45-0500 Refreshed elevations from N32W098.hgt
2024-05-30T20:09:15-0500 Replay "bbox" log /home/ricky/flightlogs/blackbox_log_2024-05-05_135827.TXT model -1
2024-05-30T20:09:15-0500 fl2ltm -device udp://localhost:40419 --index 1 --fast --type -1 -interval 100 /home/ricky/flightlogs/blackbox_log_2024-05-05_135827.TXT  # pid=17670
2024-05-30T20:09:15-0500 set mrtype=3 cap =0
2024-05-30T20:09:15-0500 Using MSP v2 0200
2024-05-30T20:09:15-0500 set TZ offset -300
2024-05-30T20:09:15-0500 INAV v7.1.0  SPEEDYBEEF405MINI (59a6ee6)
2024-05-30T20:09:15-0500 Model name: "FLYWOO EXPLORER"
2024-05-30T20:09:15-0500 Arming flags: Box failsafe,Box killswitch,RC Link,Throttle,OSD Menu,Out of memory,PreArm,Landed (524f0000), load 0% MSP_CMDS_INAV_STATUS
2024-05-30T20:09:15-0500 Sensors: Acc Baro Mag GPS  (000f)
2024-05-30T20:09:15-0500 INAV v7.1.0  SPEEDYBEEF405MINI (59a6ee6) API 2.0 Nav Pr 0 «FLYWOO EXPLORER»
2024-05-30T20:09:15-0500 init icon 3 0 3 (false)
2024-05-30T20:09:15-0500 Poller cycle for  9 items, 81 / 226 bytes (INAV_STATUS,ANALOG2,NAV_STATUS,RAW_GPS,COMP_GPS,GPSSTATISTICS,WP,ATTITUDE,ALTITUDE)
2024-05-30T20:09:15-0500 Craft is armed, special=0

stronnag, 2024-05-31:
1. Can I see the log that causes the problem?
2. Please have a look at [this Microsoft bug report](https://github.com/microsoft/WSL/issues/10947), in particular the solution endorsed by Marc Hoffman near the end.

rickgode, 2024-05-31:
1.  I've attached the mwp_stderr.log [mwp_stderr_2024-05-30.txt](https://github.com/user-attachments/files/15513004/mwp_stderr_2024-05-30.txt)
2. I'll check out that link this afternoon.

Thank you again for your time on this.  It's AWESOME!  I didn't even realize there was a windows version of the fl2xui  https://www.youtube.com/watch?v=wGj5zAQjDH0
[mwp_stderr_2024-05-30.txt](https://github.com/user-attachments/files/15513004/mwp_stderr_2024-05-30.txt)

stronnag, 2024-05-31:
"log that causes the problem" ==> BBL.

rickgode, 2024-05-31:
It would help if I could read...  Thanks for your patience.  The file is too big to upload here (I had it set to 1/1 logging)  I've since switched it to 1/128 or 1%.  Here is a google drive link to it: https://drive.google.com/file/d/1tyVX-KSUaeSTioczr7athtYa9LjmeHbT/view?usp=sharing

stronnag, 2024-05-31:
The log replays just fine on various OS here (including in my Win11 VM); so it would appear to be the latest Microsoft _breakage du jour._

rickgode, 2024-05-31:
You're probably correct about the latest Microsoft breakage du jour.  I do think I came up with a workaround.  You were on to something about that Microsoft Bug report that worked for Marc earlier.

I added these two lines to the .bashrc and it's now working

export $(dbus-launch)
dbus-launch > /dev/null

I added the > /dev/null because I didn't want the dbus lines showing up every time I started Ubuntu WSL.
[image]

Solved! Thanks again!

stronnag, 2024-05-31:
You are just doing the same thing twice ...

closed 2024-05-31 by stronnag

rickgode, 2024-05-31:
That's the exact same thing my Linux guy told me.  But on this particular PC, I tried the export $(dbus-launch) by itself and it didn't work, then I tried the dbus-launch > /dev/null by itself and it didn't work.

It only worked with both.  So who knows with Windozz?

rickgode, 2024-05-31:
Also for funsies, I tweaked my Windows WSL shortcut for this 

Target: "C:\Program Files\WSL\wslg.exe" -d Ubuntu --cd "~" -- dbus-launch -- mwp