stronnag/mwptools #177: Error Updating

marcdornan, 2024-07-03:
I was updating mwptools and ninja install resulted in this error:
error: Package `libmosquitto' not found in specified Vala API directories or GObject-Introspection GIR directories
Compilation failed: 1 error(s), 0 warning(s)

Could this be related to me updating meson as ninja squawked that it was too out of date when I first tried.

Any ideas? I posted this in rcgroups as well as I was not sure how often you looked at this.

stronnag, 2024-07-03:
It helps to state which distro you're using .... they are all different.

I'd guess that either:

* You don't have the `-dev` package for `libmosquitto` installed (f your distro makes such a distinction) ; and / or
* You need to blow away the old `_build` (or `build` depending on when you set things up) directory and re-run the `meson` phase from scratch.

There's no harm in also ensuring the OS is completely up to date as well.

And you're right, I look a RCG about once a month, so Github issues are much preferred.

marcdornan, 2024-07-03:
Sorry. Ubunto 20.04. I will work through your suggestions.

closed 2024-07-10 by stronnag