stronnag/mwptools #184: LQ in MWPTools

Stemplar42, 2024-12-10:
Hello: I am using INAV 8.0.0 RC1 with MSP for RX and Telemetry. TX and RX are from Matek under mLRS mode.

My need is to see the LQ on screen along with the RSSI. I can only see RSS and it would be very useful to see the LQ and even the SNR (Signal to Noise Ratio) together.

Is this possible?

b14ckyy, 2024-12-10:
Do you know if it would be feasible to extend `MSP2_INAV_ANALOG` or would that be 100% compatibility breaking? Just thinking if a new MSP message is needed. 
A complete new message for LQ would be a bit overkill so maybe in the future we can think of a better way to do that without breaking anything.

stronnag, 2024-12-10:
It has been considered acceptable to extend messages by appending fields (not inserting other than at the end). Then consumers can check the length to see of the extended data is available.

Otherwise the options would be:
* A new message (eg) `MSP2_INAV_ANALOG2` that just extends `MSP2_INAV_ANALOG` with new fields and the consumer can decide which to use
* A new message `MSP2_INAV_LQ_SNR` that just gets LQ/SNR etc.

My preference would be to add the required fields to `MSP2_INAV_ANALOG` rather than create new messages and the consumer then having to decide which to use based on FC version.

Stemplar42, 2024-12-10:
As a regular user, I appreciate the ideas you have for expanding the download of information from INAV to MWP. I look forward to your progress, I am very happy with MWP and I think it would be very useful to have it under MSP modality.

b14ckyy, 2024-12-10:
as the Limitation is INAV here I opened a Request there. Guess we can close here for now unless stronnag want to keep it open for reference.

Stemplar42, 2024-12-10:
> as the Limitation is INAV here I opened a Request there. Guess we can close here for now unless stronnag want to keep it open for reference.

Ok. For my part, I will be attentive to that INAV Request.