stronnag/mwptools #186: Mwptools Black Box replay speed

fatbaldmen, 2024-12-19:
Current Behavior of blackbox replay speed is fast, and normal speed. Would it be possible to have different speed levels, such as -4 -3 -2 -1 0 1x 2x 3x 4x ?. Another option would be a pause feature. 

The reason I ask/inquire about this, is to be able to slow down, or pause the drawing. It would be handy to be able to study telemetry. Thank you for all you do, mwptool is working well. 💪

stronnag, 2024-12-19:
Pause is an existing feature (and now works on Windows). Pressing the space bar during replay pauses / resumes the playback.

Finer control of playback is complex because of the chain of processing log file => `blackbox_decode` => `fl2ltm` => `mwp`. Setting a fixed rate at the start would be relatively easy and not too disruptive (only affecting `fl2ltm`). 
I will probably do that.  Much more (e.g #122) would require a complete design of this replay chain.

stronnag, 2024-12-29:
Not convinced of the usefulness of this. I see no use case for slowing down replays; it would be possible to allow a range of speedups (where the current default equates to 10), with a range 1-20 (after which we'll probably just drop points anyway). 
[image]
Requires `fl2ltm` version 1.0.27 or later and mwp `replay_speeds` branch.

fatbaldmen, 2024-12-29:
Sorry for the delay in response. You are right, after playing with BB replay over the holiday, the pause feature with the space bar is adequate. Having different speeds really adds no value at all.

closed 2024-12-29 by stronnag