stronnag/mwptools #189: Mwptools crash

fatbaldmen, 2024-12-30:
After a succesfull mission upload, mid flight, and the mission ended, I had it set to loiter at the last waypoint. With waypoint mode from inav still active, I tried to create and upload another mission while my plane loitered. Mwptools instantly crashed. Waypoimt mode needs to be disabled before another mission is created. I do understand now that this happens, because we cannot write to the flight controller with WP is active. 

Could we get a small waring of some kind, maybe a pop-up box stating that we are still in Wap mode and it needs to be disabled, instead of the application crash?

stronnag, 2024-12-30:
Marc told me about this. In an abundance of caution, I've just pushed an update that disables any "potentially dangerous" MSP activity when armed.
Relaxing this to allow uploads when not it WP mode will be the next step.
May I see the `mwp_stderr_YYYY_MM_DD.txt` log when mwp crashed please? (in`Documents/mwp` by default on Windows).

fatbaldmen, 2024-12-30:
[mwp_stderr_2024-12-30.txt](https://github.com/user-attachments/files/18277637/mwp_stderr_2024-12-30.txt)

Marc had asked me to add this .txt file for your viewing pleasure.

stronnag, 2024-12-30:
> [mwp_stderr_2024-12-30.txt](https://github.com/user-attachments/files/18277637/mwp_stderr_2024-12-30.txt)
> 
> Marc had asked me to add this .txt file for your viewing pleasure.

Thank you. welcome to the programme ...

stronnag, 2024-12-30:
The `mwp.exe` in the attached archive should:
* Disable all  "potentially dangerous" MSP activity when armed, other than "Upload Active Mission"
* Other than in FC WP mode, when "Upload Actrive Missio" _should_ also  be disabled.

[image]

[mwp-exe-2024-12-30-wpload.zip](https://github.com/user-attachments/files/18277891/mwp-exe-2024-12-30-wpload.zip)

I can't (easily) test the WP enabled case, Marc has volunteered to test in the HITL later.

stronnag, 2024-12-30:
develoment branch 11d0ce11 (for the record)

fatbaldmen, 2024-12-30:
> The `mwp.exe` in the attached archive should:
> 
> * Disable all  "potentially dangerous" MSP activity when armed, other than "Upload Active Mission"
> * Other than in FC WP mode, when "Upload Active Mission" _should_ also  be disabled.
> 
> ![image](https://private-user-images.githubusercontent.com/158229/399373642-5d68da21-a73b-435c-91cb-d7dc28da8cec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1OTU0NjMsIm5iZiI6MTczNTU5NTE2MywicGF0aCI6Ii8xNTgyMjkvMzk5MzczNjQyLTVkNjhkYTIxLWE3M2ItNDM1Yy05MWNiLWQ3ZGMyOGRhOGNlYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMFQyMTQ2MDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YTNjMzg5YzZiYjQ4MjNmMDVkM2EwZTQzYTU2MTVlNGZiZTk4NGUxNTY1YzI0MDIyODhmMWZiZjQyMDY5ZDU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.D5b0FWeklu8IYKLES6-DPSQYq6WC-xEqt1-fRtbIMdM)
> 
> [mwp-exe-2024-12-30-wpload.zip](https://github.com/user-attachments/files/18277891/mwp-exe-2024-12-30-wpload.zip)
> 
> I can't (easily) test the WP enabled case, Marc has volunteered to test in the HITL later.

Thank you, I will also collectively test in HITL later on, and converse with Marc,  and possibly test in real flight tomorrow, if HITL goes well.

b14ckyy, 2024-12-31:
Update with HITL Tests: 

Active Mission Upload is still possible in flight when in WP mode but no crash happens. The uploaded mission while in WP Mode is ignored by INAV as expected. 
[image]

Geozone Upload and Download mid flight is disabled, is available again when disarmed.
Safehome upload and download is still available when armed and causes MWP to crash. Log attached. the last session was fine, the second last and the one before show the crash: 
[mwp_stderr_2024-12-31.txt](https://github.com/user-attachments/files/18280220/mwp_stderr_2024-12-31.txt)

When Connecting mid flight, Geozones and SafeHomes are downloaded if enabled to auto-download. I think this is fine. 

The test mwp.exe triggers a false alert in Windows Defender. Same happened when we tested the IDLE/Screensaver control in Windows on 2024-12-16. I think there is a function that Windows defender really hates, when accessing system functions. Maybe some undocumented API? 
[image]

Remember we also had that in fl2xui a while ago.

stronnag, 2024-12-31:
1. Thanks
2. I have yet to address auto-upload / download. In fact I'd forgotten about them
3. Hadn't looked at "connect when armed" - todo
4. Windows OS popups I have no control over; mwp just uses public documented APIs.

stronnag, 2024-12-31:
Don't support you have raw log for the crasher?

b14ckyy, 2024-12-31:
no raw log at this point but I can create one.

stronnag, 2024-12-31:
I've now tested  uploading Missions when in WP mode on the SITL. Absolutely no problem. 
The FC accepts the mission, the MSP conversation is the same as when unarmed (the FC just doesn't process the new mission, but mwp couldn't care less about that).

Uploading / downloading safehome works just fine (other than as expected, save to EEPROM doesn't work).

So it's some weird / unexpected / unhanded  condition on Windows (race condition, invalid memory access) that I am unlikely to be able to reproduce in any way. 

So unless someone can come up with a way to reproduce this issue on a supported OS, with generic (HC12, 3DR) low bandwidth telemetry , it's heading for "not reproducible /  WONTFIX".

b14ckyy, 2024-12-31:
Is a fix actually needed? I understood that you wanted to disable SafeHome Upload when armed anyway. Same as for Geozones. 
Download works fine though and should work due to Connect mid flight.

stronnag, 2024-12-31:
Indeed, I'm going to complete the menu enable / disable and maybe the auto functions.
Doesn't change the fact that there is some weird / unexpected / unhanded condition on Windows (race condition, invalid memory access) that I am unlikely to be able to reproduce in any way, and may crash mwp at time.

synersignart, 2024-12-31:
I have issues connecting for windows 
[image]

b14ckyy, 2024-12-31:
That's unrelated to this issue ticket.

But holy moly please clean up your COM port assignments. Delete them all including hidden devices. Then it will work.
[image]

MWP has issues if the con port number is 2 digit. We don't know why and someone with more windows coding esperience need to look at this

synersignart, 2024-12-31:
I use COM41@115200 and still have error connecting

stronnag, 2024-12-31:
> MWP has issues if the con port number is 2 digit. We don't know why and someone with more windows coding esperience need to look at this

No, we need people to provide coherent reports and logs. This is probably addressable quite easily.

synersignart, 2024-12-31:
> That's unrelated to this issue ticket.
> 
> But holy moly please clean up your COM port assignments. Delete them all including hidden devices. Then it will work. ![Screenshot_20241231-155956.png](https://private-user-images.githubusercontent.com/33039058/399490535-0287f798-3f46-45c6-9cb5-4f2350645d9b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2NTc3NzksIm5iZiI6MTczNTY1NzQ3OSwicGF0aCI6Ii8zMzAzOTA1OC8zOTk0OTA1MzUtMDI4N2Y3OTgtM2Y0Ni00NWM2LTljYjUtNGYyMzUwNjQ1ZDliLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMxVDE1MDQzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyYjFmNzZhNjdhMmM1MzE3YjJiZWNjZmVlNGVmM2U5NzA2MTc3OGM2ZWJhNThjNGRmOGQ1YzVmMmE2NTE0MDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Sv3ocru16BovsHmtoUhw6gH7Gg3HAYOnvOakLLO3UNY)
> 
> MWP has issues if the con port number is 2 digit. We don't know why and someone with more windows coding esperience need to look at this

so I have to reassign to single digit 
also for Bluetooth?

synersignart, 2024-12-31:
sorry ,  im surprise there is error on 2 digits  my Com assignments is a bit mess up as I have several drones in the line

stronnag, 2024-12-31:
The menu  options enable /disable should now be fixed by  c3b773b9.
Tested on Windows 10 via  SITL/fl2sitl over  HC12 9600 baud link  c. 15 minutes with WP mode. No errors of any kind.

closed 2024-12-31 by stronnag

fatbaldmen, 2024-12-31:
Thank you. Sorry, I got busy yesterday and couldn't test anything. Too bloody cold to put a plane in the sky, but ill get hitl running today, and play around a bunch. Will report back if I see anything abnormal.

stronnag, 2024-12-31:
Thanks, but I think Marc's already reported all I needed to know (about the cases I'd missed).

stronnag, 2024-12-31:
> Thank you. Sorry, I got busy yesterday and couldn't test anything. Too bloody cold to put a plane in the sky, but ill get hitl running today, and play around a bunch. Will report back if I see anything abnormal.

If you do want to play with something:
[mwp-exe-2024-12-31.zip](https://github.com/user-attachments/files/18283727/mwp-exe-2024-12-31.zip)
which should provide MSP "blocks" when armed.