stronnag/mwptools #190: Wind direction, Virtual air speed and Cross track error display

Sibre3, 2025-01-05:
Recently I stumbled across mwptools on iNAV discord, after a Windows version had been released.
This tool is very useful when viewing flight data.

However I would like to make a feature request. 
Would it be possible to add the estimated wind and virtual airspeed related telemetry data from iNAV, to the telemetry window? Or even displaying a wind direction and speed indicator over the flight map.
I consider such data as a must when navigating a mission at a distance. 

I believe this would be useful data when flying a mission in the wind, or even when reviewing logs. 
I understand this is mostly useful for fixedwing flight. However I have placed the horizontal wind speed indicator on the OSD of a multicopter. And it too displays the wind direction with good accuracy. It just doesn't display the wind speed. So its half way there and I believe it would still be helpful information.
Thank you Mr Hudson for your work on this tool and consideration of this request.

stronnag, 2025-01-05:
Interesting suggestion. Sure, I'll have a look at that.
Would you like to volunteer a couple of BBL together of _your estimate_ of the wind direction (and perhaps speed)?

stronnag, 2025-01-05:
And, FYI, there exists already a [standalone script]( https://github.com/stronnag/mwptools/blob/master/src/bbox-replay/inav_bbl_wind.rb) that will list wind speed and direction from a BBL, for examlpe:
```
$ inav_bbl_wind.rb  sample.TXT
Time	Dirn	m/s	knots	alt(m)
180.210	0	0.0	0.0	-1
186.376	275	0.2	0.0	2
186.633	85	0.5	1.0	3
187.918	143	0.2	0.0	7
188.432	222	0.7	1.0	8
190.744	241	1.3	3.0	11
197.424	242	2.1	4.0	32
197.938	236	2.6	5.0	34
198.195	238	2.7	5.0	35
199.223	233	2.6	5.0	38
199.737	231	3.1	6.0	39
200.251	233	3.7	7.0	41
201.022	238	4.4	9.0	44
201.278	234	4.0	8.0	44
201.792	230	4.3	8.0	41
203.591	239	4.9	10.0	30
```
Given the location, the prevailing direction would indeed be SW.

Sibre3, 2025-01-05:
> Interesting suggestion. Sure, I'll have a look at that. Would you like to volunteer a couple of BBL together of _your estimate_ of the wind direction (and perhaps speed)?

Unfortunately I'm away over this next month. And recently crashed the aircraft I took along. 
I also checked and don't have any old logs on this laptop. :(

stronnag, 2025-01-05:
OK. If  / when some useful (pilot validated) data becomes available, this feature request can proceed.

stronnag, 2025-01-05:
Mockup:
[image]

stronnag, 2025-01-06:
Thusly (not yet published).
[image]

stronnag, 2025-01-07:
Now pushed to github, requires development branch of  both mwp and [fl2ltm](https://github.com/stronnag/bbl2kml)
To do: 
- [ ] Display speeds in user's preferred speed ~freedom~ unit vice always `m/s`.

stronnag, 2025-01-11:
merged to master

closed 2025-01-11 by stronnag

Sibre3, 2025-01-17:
@stronnag I see you acquired a couple of logs.
I'll watch for when its released on the WIndow's installer version. 
If your interested in feedback. I used the installer version for a week over the christmas new year period. I haven't used all its functionality. But from what I did see; it worked fine.

Thank you once more.

stronnag, 2025-01-17:
It is in the current Window's installer on Github.