stronnag/mwptools #195: mwptools_25.01.18_x86_64_installer.exe wont start

jasc76, 2025-01-24:
Hi

I installed this version over an existing version, so it was running fine before
Installation works so far (standard not portable)

It doesn't start. A mwp process is created, but GUI never opens

Regards
Jan

stronnag, 2025-01-24:
[reporting requirements](https://www.daria.co.uk/mwptools/mwp_support/)

b14ckyy, 2025-01-26:
Did you reboot your PC? in older versions there was a process stuck if mwp.exe got terminated unexpectedly and prevented it from starting again. 
If its still an issue, please provide logs and information as in the link stronnag posted. 
If not, please update here or close.

stronnag, 2025-01-26:
I know how much you Windows users love rebooting, but it is [not necessary](https://www.daria.co.uk/mwptools/mwp-windows-msys/#known-issues)

jasc76, 2025-01-26:
I used the one build older version and now everything works. 
closed

closed 2025-01-26 by jasc76