stronnag/mwptools #197: Integrate Live ADS-B Air Traffic Data into MWP (add additional sources)

stronnag, 2025-01-26:
### Discussed in https://github.com/stronnag/mwptools/discussions/196

<div type='discussions-op-text'>

<sup>Originally posted by **ValkyrieDomi** January 26, 2025</sup>

Hi, first off, thank you for the amazing work on **MWP**! It's an invaluable tool for the community, and I truly appreciate the effort put into it.

I’d like to propose an exciting new feature: integration of live ADS-B air traffic data from ADSBexchange.com into **MWP’s** map interface. This would allow users to overlay real-world aircraft traffic onto their mission planning view, enhancing situational awareness and safety without requiring dedicated ADS-B receivers.

**Why This Would Be Valuable:**

**Enhanced Situational Awareness:**

Pilots, UAV operators, and mission planners could instantly see real-time air traffic in their vicinity, improving decision-making and flight safety.

**No Extra Hardware Required:**

ADSBexchange provides global, real-time aircraft tracking, removing the need for users to invest in or deploy dedicated ADS-B receivers.

**Pre-Flight and In-Flight Risk Assessment:**

Users can assess potential risks in their operational area before takeoff and dynamically adjust their missions based on live traffic data.

**Useful for BVLOS (Beyond Visual Line of Sight) Operations:**

Provides an additional layer of awareness for long-range UAV missions where knowing surrounding air traffic is crucial.

**Integration Opportunities:**

**MWP** could provide an optional map layer that fetches aircraft positions via the ADSBexchange API and displays them in real time.

**Potential Implementation Approach:**

Use the ADSBexchange public API to retrieve aircraft data based on the user's geolocation.

Implement an optional toggle in the UI to enable/disable the ADS-B overlay.

Display aircraft data with position, altitude, speed, and heading.

Consider configurable refresh rates to balance performance with real-time updates.

I believe this feature would make **MWP** an even more powerful tool for the community, offering a unique blend of UAV operations and air traffic awareness.

Would love to hear your thoughts on this! Thanks again for your hard work and dedication to this project.

Valkyrie

**"Concept Art:"**

[image]
</div>

stronnag, 2025-01-26:
## Overview

The first iteration of adding new public web based ADSB sources is now available in the **development** branch. 

* Requires the development branch
* Requires a new definition of that "radar device" (see below)

## Supported providers

The intention is to support any provider that offers the AdsbExchange V2 API response. The query to get such a response may differ between providers, so for now the following providers are supported:

* adsb.lol
* adsb.one

This sources have the benefits of:

* Use the same REST API (`https://host/v2/point/lat/lon/range`)
* Are free to use
* Do not require registration
* Have few usage limits

## Configuration

Using (one of) the above providers requires that the user defines a `--radar-device` option using the new `adsbx://` URI schema. It is necessary to supply the host name, the range (in nautical miles) and a refresh interval in milliseconds, in the format:
```
adsbx://host?range=R&interval=T
```
where
* host is the provider host (assumed available via https)
* range defines the range (R)  in nautical miles 
* interval defines the refresh interval (T) in milliseconds

`--radar-device` may be given on the command line, or more conveniently in `~/.config/mwp/cmdopts`, for example: 

```
--radar-device adsbx://api.adsb.lol?range=40&interval=1000
## or
--radar-device adsbx://api.adsb.one?range=40&interval=1000
``` 

## Notes

* It is pointless to define both providers as they have much the same data
* The providers permit a maximum range of 250NM
* The providers have a minimum query interval of 1s (1000ms)
* As the APIs return the data as JSON, the results are verbose. Requesting a range of 250NM may seem like a good idea; however it is not useful for our purposes and will return a large volume of data, which may be sub-optimal using mobile internet in the field.
* `mwp` uses the current view centre latitude and longitude for the query. In future other options may be offered (for example home location, vehicle location).

b14ckyy, 2025-01-26:
Awesome work mate. Would you mind to upload a windows binary for us (@ValkyrieDomi also has an Ubuntu running if you have a .deb) to test?
I have still not found the time to set up a Windows Build environment 😇

ValkyrieDomi, 2025-01-26:
I am speechless, awesome! I will try it when I got some extra free time. ✈

Yes @b14ckyy , running Ubuntu too. 

@stronnag  Thank you very much!

stronnag, 2025-01-26:
I don't have either ... but I'll build a Windows .exe and a Ubuntu binary that can replace the file from the current release.

stronnag, 2025-01-26:
For Ubuntu (should actually work on most modern Linux)

[mwp.gz](https://github.com/user-attachments/files/18551431/mwp.gz)

* `gunzip mwp.gz`
* `chmod +x mwp`
* `sudo mv mwp /usr/bin/`

stronnag, 2025-01-26:
Windows binary. Unzip and (Admin) copy `mwp.exe` over existing.
Add one of the above `--radar-device` lines to `c:\Users\<USER>\AppData\Local\mwp\cmdopts`

[mwp-exe-2025-01-26-adsbx.zip](https://github.com/user-attachments/files/18551497/mwp-exe-2025-01-26-adsbx.zip)

[image]

stronnag, 2025-01-26:
Feel free to alter the `range=` and `interval=` parameters to suit.

b14ckyy, 2025-01-26:
Looking good on first glance. 
Maybe the color scheme could be a bit different? somewhat brighter for satellite maps? 
I tried 250NM for the giggles 

I also had to install the latest snapshot first and THEN use the test .exe or it won't start. @ValkyrieDomi also had a crash when she just copied the binary. She tries the snapshot update first now. 

[image]

ValkyrieDomi, 2025-01-26:
@stronnag You are hero. Awesome work! 

**250nm** range to test it. 
[image]

b14ckyy, 2025-01-26:
Not sure if this is missing data in the ADSB stream or from the renderer but some Helicopters seem to have the wrong heading. 
2 of 3 helis I found are locked to north and moved backwards or sideways. Not a big deal but can be confusing if the date updates are stuck for a bit and I want to know the direction on the map 

[image]

stronnag, 2025-01-26:
* You would indeed need the current installer / .deb, as mwp's internal libraries will have changed (particularly on Linux)
* Colours. Well. Zoomed in to typical GCS usage, the icons are more obvious. 
* And mwp is a GCS / planner, not primarily an ADSB viewer
* If you have aircraft within alarm ranges, they are (more visibly) displayed in bright red.
* The icons are all under `<install_base>/share/mwp/pixmaps/adsb` The initial letters (`A0` etc.) are the (useful) ADSB emitter categories, `_red` refers to the alert variant https://www.adsbexchange.com/emitter-category-ads-b-do-260b-2-2-3-2-5-2/. Contributions of better icons are welcome.

stronnag, 2025-01-26:
re helos. mwp just takes the data from the ADSB feed.

b14ckyy, 2025-01-26:
Thanks for the update. Having alert in red and "irrelevant" aircraft more descent, thats actually okay and makes sense. 

Thanks a lot for that!

stronnag, 2025-01-26:
It is also the case that ADSB feeds have have incomplete data (depending on the range of the target from the receiver). And we don't know how the aggregators fuse or average out multiple reports from different stations.

ValkyrieDomi, 2025-01-26:
@b14ckyy The issue is prob. that the data as heading is not provided. See the data from aircraft below. On those that are working in MWP / correct heading these data are available also from ADSB. 

[image]

stronnag, 2025-01-26:
You can thank Charles Blackburn "Florida FPV" / @wx4cb for nagging me some years ago to build the infrastructure that made this relatively seamless.

b14ckyy, 2025-01-26:
Then thank you both guys! Great new addition. In combination with a ADSB receiver on the GS itself, the online data will perfectly cover any LOS shadows behind mountains while the receiver gives guaranteed data no matter of public database coverage. 

If I remember correctly, INAV can only send ADSB/Radar data down to the GS or also MWP up to INAV? A friend asked me that earlier but was not sure anymore.

stronnag, 2025-01-26:
mwp can only receive ADSB from the vehicle. You only get the five nearest aircraft and it's a real bandwidth hog (152 bytes) for ` MSP2_ADSB_VEHICLE_LIST` (regardless of whether the vehicle sees 0 or 5 aircraft).

b14ckyy, 2025-01-26:
Just one last thing out of curiosity: Having Info tags on the aircrafts in general would probably be pointless and a performance pit but maybe for aircraft in alert range, additional to the color change, a info flag with course, altitude speed and distance? Just to have the info right at hand without opening the list, if an aircraft comes into alert range.

stronnag, 2025-01-26:
> Just one last thing out of curiosity: Having Info tags on the aircrafts in general would probably be pointless and a performance pit but maybe for aircraft in alert range? Additional to the color [...]

That's a good idea. Maybe an information "toast" at the bottom of screen (so Gtk then takes care of removing the message etc.).

stronnag, 2025-01-26:
Apropos `MSP2_ADSB_VEHICLE_LIST` : By default, mwp only polls for `MSP2_ADSB_VEHICLE_LIST` if it detects a robust telemetry link (by some arbitrary heuristic  I've already forgotten). mwp instruments the pre-poller (startup) data exchange and based on the data rate / message rate decides whether to poll for `MSP2_ADSB_VEHICLE_LIST` .  Subject to `msp2-adsb` setting "Options for requesting MSP2_ADSB_VEHICLE_LIST. "off": never request, "on:" always request, "auto:" heuristic based on serial settings / bandwidth". Default is "off" to prevent users unexpectedly flooding the link with empty MSP ADSB messages.

b14ckyy, 2025-01-26:
I remember you told me that before. Will dig into the documentation as soon as the PicoADSB arrives. 

Speaking of bandwidth. I know its off topic here but do you see a chance of having a configurable polling rate over MSP? There is a lot of spare bandwidth on mLRS and it would be nice to have the AHRS and position updated a bit faster than 1Hz. up to 5Hz should be easily possible for position and attitude.

wx4cb, 2025-01-26:
You won’t be able to send ADSP data as the facility to transmit ADSP data
is tied to an end number registration, and all the cell call of the
aircraft that is why ADSP does not exist in remote ID does at least in the
US but the ability to send ADSB data, although technically feasiblewould be
something that could cause a whole bunch of issues with full size aircraft
not to mention it would have to be again at least in the US FAA approved

wx4cb, 2025-01-26:
The original hardware was a  uavionix pingrx Receiver, which is about the
same size as the TBS nano receiver with the PWM board

Theoretically, you could install it on an aircraft as long as you had a
separate RS-232 telemetry link down which I had originally tried, but I
found that it got just as many aircraft on a tripod 6 feet off the ground
as it did in the plane

wx4cb, 2025-01-26:
Just thinking another way to do it would be if you had an RTLSDR on a pie
zero running dump 1090 and then you could pull that in over the network
that is actually how I do it in my full size plane when I fly and it
integrates With ForeFlight on my iPad

The actual firmware name is called a stratux.

stronnag, 2025-01-26:
> I remember you told me that before. Will dig into the documentation as soon as the PicoADSB arrives.
>
> Speaking of bandwidth. I know its off topic here but do you see a chance of having a [...]

mwp rate limits polling to c 120 MSP messages / second. This is mainly for sanity on SITL, as most FCs are limited to c. 100 MSP /second anyway.

b14ckyy, 2025-01-26:
@wx4cb not talking about transmitting ADSB. Only sending ADSB data that MWPTools has available (ground receiver or online database) to the UAV over a Datalink so INAV can show them on the OSD, without extra ADSB hardware on the aircraft. 

Ardupilot can do that. Both ways. Feed the UAV with ADSB data from MP and show ADSB data from FC on MP. INAV is one way at the moment.

wx4cb, 2025-01-26:
Another idea would be maybe streaming the AR data constantly and then
injecting MWP/MSP into the data stream so you’re not overloading the
polling rate?

Just a thought they do a similar thing at work with the lightning detector
hardware where they stream the lightning data and then just interject GPS
and other data wants a second

wx4cb, 2025-01-26:
Apologies for the multiple emails. I’m driving and using voice to text
another way to potentially save Burnworth would be using some form of
encoding or compression, but that would require other things, but if there
was a way to just send it as binary as opposed to ascii That should save
some bandwidth too

wx4cb, 2025-01-26:
Oh, OK like I said before I’m driving so I may have misread it.

That said I haven’t touched any of my models in over three years  Been
having too much fun flying one of these

stronnag, 2025-01-26:
> That's a good idea. Maybe an information "toast" at the bottom of screen (so Gtk then takes care of removing the message etc.).

The whole alert thing is a bit broken for ADSBEx reports. For SDR, mwp uses the SDR location, for SBS1 it uses mwp home (both atre OK). For ADSBEx is uses the centre of the map, which is just wrong. It will use home when I fix it.

stronnag, 2025-01-26:
Not yet published (needs units shown in user's preferred setting of GA units). And it's also beeping (which you can't see :) .... (alert range and altitude set to 10000m for this example).

[image]

fatbaldmen, 2025-01-26:
So far, it is not working on my windows 10 pro. I drag and dropped the mwp.exe and replaced the existing. I open the rear view within the app,  the radar and telemetry tracking screen shows nothing. 

I noticed I do not have this file     c:\Users\<USER>\AppData\Local\mwp\cmdopts 

(The cmdopts is non exsistant)

fatbaldmen, 2025-01-27:
Seems voltage windows is missing in 2025/01/26 snapshot.

[image]

fatbaldmen, 2025-01-27:
> [@fatbaldmen](https://github.com/fatbaldmen) Did you create the **cmdopts** file and included the command?

I am not to clear on that part, no I did not. Am I creating a new folder called "cmdopts? And which command am I putting in that folder. Also what format is the command in? Sorry in advance, I am not the most computer nerdy person. I know my way around one, thays about it.

stronnag, 2025-01-27:
> Nevermind, i had to delete old files, voltage showing again. Still no adsb.
> 
> Seems voltage windows is missing in 2025/01/26 snapshot.
> 

OK. Now I'm confused. You say that the voltage widget is showing again and you post an image without it.

* If the file  `c:\Users<USER>\AppData\Local\mwp\.pamed` exists, then deleting it should restore the default sidebar widget placement. 
* As @ValkyrieDomi  stated, `c:\Users<USER>\AppData\Local\mwp\cmdopts` is an optional file of user specified options. The directory `c:\Users<USER>\AppData\Local\mwp` should have been created by `mwp` if it does not exist.
* Create the file  `c:\Users<USER>\AppData\Local\mwp\cmdopts` with your favourite text editor (`notepad` for example).
* Insert the line `--radar-device adsbx://api.adsb.one?range=40&interval=1000` 
* Save the file.

In future, mwp should create a blank file if it does not exist.

Note in the above `<USER>` represents your user name.

ValkyrieDomi, 2025-01-27:
@stronnag I was just typing on that topic, so to add, on Windows the file should be **Saved As** without any extension (.txt) and as *All files?    Similarly as on Linux? 

(The blank files would be great for other settings, to avoid user confusion, as in what format a file should be, you know....)

fatbaldmen, 2025-01-27:
> OK. Now I'm confused. You say that the voltage widget is showing again and you post an image without it.
>
> * If the file  `c:\Users<USER>\AppData\Local\mwp\.pamed` exists, then deleting it should [...]

[image]
[image]

If this looks right, nothing is appearing still, in the radar viewer.

b14ckyy, 2025-01-27:
In Explorer, enable to show file extensions. Then remove the .txt at the end.

The file has to be named just cmdopts without ending. Yours is still a txt. 😊

ValkyrieDomi, 2025-01-27:
> In Explorer, enable to show file extensions. Then remove the .txt at the end.
> 
> The file has to be named just cmdopts without ending. Yours is still a txt. 😊

Hah. Funny. Just tried on Windows - and because I use Notepad++, does the file type automatically, the regular Notepad does not, tho you need to do, what you just wrote.

So, I can also confirm it works on Windows no problemo! 👍

stronnag, 2025-01-27:
Now back to how we handle location and alerts for "ADSBX" targets, proposal:

### No location

* pre-connect, pre-arm, none of other setting conditions below: `mwp` will use the centre of the map as the centre location for web service query. No alerts, list will show range (when provided by the web service, but no bearing).

### User defined location

* User has set "GCS Icon" (View -> GCS Location or local `gpsd`) :  GCS location is used regardless of any following conditions. Alerts will be raised against GCS location, list will show range and bearing from GCS location.

[image]

* User has set "fake / planned home" (either `Ctrl-Shift-H` or loading a mission file with planned home): Planned home location is used in absence of any following (connected / armed) conditions. No _audible_ alert will be generated against  planned home, list will show range and bearing from planned home.

### Vehicle supplied locations

* Vehicle is connected and showing a 3D fix location. Vehicle location is used for alerts and list. Alerts and list range / bearing will be to (possibly flying) vehicle.

stronnag, 2025-01-27:
> Not sure if this is missing data in the ADSB stream or from the renderer but some Helicopters seem to have the wrong heading. 2 of 3 helis I found are locked to north and moved backwards or [...]

Just had a helo appear locally. Locked to N / 0. On inspecting the returned JSON I see
```
 "hex": "4079b3",
      "type": "adsb_icao",
      "flight": "ETP77   ",
      "r": "G-RNGS",
      "t": "EC45",
      "desc": "AIRBUS HELICOPTERS EC-145",
      "alt_baro": 2900,
      "alt_geom": 2150,
      "gs": 71.1,
      "track": 272.42,
```
So no `mag_heading` key, rather `track`. In future I'll look for both.

b14ckyy, 2025-01-27:
Makes sense to me I just wanted to clarify :D

fatbaldmen, 2025-01-27:
So i enable "show file extensions" which changed the cmdopts to cmdopts.txt. i renamed the file "cmdopts". Double click on it, it asks which program i want to run the file, i select Notepad. If shows that i have the command in the file bit still nothing shows in radar view.

[image]

stronnag, 2025-01-27:
> So i enable "show file extensions" which changed the cmdopts to cmdopts.txt. i renamed the file "cmdopts". Double click on it, it asks which program i want to run the file, i select Notepad. If shows that i have the command in the file bit still nothing shows in radar view.
> 

* Delete your `cmdopts` file 
* Drag the `cmdopts` from the zip file below into its place.
* Hopefully welcome your new  aerial overlords

[cmdopts.zip](https://github.com/user-attachments/files/18559033/cmdopts.zip)

b14ckyy, 2025-01-27:
Ahhhh stupid Windows Notepad. @stronnag remember when my cmdopts had weird in-line spaces? 
Had to use Notepad++ since Windows Notepad added invisible signs that mess with the readout.

ValkyrieDomi, 2025-01-27:
Whiskey Tango Foxtrot? Yeah just use proper notepad not this windows abomination....

fatbaldmen, 2025-01-27:
> * Delete your `cmdopts` file
> * Drag the `cmdopts` from the zip file below into its place.
> * Hopefully welcome your new  aerial overlords [...]

Nope, nothing still

stronnag, 2025-01-27:
Let's never let the obvious get in the way of wilder theories. Maybe there is no coverage in your area:
```
 curl https://api.adsb.one/v2/point/53.92/-122.77/40
{"ac":[
]
,"msg": "No error"
,"now": 1737993628002
,"total": 0
,"ctime": 1737993628002
,"ptime": 0
}
```

stronnag, 2025-01-27:
Add more
```
$ curl https://api.adsb.lol/v2/point/53.92/-122.77/40
{"ac":[
]
,"msg": "No error"
,"now": 1737993711001
,"total": 0
,"ctime": 1737993711001
,"ptime": 0
}
```
However if we look out 250NM:
```
$ curl https://api.adsb.one/v2/point/53.92/-122.77/250
{"ac":[
{"hex":"42475d","type":"adsb_icao","flight":"TBJ78   ","r":"VP-CYA","t":"GLEX","desc":"BOMBARDIER BD-700 Global 6000/6500",
"ownOp":"Tag Aviation Asia","alt_baro":44975,"alt_geom":45500,"gs":507.0,"track":145.93,"geom_rate":0,"squawk":"5753","emergency":"none","category":"A3","nav_qnh":1012.8,"nav_altitude_mcp":44992,"nav_heading":127.97,"lat":52.117172,"lon":-128.966827,"nic":8,"rc":186,"seen_pos":0.632,"recentReceiverIds":["dd66aa89-863e-4b79"],"version":2,"nic_baro":1,"nac_p":10,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"alert":0,"spi":0,"mlat":[],"tisb":[],"messages":18230,"seen":0.6,"rssi":-18.9,"dst":248.499,"dir":246.7},
{"hex":"a81dbc","type":"adsb_icao","flight":"ASA61   ","r":"N622AS","t":"B737","desc":"BOEING 737-700",
"ownOp":"ALASKA AIRLINES INC",
"year":"2000","alt_baro":39975,"alt_geom":40475,"gs":421.6,"track":326.12,"baro_rate":0,"squawk":"1654","emergency":"none","category":"A3","lat":50.858227,"lon":-126.349487,"nic":8,"rc":186,"seen_pos":0.197,"recentReceiverIds":["dd66aa89-863e-4b79"],"version":2,"nic_baro":1,"nac_p":10,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"alert":0,"spi":0,"mlat":[],"tisb":[],"messages":11388,"seen":0.2,"rssi":-21.6,"dst":225.764,"dir":216.9},
{"hex":"a54535","type":"adsb_icao","r":"N439GT",
"ownOp":"ATLAS AIR INC",
"year":"2002","alt_baro":33000,"gs":525.0,"track":123.50,"baro_rate":55,"squawk":"7267","lat":55.168762,"lon":-125.606850,"nic":8,"rc":185,"seen_pos":5.267,"recentReceiverIds":[],"nic_baro":1,"mlat":[],"tisb":[],"messages":215,"seen":5.3,"rssi":-49.5,"dst":124.029,"dir":308.3},
{"hex":"c00dce","type":"adsb_icao","flight":"FLE820  ","r":"C-FFFX","t":"B38M","desc":"BOEING 737 MAX 8","alt_baro":26350,"alt_geom":26900,"gs":409.2,"track":50.55,"baro_rate":1856,"squawk":"2661","emergency":"none","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":39008,"nav_heading":33.05,"lat":49.825378,"lon":-122.266557,"nic":8,"rc":186,"seen_pos":0.410,"recentReceiverIds":["ee0bd195-eaed-44b7","dd66aa89-863e-4b79","36be55ca-d446-46fd"],"version":2,"nic_baro":1,"nac_p":10,"nac_v":2,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"alert":0,"spi":0,"mlat":[],"tisb":[],"messages":3201,"seen":0.4,"rssi":-17.1,"dst":246.549,"dir":175.5},
{"hex":"c05271","type":"adsb_icao","flight":"ASP595  ","r":"C-GFFS","t":"C25B","desc":"CESSNA 525B Citation CJ3","alt_baro":43000,"gs":351.0,"track":212.30,"baro_rate":53,"squawk":"1367","lat":53.854980,"lon":-119.870964,"nic":0,"rc":0,"seen_pos":15.279,"recentReceiverIds":[],"nic_baro":0,"mlat":[],"tisb":[],"messages":47,"seen":15.3,"rssi":-49.5,"dst":102.660,"dir":91.0},
{"hex":"a97cb5","type":"adsb_icao","flight":"GTI360  ","r":"N710GT","t":"B77L","desc":"BOEING 777-200LR",
"ownOp":"TITAN AVIATION LEASING LTD - AMERICAS INC",
"year":"2023","alt_baro":35975,"gs":440.0,"track":298.60,"baro_rate":51,"squawk":"3154","category":"A5","lat":51.877068,"lon":-119.388428,"nic":0,"rc":0,"seen_pos":13.273,"recentReceiverIds":[],"version":2,"nic_baro":0,"mlat":[],"tisb":[],"messages":39610,"seen":13.3,"rssi":-49.5,"dst":173.302,"dir":133.7},
{"hex":"c08335","type":"adsb_icao","flight":"WJA160  ","r":"C-GXRW","t":"B738","desc":"BOEING 737-800",
"ownOp":"Swoop Inc.","alt_baro":35000,"alt_geom":35250,"gs":445.4,"track":55.70,"baro_rate":0,"squawk":"4553","emergency":"none","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":10016,"nav_heading":33.75,"lat":52.671707,"lon":-116.183472,"nic":8,"rc":186,"seen_pos":0.225,"recentReceiverIds":["e30441a9-f599-9446","a0fa9775-f435-4413","91e95114-324f-449a"],"version":2,"nic_baro":1,"nac_p":10,"nac_v":2,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"alert":0,"spi":0,"mlat":[],"tisb":[],"messages":7789,"seen":0.2,"rssi":-21.8,"dst":247.851,"dir":104.9}
]
,"msg": "No error"
,"now": 1737993735001
,"total": 7
,"ctime": 1737993735001
,"ptime": 0
}
```
So just 7 aircraft reported with 250NM radius of your location.

stronnag, 2025-01-27:
[image]

stronnag, 2025-01-27:
Commercial services like FlightRadar24 don't see many more aircraft in the area.

fatbaldmen, 2025-01-27:
> Let's never let the obvious get in the way of wilder theories. Maybe there is no coverage in your area:
> 
> ```
>  curl https://api.adsb.one/v2/point/53.92/-122.77/40
> {"ac":[
> ]
> ,"msg": "No error"
> ,"now": 1737993628002
> ,"total": 0
> ,"ctime": 1737993628002
> ,"ptime": 0
> }
> ```

I will be headed towards our major city named Vancouver. Where air traffic is of abundance. I will try there when I get there which is roughly 800 km away from where I'm trying now

stronnag, 2025-01-27:
You just zoom there in mwp and see ... (40NM radius)

[image]

ValkyrieDomi, 2025-01-27:
@fatbaldmen 

You configured MWP with the cmdopts file to get data of air traffic from a source either ADSB.one or ADSB.lol

With a radius and refresh rate. 

The radius moves with position of the map - where are you looking at.

So move you view to JFK or whatever with high traffic.

Don't understand why you are talking about a city 800km away and trying this when you will get there. 

We test this intentionally with 250nm radius and you can literally browse entire globe and see air traffic anywhere. Whether zoomed in or out.

Or am I missing something with he Vancouver in particular?

b14ckyy, 2025-01-27:
@ValkyrieDomi It was a misunderstanding. Just explained it in chat to him :) 

btw interestingly, none of the airplanes in the US seem to transmit any Compass heading. Europe its much more common. 

[image]

ValkyrieDomi, 2025-01-27:
@b14ckyy Great! 

Ah, good finding! So better to use Ground Track (as @stronnag confirmed above)

fatbaldmen, 2025-01-27:
> [@fatbaldmen](https://github.com/fatbaldmen)
> 
> You configured MWP with the cmdopts file to get data of air traffic from a source either ADSB.one or ADSB.lol
> 
> With a radius and refresh rate.
> 
> The radius moves with position of the map - where are you looking at.
> 
> So move you view to JFK or whatever with high traffic.
> 
> Don't understand why you are talking about a city 800km away and trying this when you will get there.
> 
> We test this intentionally with 250nm radius and you can literally browse entire globe and see air traffic anywhere. Whether zoomed in or out.
> 
> Or am I missing something with the Vancouver in particular?

Marc showed me how to change providers. And now that you mention it I'm just using the default 40 nautical miles and 1hz refresh. That could be why I'm not seeing anything cuz according to adsp exchange there was nothing within 40 nautical miles so if I change that command to 250 nautical miles. I should then be able to see something. I was talking about being 800 km away tonight because I thought I needed to be physically closer to more air traffic. But now I'm in the understanding that I can just zoom anywhere in the world and within, for an example, $250 nautical miles it'll pick up air traffic if it's there. Please excuse my ignorance I'm still trying to learn all this as well. I appreciate you guys walking me through this I'm getting a better understanding now

fatbaldmen, 2025-01-27:
Ill just leave this here, and show myself out :D

 [image]
[image]

ValkyrieDomi, 2025-01-27:
@fatbaldmen No worries as well as problem to help :)) And congrats! 

Also I would rather use this thread for suggestions, testing feedback, updates from Stronnag, features, you know. 
For tailored help, discord or smth. 
That´s just my idea/opinion.

[98 more replies]

closed 2025-02-03 by stronnag