stronnag/mwptools #200: [Feature Request] Add UBLOX AssistNow support

mmosca, 2025-01-31:
I think assistnow would make a lot of sense in MWPTools.

The API is fairly simple, the hardest part is splitting the ublox commands in chuncks, so they can be encapsulated into the MSP message.

The code to load the data and split it in chunks from inav-configurator is here:
https://github.com/iNavFlight/inav-configurator/blob/cf64225629a78c23921e6cd5946c2bf44072d065/js/ublox/UBLOX.js#L187

The code to send MSP messages, is here:
https://github.com/iNavFlight/inav-configurator/blob/cf64225629a78c23921e6cd5946c2bf44072d065/tabs/gps.js#L440

Or all in one, in the protype python script:
https://github.com/iNavFlight/inav/blob/master/src/utils/assistnow.py

stronnag, 2025-01-31:
Thanks, noted. Once I get the insatiable ADSB monster off my back. "_Give some people an inch and they take a mile_".

stronnag, 2025-02-03:
OK, pretty simple to do.

Typically  233 messages for the online case
Typically  4249 messages for the offline case

On the basis that most telemetry devices can round-trip  sustain between 8 MSP/sec and 25 MSP /sec,  it might be quicker just to wait for a fix .....

stronnag, 2025-02-03:
[image]

development branch
```
 gsettings  set  org.stronnag.mwp  assist-key  'xxxxxxxxxxxxx'
```

mmosca, 2025-02-04:
For the offline messages, you may select to only upload things that are relevant now. Like, 1 or 2 days, and cache the bulk.

They claim assitnow info may also improve the fix, not only make it faster.

Not sure how much of that is marketing, and how much is reality :).

stronnag, 2025-02-04:
You are a mind reader .....

[image]

stronnag, 2025-02-04:
Just uploaded the location specific data and the  (never gets an indoor's fix) old M8  has a location . 0 sats so probably just echoing what the Assist data provided. No exactly where I am, but not far off.

stronnag, 2025-02-04:
Filtering the offline data looks like a good idea. By message ID, we have:
```
msgid 0x0 /0, count = 32
msgid 0x2 /2, count = 23
msgid 0x3 /3, count = 27
msgid 0x5 /5, count = 3
msgid 0x6 /6, count = 24
msgid 0x20 /32, count = 4176
```
msgid 0x20 are the time stamped `UBX-MGA-ANO`, of which there are c. 110 per year at the moment.
So just uploading the current year and next one or two years would save a lot of nugatory message uploads.

mmosca, 2025-02-04:
This is what I do on INAV Configurator:

So you are in the right track.  I probably haven't checked if there is difference between m8 and newer devices messages.

```
self.isAssistnowDataRelevant = function(ubxMessage, cy, cm, cd) {
        if ((ubxMessage[2] == 0x13 /*UBX_CLASS_MGA*/) && (ubxMessage[3] == 0x20 /*UBX_MGA_ANO*/))
            {
                // UBX-MGA-ANO
                const payloadOffset = 6;
                if (((ubxMessage[payloadOffset + 4] + 2000) == cy) && (ubxMessage[payloadOffset + 5] == cm) && (ubxMessage[payloadOffset + 6] == cd))
                {
                    //console.log("UBX-MGA_ANO date matches");
                    return true;
                }
            } else {
                //console.log("UBX-CMD: class: 0x" + ubxMessage[2].toString(16) + " id: 0x" + ubxMessage[3].toString(16));
                return true;
            }
        
            return false;
    }
```

stronnag, 2025-02-04:
Even makes more sense when I look at the right payload offsets. So there about 110 *daily* ANO data items.
I understand there is no difference for M8 and later models.

mmosca, 2025-02-04:
> Even makes more sense when I look at the right payload offsets. So there about 110 _daily_ ANO data items. I understand there is no difference for M8 and later models.

I remember there being 2 different api depending on the gps version. I am not sure if m8 was before, or after the cutoff for the current api. The ubx commands returned would be different.

stronnag, 2025-02-04:
The APIs are M7  or M8 and later.  M7 I'm ignoring.

stronnag, 2025-02-05:
implemented in c52f914d

closed 2025-02-05 by stronnag