stronnag/mwptools #205: Mavlink-RC (ELRS) connection issue on Windows

ValkyrieDomi, 2025-02-07:
Hello, I was testing MWP on Windows few days ago, with Mavlink-RC. 

The issue is, as I usually connect on 14550 port, to my TX backpack, which however works flawlessly on Ubuntu (or automatically connects in MP on Win.), no successful connection in MWP on Win.

[mwp_stderr_2025-01-23.txt](https://github.com/user-attachments/files/18702009/mwp_stderr_2025-01-23.txt)

This is a separate issue but I would like to also mention that on Ubuntu with Mavlink-RC successfully connected, AHI banks only right. Thus when FC banked left, AHI stays horizontal.

ValkyrieDomi, 2025-02-07:
1. Yes, tested with MWP whitelisted and also firewall off. Both tests not successful. 

2. I will check out the log in the evening and report.

stronnag, 2025-02-07:
The first issue is because Windows is stupid and appears not to work correctly on dual-stack (IPv6 and IPv4 sockets). Invoking mwp as:
```
# --no-poll is optional, just keeps things tidy in the log ....
mwp --no-poll --force4
```
resulting in data being received.

 However, this generates a second problem .... mwp crashes when some Mavlink packet is received on Windows (but not on Linux or FreeBSD). I'll look at that later, probably after I get your raw log as my Mavlink sample is not recent.

ValkyrieDomi, 2025-02-08:
I just connected to MWP and got the RAW LOG.
I did right bank - AHI banks right, then following left bank - AHI stops at horizontal position.

I am unable to upload the RAW here, so I will tag you in the Discord channel, with the file.

stronnag, 2025-02-08:
Many thanks. I believe it's all fixed (including a workaround for Windows `V6ONLY`); your log will confirm it.

stronnag, 2025-02-08:
Both issues are fixed in the `development` branch (0a7cb5d8).

closed 2025-02-08 by stronnag