stronnag/mwptools #208: ADSB Targets disappear on map and miss heading, from dedicated "ADSBee" Receiver.

b14ckyy, 2025-02-15:
I am using a ADSB receiver called ADSBee 1090. See Project GH here: https://github.com/CoolNamesAllTaken/adsbee
The module is capable of outputting Mavlink-ADSB Messages in MAVLINKv1 and MAVLINKv2 format through USB and UART. 

When the Module (currently running 0.7.3 Beta firmware release) is connected to INAF 8.0 and MAVLINK2 as the protocol, INAV can see the ADSB contacts and reports them correctly to the INAV Mission control over MSP.

When the Module is connected to MWP though, the contacts are seen in the Radar List with the correct protocol and at first also on the Map. 

All Contacts miss Heading information and also go into a transparent "stale" representation on the map after just a few seconds. The Icons still move around and get updated locations but stay in "stale" state on the map. After 60s they disappear but stay in the radar view list. 

Only new contacts that come in range will appear for a minute on the map again and then disappear the same way. 

Here is a raw serial log from the module over USB: 

[mwp.COM7.2025-02-15T133736.zip](https://github.com/user-attachments/files/18810727/mwp.COM7.2025-02-15T133736.zip)

STDERR log (multiple sessions including the last one from the raw log: 

[mwp_stderr_2025-02-15.txt](https://github.com/user-attachments/files/18810729/mwp_stderr_2025-02-15.txt)

The ADSBee firmware had a bug until the previous 0.7.2 release that broke the compatibility with MWP and INAV and just started working with the latest release. If you can still see any issues on that end, I will report that to the Developer.

stronnag, 2025-02-15:
The reports are not setting the validity flags correctly. For example, from mwp's `mavreader` tool:
```
0000.00 Mav2: mavlink_traffic_report 38 : ICAO 4952c1, callsign TAP31CL valid 193 : 50.938531 11.549225 10670.0m swk=0 etyp 3, tslc 60
0000.00 Mav2: mavlink_traffic_report 38 : ICAO 4d2350, callsign RYR9WD valid 193 : 51.510227 11.610842 11585.0m swk=0 etyp 3, tslc 46
```

The validity flag is set to 0x193 in both cases, which excludes heading 0x4 and velocity 0x8.
mwp declines to display not valid data.

CoolNamesAllTaken, 2025-02-15:
Thanks @stronnag ! I was indeed missing the flags in my firmware.

stronnag, 2025-02-15:
> Thanks [@stronnag](https://github.com/stronnag) ! I was indeed missing the flags in my firmware.

Cool. I now need to fix mwp's logging for Marc's next experiment, as mwp's Windows' support is new, and there is also a bug there (not setting `O_BINARY`). 

Still `validty flags` sounds like the reason.

So that's three bugs in three products from one test ... (INAV f/w should check the flags as well).

b14ckyy, 2025-02-15:
Alright so the Heading and speed is fixed and contacts stay fully colored. 
But one issue remains, that the icons on the map randomly disappear. I can occasionally trigger that by just clicking on the map without dragging. 
moving the map or zooming again, and they reappear.

stronnag, 2025-02-15:
> Alright so the Heading and speed is fixed and contacts stay fully colored. But one issue remains, that the icons on the map randomly disappear. I can occasionally trigger that by just clicking on the map without dragging. moving the map or zooming again, and they reappear.

Once I can fix the brain deal Windows text/binary nonsense, I'll ask you another log.

stronnag, 2025-02-15:
@b14ckyy Can you run the attached with `--raw-log`? 

Please ensure you unzip it into your existing installed `mwptools` directory keeping the sub-directory structure; I'm in the middle of a rework on the symbol scaling and settings schema has changed, so you need the schema files.

The raw log will still be in the starting directory as I haven't got to moving that yet.

```
  Length      Date    Time    Name
---------  ---------- -----   ----
  1542656  2025-02-15 15:37   bin/mwp.exe
    48612  2025-02-15 15:37   share/glib-2.0/schemas/gschemas.compiled
    25439  2025-02-15 15:28   share/glib-2.0/schemas/org.stronnag.mwp.gschema.xml
---------                     -------
  1616707                     3 files
```

[mwp-exe-2025-02-15.zip](https://github.com/user-attachments/files/18811707/mwp-exe-2025-02-15.zip)

b14ckyy, 2025-02-15:
Done. 

[mwp.COM7.2025-02-15T171010.zip](https://github.com/user-attachments/files/18811787/mwp.COM7.2025-02-15T171010.zip)

again plenty of planes missing or shown with high transparency. 

[image]

stronnag, 2025-02-15:
> Done.
> 
> again plenty of planes missing or shown with high transparency.
> 

I believe that's because many of the targets are reported with quite high `tlsc`.
For ADSB, mwp considers reports older than 15s to be 'stale' and hides reports more than 30s old.
These figures work fine with other sources (PingRX, local SDR, Internet) and prevent the picture  being overwhelmed with expired targets (which is always a problem in such tracking applications). 

You can use the `mavreader` tool in `mwptools/src/samples/mavtest` to decode your log.
[mavreader.win64.zip](https://github.com/user-attachments/files/18811986/mavreader.win64.zip)
```
mavreader.exe mwp.COM7.2025-02-15T171010.raw
```
 And see how many reports have "stale" / excessive `tslc`

Given that all mwp's other ADSB sources behave rationally with mwp's empirical timeouts, I don't consider this to be a mwp problem (at the moment anyway ...)

b14ckyy, 2025-02-15:
I see Stale in the Status relatively often ineed especially since I am indoors there are a lot of messages not coming trhough. But as of right now I have apparently 15 targets and 11 Live with 3 stale. This quite matches the expected reception radius compared to adsbexchange. Yet on the map I only see one single contact. Thats what confuses me. 

[image]

stronnag, 2025-02-15:
> I see Stale in the Status relatively often ineed especially since I am indoors there are a lot of messages not coming trhough. But as of right now I have apparently 15 targets and 11 Live with 3 stale. This quite matches the expected reception radius compared to adsbexchange. Yet on the map I only see one single contact. Thats what confuses me.
> 
> But I get your point that the map should be cleaned in time and 30s is a reasonable timout.
> 
> [image]

 Fixed in `development` 08a1d2f66. mwp did not like the extreme staleness  / "time late" and would fail to establish visibility as the oscillated between timely and late. 
Somewhat strange behaviour from this device compared to other options; mwp should now cope with it.

b14ckyy, 2025-02-15:
Its still in beta state and gets optimized and @CoolNamesAllTaken just replaced the decoder and I guess its maybe not 100% smooth yet. Feedback is always welcome on that end :D 
Thanks a lot and glad we smashed a few more bugs today.

stronnag, 2025-02-16:
mwp (develep[ment branch) should hopw cope somewhat better with  wacky oscillating tlsc.
If you run mwp as:
```
MWP_ADSB_LABEL=1 mwp 
```
[image]
you see the 'planes as labels: name, tlsc and either "Stale" or a (number), which should be zero.

b14ckyy, 2025-02-16:
That's cool. So just to confirm: the ltsc info comes from the ADSB receiver, right? In this case if the plane's data is flagged as stale despite still updating its location, its a ADSBee bug? Since I can even with these flags see them as "stale" while moving. 

I think there is not much more that you could do here and we wait until CNAT is back up. Early morning for him ^^ 
Thanks a lot, as always. 

I have not yet messed around with the new library to get the pointy flags. Will do another time when I set up my GCS on Ubuntu. Hope it will not take too long until this rolls out in libshumate.

stronnag, 2025-02-16:
> That's cool. So just to confirm: the ltsc info comes from the ADSB receiver, right? In this case if the plane's data is flagged as stale despite still updating its location, its a ADSBee bug? Since I can even with these flags see them as "stale" while moving.

`tslc` is a MAVLink ADSB field (from the superb PingTX dev guide):

[image]

ADSBee bug or limitation ... maybe. You can see this, as you have mwp's `mavreader` tool. For example `RYR58DM` is a good example / serial offender: 

```
$ mavreader /tmp/mwp._dev_ttyACM0.2025-02-16T103944.raw  | grep RYR58DM
...
0303.08 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 57
0304.13 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 58
0305.14 Mav2: mavlink_traffic_report 37/38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 0
0306.18 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 1
0307.20 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 2
0308.18 Mav2: mavlink_traffic_report 37/38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 0
0309.23 Mav2: mavlink_traffic_report 37/38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 0
0310.27 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 1
0311.28 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 2
0312.29 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 3
...
0363.29 Mav2: mavlink_traffic_report 38 : ICAO 4d2346, callsign RYR58DM valid 19f : 56.947034 1.638680 11890.0m 301° 301m/s swk=0 etyp 3, tslc 54
```
and see how the `tslc` jumps around (so in many cases getting 1 aircraft 'ping' per minute. The first number is the raw log timestamp (seconds) . You can replay the raw log in mwp and correlate with the `mavreader` output.

> I have not yet messed around with the new library to get the pointy flags. Will do another time when I set up my GCS on Ubuntu. Hope it will not take too long until this rolls out in libshumate.

Unfortunately, I think it will be a while. 1.4 is targeted for GNOME 48 in March 25, 1.5 will therefore be August / September 25 (ties in with Ubuntu and Fedora release dates). If you're running a rolling release (Arch, Alpine, Chimera, Debian Testing / Unstable) then probably sooner (and as Msys2 is basically Arch, Windows will see it before Ubuntu!).

stronnag, 2025-02-16:
And just to clarify / follow up the last.
* `MWP_ADSB_LABEL=1` does not depend on pointy labels 
* You will see from the `mavreader` output that in many cases the location is not changing. This is fine, just saying "haven't heard from this one for a while, last time I did, it was here".

b14ckyy, 2025-02-16:
thanks for clarifying and the extra info. Yes I used MWP_ADSB_LABEL=1 as it makes the visualization of the issue very easy and it shows the centered flags as expected. 

I guess we can close this here now. Looking forward to the next snapshot :)

closed 2025-02-16 by b14ckyy

stronnag, 2025-02-16:
Actually one other thing for @CoolNamesAllTaken ; in the above log extract you see annotations ` mavlink_traffic_report 38 ` and `mavlink_traffic_report 37/38`. 38 is the expected payload size, 37/38 means that the Mavlink payload size was set to 37 in the mavlink header, which is probably not helpful (and the packet passed CRC based on the smaller size) Unless there's some Mavlink trick I'm missing here..

wiedehopf, 2025-02-20:
i've checked the video against the log he sent with it and parsed the log with mavreader.
```
0270.50 Mav2: mavlink_traffic_report 37/38 : ICAO 4855d0, callsign KLM55Z valid 19f : 51.577235 11.979254 11280.0m 94° 94m/s swk=0 etyp 3, tslc 0

```

[image]

4 seconds into the video this position comes up and shows as Stale / 36 despite mavreader showing tslc 0

I've tried to look at the source but nothing stood out to me.

wiedehopf, 2025-02-23:
Ok so i'm a bit curious now, i've gone and made myself a VM and was trying to replay the raw log but the button is greyed out in mwp.

[image]

Did i build wrong or does it need a serial device connected to allow for a replay (don't have any of this hardware just messing with the software).

wiedehopf, 2025-02-23:
```c
/**
 * @brief Trim payload of any trailing zero-populated bytes (MAVLink 2 only).
 *
 * @param payload Serialised payload buffer.
 * @param length Length of full-width payload buffer.
 * @return Length of payload after zero-filled bytes are trimmed.
 */
MAVLINK_HELPER uint8_t _mav_trim_payload(const char *payload, uint8_t length) {
    while (length > 1 && payload[length - 1] == 0) {
        length--;
    }
    return length;
}

```
And this is the code in adsbee that causes the shorter payload.

Code on github:
https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/pico/application/comms/mavlink/README.md

stronnag, 2025-02-23:
> /**
>  * @brief Trim payload of any trailing zero-populated bytes (MAVLink 2 only).
>  *
>  * @param payload Serialised payload buffer.
>  * @param length Length of full-width payload buffer.
>  * @return [...]

Yeah. mwp has never implemented the MAV2 nul pruing code. 
I should fix that.