stronnag/mwptools #210: [Win] COM port Radar devices disappear from new Radar manager when deselected

b14ckyy, 2025-03-02:
First of all, the Radar Device Manager is a great addition! Especially with multiple sources to switch on demand. 

But I found a small bug in Windows. 
in cmdopts i defined `--radar-device COM7` for my ADSB receiver. Its listed in the manager after startup and works but when I deselect it (just the checkbox) it gets removed from the list. I can manually add it again and it works but then disappears again when unselected. 

For online sources this does not happen and they stay in the list as expected.

stronnag, 2025-03-02:
That's a generic bug ...

stronnag, 2025-03-02:
Fixed by 9fbe3d4e  (development)

stronnag, 2025-03-03:
Closing, now in master.

closed 2025-03-03 by stronnag

b14ckyy, 2025-03-03:
Nice. Could not test yesteday but will update my Pi today.