stronnag/mwptools #215: Question: ELRS Backpack

wx4cb, 2025-04-22:
hey stronnag.. LTNM 😄 

just a quick question. I checked here: https://www.daria.co.uk/mwptools/mwp-multi-procotol/#serial-devices and i already knew about tcp/udp connections for data, but couldnt find the answer to this:

Is the elrs wifi backpack directly supported as a provider for telemetry data? if so, what would the URI be for it (if you know 😄 ).

btw, great work from what i've played with so far 😄 

Charlie

stronnag, 2025-04-22:
> Is the elrs wifi backpack directly supported as a provider for telemetry data? if so, what would the URI be for it (if you know 😄 ).

mwp just requires a transparent serial device. I have no idea what an "elrs wifi backpack" might be.

stronnag, 2025-04-22:
and transparent serial covers a multitude of things like UARTS, BT, UDP, TCP

wx4cb, 2025-04-22:
> > Is the elrs wifi backpack directly supported as a provider for telemetry data? if so, what would the URI be for it (if you know 😄 ).
> mwp just requires a transparent serial device. I have no idea what an "elrs wifi backpack" might be.

it's part of the transmitter module that creates a wifi network (similar to the way crossfire works) and you can set it to send telemetry out. it's not a big deal. i'm just getting back into things and was just curious as ELRS is new to me 😄

stronnag, 2025-04-22:
well mwp demonstrably works with unsolicited telemetry (LTM, INAV favour Mavlink at least) over TCP or UDP.  If it's UDP and Mavlink, there are various query options `bind=port` ,  `mavlink=VERSION` to coerce providers into cooperation.

wx4cb, 2025-04-22:
> well mwp demonstrably works with unsolicited telemetry (LTM, INAV favour Mavlink at least) over TCP or UDP. If it's UDP and Mavlink, there are various query options `bind=port` , `mavlink=VERSION` to coerce providers into cooperation.

ok cool..

as i say, just getting back into the swing of things :D 

you can close this for now. I'll work on it and let you know if i can get it working

Charlie

closed 2025-04-22 by wx4cb