stronnag/mwptools #217: AHI pitch reversed (Again 😆 )??

wx4cb, 2025-05-02:
Not sure if it's me or..... but with reference to the old versions issue of the ahi being reversed in pitch (see: https://github.com/stronnag/mwptools/issues/92 )

i'm guessing we can do something similar that you did in the old version to invert it by just putting a "-" in front of it on this line?

```_trans_y = (double)pitch;```  to ```_trans_y = -(double)pitch;```

trying to figure out what the code does without comments 😃 and this seems A logical place as it only affects the drawing and not anything else that may be using the same values. I found the invert roll option in the settings, but obviously not the pitch.

feel free to chew me out 😺 

```
FILE: mwp-ah_widget.vala

private void update_ahi(int roll, int pitch) {
            if(roll < 0)
                roll += 360;

			if(roll > 360)
                roll %= 360;

            if(pitch > 70)
                pitch = 70;
            if(pitch < -70)
                pitch = -70;

			_angle = (double)roll;
			_trans_y = (double)pitch;    <<<<<<<<<<<<<<<<< Here
            queue_draw();
        }
```

wx4cb, 2025-05-02:
yes... just did it and it works, although i did notice that ahi_invert_roll doesnt do anything on mine (and is also backwards)

stronnag, 2025-05-02:
Whcih telemetry protocol (they are somewhat different). 
That is probably not the correct place to fix it, as it may break other protocols.

stronnag, 2025-05-02:
The error will be in the normalisation of the actual sensor to the normalised data displayed by the widget.
It therefore needs to be fixed in the CRSF deserialisation / normalisation. 
Changing the display widget will break the display of (validated) normalised data for other protocols.

I'll find a CRSF RX and have a look later.

wx4cb, 2025-05-02:
I understand that, however, if there is a way through the settings to just
invert it on display, it shouldn’t break anything else. You already have
AHI_invert_role in the settings couldn’t we just add one for pitch in the
settings? That said I know on mine. The invert role doesn’t do anything for
the widget.

I’m not sure if that is implemented yet, but I do think to maintain
compatibility and not break other protocols doing it in the “display
routine “would be the safest place to do it

The INAV LUA script just display it correctly, but changing it in the D
sterilization section of the protocol decoder potentially turned break TBS
compatibility if it is in fact in the ELRS thing if that makes sense

wx4cb, 2025-05-02:
And please excuse grammatical errors I’m using voice to text and Siri
doesn’t understand English half the time lol

stronnag, 2025-05-02:
> I understand that, however, if there is a way through the settings to just
> invert it on display, it shouldn’t break anything else. You already have
> AHI_invert_role in the settings couldn’t we just [...]

It has to be changed in the deserialisation, otherwise logging values are wrong, forwarded values are wrong and the dislplay is broken for everyone else.

wx4cb, 2025-05-02:
I see that does make sense. I guess I’ll take a look when I get home right
now. I’m on my way to spend half a day at the beach making hurricane
repairs from last year because typical local government they don’t wanna
pay lol

stronnag, 2025-05-02:
I't's already fixed. I'll push to GH shortly.

stronnag, 2025-05-02:
And there is no ahi-reversal setting; it was removed a while ago when I fixed the normalisation for MSP/LTM/Mavlink.

stronnag, 2025-05-02:
> If you go into the “advanced settings “the 2nd or third line down there is
> an option to invert role. That was why I mentioned it.
> 
> That maybe should need to get removed in that case and would explain why it
> doesn’t do anything lol
> […](#)

Huum, that should not be there ...

and the CRSF normalisation is fixed in 4cd78b34

stronnag, 2025-05-02:
Legacy roll inversion setting removed from gsettings xml in f26a1d7e

closed 2025-05-03 by stronnag