stronnag/mwptools #220: HID initialization on connect

arealmess, 2025-05-16:
I have been trying to find the cause of this but not having much luck, 

The way I test it right now is to have my FC plugged into iNav on the receiver tab on another machine. Then hit connect in MWP. What I'm looking for is the behavior of the RC compared to how the mapping file is configured.

By watching the initial values of the channels, you can tell whether or not extra mapping file features have been taken into account. For example, buttons tagged as `latch` will behave as regular momentary switches if you see one set of initial channel values, or will behave as latched after another.

It appears random, maybe it has something to do with async things happening, but if you wait out the popup after connecting, ie no inputs during init, there's a better chance to have all features from the mapping file read and behaving as expected.

stronnag, 2025-05-16:
works fine here. I don't understand what you mean by "The way I test it right now is to have my FC plugged into iNav on the receiver tab on another machine. Then hit connect in MWP. What I'm looking for is the behavior of the RC compared to how the mapping file is configured.". 

Please explain .

stronnag, 2025-05-16:
You might start by explaining exactly how you have everything configured:
* Telemetry radio
* Data rate available on telemetry radio

log file

arealmess, 2025-05-16:
Maybe it's only latch states not initializing properly every connect/disconnect cycle. But in this video you can see both outcomes from connecting, and two different initial states of channels 6 and 8, which are extra axes and not related to latched states at all. 

Basically I have my FC plugged into my laptop running iNav configurator, watching the RC values, and MWP on my steamdeck. 

TX: ELRS 2.4ghz Radiomaster Ranger Micro
Rate: 500hz

https://github.com/user-attachments/assets/6b42c723-c50b-418f-af9f-6117b63e5b81

[message.txt](https://github.com/user-attachments/files/20257829/message.txt)

stronnag, 2025-05-16:
OK, I understand now. BTW Inav is the firmware, INAV Configurator is the configurator. If you confuse the two, you confuse me.

I'll take a look later. After I sort out the Windows dual stack nonsense.

stronnag, 2025-05-16:
I have no idea what `joyreader-toggles` might be. Please test `development`

arealmess, 2025-05-16:
Oh I was behind. I'm kind of new to real git and github stuff so i'm still learning. It's not a bug in the new development commits already.
show-channels is very cool also. Nevermind then, time to go fly. Thanks!

closed 2025-05-16 by arealmess

stronnag, 2025-05-16:
No problem, I needed some sort of "latch" to progress things. Thanks for the inspiration. 
Good luck with the flight.