stronnag/mwptools #221: On windows link to aircraft via UDP doesn't stay active.

xNuclearSquirrel, 2025-05-16:
On Windows, MWPtools receives MAVLink packets on its dual-stack IPv6 UDP socket,
but the first outbound **status/heartbeat** send fails.  MWP then closes the link immediately.

### To reproduce
1. INAV is broadcasting IPv4 MAVLink on **10.0.0.1:14550**.
2. Launch current `mwptools` Windows snapshot  
   ([mwptools_25.05.12_x86_64_installer.exe](https://github.com/stronnag/mwptools/releases/download/25.05.10/mwptools_25.05.12_x86_64_installer.exe)).
3. In the connection dialog enter `udp://:14450.`
4. HUD shows voltage/horizon/heading for < 1 s, then disconnects.

### Expected behaviour
Link should stay active.

### Environment
| Item | Value |
|------|-------|
| MWPtools build | snapshot <2025-05-12> |
| OS | Windows 11 Pro 23H2 (22631.3461) |
| NIC | Intel AX211 Wi-Fi; IPv4 10.0.0.2/24, IPv6 enabled |
| FC, firmware | Speedybee F405 Wing, INAV 8.0 MAVLink |
| Firewall | Windows Defender default (Mwp.exe allowed) |

Raising this issue as requested so the mitigation can be merged into the next Windows build.
Happy to test any PR or nightly you provide.

stronnag, 2025-05-16:
Thanks. It's a Windows incompatibility with the way dual stack IP works on any other OS.

stronnag, 2025-05-16:
Fixed in development branch 0c2cbbbe 
Also reinstates `--force4` (force Ipv4 single stack).

stronnag, 2025-05-16:
There's an untested development [Windows installer](https://seyrsnys-mwptools-win64.surge.sh/)
If you have any problems with it, please prvide a log file, preferably generated with `mwp --debug-flags 4100`

xNuclearSquirrel, 2025-05-16:
Amazing! I tested the Installer, and it works flawlessly on the bench. The connection is stable. Thanks a lot for the quick help! If you need any additional testing, let me know.

closed 2025-05-16 by stronnag