stronnag/mwptools #226: MSP RC

arealmess, 2025-06-30:
I have been cleaning up and updating my steamdeck dev drone with new matek mLRS hardware, and trying to focus in on getting a good understanding of everything so I can build and troubleshoot these kinds of setups in the future.

Where I'm at now:
  clean mwp master branch install
  mLRS works in handset with the lua RC override setting and mspx
  i try different combinations of mspx/transparent and rc override etc
  the UI sticks icon in mwp doesn't appear unless I disable and re-activate MSP RC
  i havent seen them move yet so far
  mwp-hid-test works
  plugging steamdeck mwp directly into FC has the same behavior as mLRS, of not registering inputs at all (i feel like this is a major clue)

So I'm just wondering about any thoughts on this, anything I can do to troubleshoot further, or any more info I can give.

Thanks!

arealmess, 2025-06-30:
I was meaning to ask, do you have a usual process for cleaning and rebuilding the project like when switching branches? My overkill solution is factory reset lol, but I saw in the INSTALL some tips like a ninja uninstall. I think I tried that and noticed some leftover artifacts

stronnag, 2025-06-30:
No, I just install everything  to `--prefix ~/.local` .
The only time special care is needed is if the settings schema changes, which doesn't happen that often/

arealmess, 2025-07-03:
Here's logs trying to connect on dev branch, if there's any clues here. i notice the msp ident error, raw rc none etc. this is with steamdeck connected to FC over usbc. i tested inav configurator and msp is working on that fine. and i get telemetry in mwp. what happens is the sticks dont activate in the UI without turning msp rc off and back on, and there's no response from the inputs. 

I wonder if this is a channel order issue but i'd think that it would work anyways or tell me there's a mismatch
That is maybe the only thing that has changed that i know of

(mwp:89388): Gtk-WARNING **: 15:03:50.209: Unknown key gtk-modules in /home/deck/.config/gtk-4.0/settings.ini

(mwp:89388): Adwaita-WARNING **: 15:03:50.377: Using GtkSettings:gtk-application-prefer-dark-theme with libadwaita is unsupported. Please use AdwStyleManager:color-scheme instead.
22:03:50.427591 mwp 
22:03:50.427708 buildinfo: 2025-07-03T21:38:47Z Linux 6.11.11-valve19-1-neptune-611-g88b36d49a5e3 x86_64 steamdeck
22:03:50.427740 toolinfo: gcc version 15.1.1 20250425 (GCC)  / GNU ld (GNU Binutils) 2.44.0 / Vala 0.56.18-dirty
22:03:50.427769 version: mwp 9fef934a (development) / 2025-06-29T16:17:04+01:00 25.06.29
22:03:50.428105 "Arch Linux" on steamdeck running Linux 6.11.11-valve19-1-neptune-611-g88b36d49a5e3 x86_64
22:03:50.428162 Gtk: 4.18.6 (build) / 4.18.6 (runtime)
22:03:50.428199 Adw: 1.7.5 (build) / 1.7.5 (runtime)
22:03:50.428230 GLib: 2.84.3 (build) / 2.84.3 (runtime)
22:03:50.428260 Shumate: 1.4
22:03:50.428300 Setting "GSK_RENDERER=cairo" for maximum GPU compatibility.
        Export an alternate value from `~/.config/mwp/cmdopts`, `.profile`,
        `/etc/environment` or `~/.config/environment.d/` to use a potentially
        more performant setting, see:
        https://stronnag.github.io/mwptools/mwp-Gtk4-migration-guide/#display-variables-tweaks
22:03:50.428346 WM: plasma-steamos-oneshot / x11
22:03:50.465785 Loaded /home/deck/.local/share/mwp/mwp.css
22:03:50.577096 Using settings schema org.stronnag.mwp
22:03:50.578534 Locale: en_US.UTF-8 (50.9 -1.5 / 50.9 -1.5)
22:03:50.578723 MSP_SET_RAW_RC / HID enabled, full duplex
22:03:50.734367 Registered serial device: /dev/ttyACM0 [0483:5740], Vendor: INAV, Model: STM32_Virtual_ComPort_in_FS_Mode, Serial: 376233653433, Driver: cdc_acm
22:03:50.734484 Registered serial device: /dev/ttyACM1 [28de:1205], Vendor: Valve_Software, Model: Steam_Controller, Serial: 123456789ABCDEF, Driver: cdc_acm
22:03:50.799013 libsecret: The name is not activatable
22:03:50.799277 Start poller sanity timer
22:03:50.819664 Loaded /home/deck/.local/share/mwp/volts.css
22:03:50.825282 Starting Bing proxy 
22:03:50.853469 Failed to find "blackbox_decode" on $PATH; see https://stronnag.github.io/mwptools/replay-tools/
22:03:50.853856 Failed to find "gnuplot" on $PATH
22:03:50.854135 Failed to find "fl2ltm" on $PATH; see https://stronnag.github.io/mwptools/replay-tools/
22:03:50.854367 Failed to find "mavlogdump.py" on $PATH; see https://stronnag.github.io/mwptools/replay-tools/
22:03:50.854418 FYI, PATH is /home/deck/.local/bin:/home/deck/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/var/lib/flatpak/exports/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/sbin:/sbin:/bin
22:03:50.854457 :DBG: No TA for you!
22:03:50.854526 Mask 0, api [espeak]
22:03:50.854570 Using speech api 0 [none]
22:03:50.854601 Initialised "none" for speech
22:03:50.857230 Generated GCS icon
22:03:50.874721 Set up http 3 radar device adsbx://api.adsb.one?range=40&interval=1000
22:03:51.073392 Main window activated false
22:03:55.290158 Trying OS open for /dev/ttyACM0 STM32_Virtual_ComPort_in_FS_Mode
22:03:55.293624 Connected /dev/ttyACM0 STM32_Virtual_ComPort_in_FS_Mode (nopoll false)
22:03:55.314649 HID process: pid=89428 81 (true)
22:03:55.314736 Requesting HID Info
22:03:55.331785 Msp Error: MSP:IDENT 0x64/100 [0b] 
22:03:55.339865 Using Msp v2 0205
22:03:55.349947 Model name: "no-name"
22:03:55.360227 V2 mixer 3 0
22:03:55.390289 Feature Mask [10480c86] : telemetry true, gps true, current true
22:03:55.425975 INAV v8.0.0  PRINCIPIOTF7 (ec2106af)
22:03:55.443406 Initial cycle 0.152089 (b=381 r=2505.1, bi=20040.9) 
22:03:55.443775 Boxids: [ 0 51 61 1 2 35 5 8 6 7 32 11 10 28 53 45 30 31 55 59 46 3 13 60 19 26 27 39 40 41 42 43 44 52 62 63 65 66 67]
22:03:55.443822 Masks arm=1 angle=8 horz=10 ph=800 rth=1000 wp=2000 crz=8000 fs=4000000
22:03:55.450824 RX_MAP 3 2 0 1
22:03:55.460778 WP_GETINFO: 120/0/0
22:03:55.473076 Activeboxes 8 04000008 00000000
22:03:55.473165 Requesting common settings
22:03:55.473202 Request setting nav_wp_max_safe_distance
22:03:55.473242 Request setting inav_max_eph_epv
22:03:55.473290 Request setting gps_min_sats
22:03:55.473326 Request setting safehome_max_distance
22:03:55.473362 Request setting nav_fw_land_approach_length
22:03:55.473396 Request setting nav_fw_loiter_radius
22:03:55.580233 Received nav_wp_max_safe_distance: 20000m
22:03:55.683592 Received inav_max_eph_epv: 1000
22:03:55.765686 Received gps_min_sats: 6
22:03:55.875459 Received safehome_max_distance: 100m
22:03:56.044536 Received nav_fw_land_approach_length: 350m
22:03:56.158807 Received nav_fw_loiter_radius: 75m (false)
22:03:56.168139 Arming flags: Navigation unsafe, • Fix quality, • 0 satellites,RC Link (40800), load 63% MSP_CMDS_INAV_STATUS
22:03:56.168440 Sensors: Acc Baro Mag GPS  (008f)
22:03:56.169130 INAV v8.0.0  PRINCIPIOTF7 (ec2106af) «no-name»
22:03:56.169201 Poller cycle for  9 items, 81 / 226 bytes (INAV_STATUS,ANALOG2,NAV_STATUS,RAW_GPS,COMP_GPS,GPSSTATISTICS,WP,ATTITUDE,ALTITUDE)
22:03:56.169268 Start poller
22:03:56.431239 Raw RC: none
22:04:05.700420 :DBG: msprc action set to false, use_rc=0
22:04:08.524520 :DBG: msprc action set to true, use_rc=86

stronnag, 2025-07-04:
You are using an **old** revision of the development branch where the **HID server is known to be broken**.
See  https://github.com/stronnag/mwptools/issues/226#issuecomment-3020324737

```
# Your log
22:03:50.427769 version: mwp 9fef934 (development) / 2025-06-29T16:17:04+01:00 25.06.29

# Git log

commit a6bbf61b81f5d94a3e44be584d3dc8ef2af682b6
Author: Jonathan Hudson <jh+github@daria.co.uk>
Date:   Mon Jun 30 19:11:50 2025 +0100

    workaround timeouts due to something upsteam not delivering MSP replies
    fix hid crash with no device

commit 9fef934aa2b4460eb9203f4cc0ac1135075d15e5 (winclose)
Author: Jonathan Hudson <jh+mwptools@daria.co.uk>
Date:   Sun Jun 29 15:37:09 2025 +0100

    clean i/o chan close for win32

```